ev3dev/ev3dev #420: Add LEGO 9V motors to motor docs.

dlech, 2015-11-02:
We are now setup to automatically generate documentation on motors. We could use someone to add any 9V LEGO motor products to [this file](https://github.com/ev3dev/lego-linux-drivers/blob/master/motors/rcx_motor_defs.c).

Basically, you just copy the existing entries and change the part name and number and url.

LEGO doesn't have any of these on their website, so we can use bricklink or another similar site. Here is the list of motors: http://www.bricklink.com/catalogList.asp?q=Electric%20Motor%209V

BertLindeman, 2015-11-03:
Took the search result and not sure what to add;

drop Microromotor 2 x 2 Base Cover and Pulley

Already in:
```
99499c01;Electric, Motor 9V Power Functions L with Dark Bluish Gray Bottom;http://img.bricklink.com/P/86/99499c01.jpg
58120c01;Electric, Motor 9V Power Functions M with Dark Bluish Gray Bottom;http://img.bricklink.com/P/86/58120c01.gif
99498c01;Electric, Motor 9V Power Functions Servo with Dark Bluish Gray Bottom;http://img.bricklink.com/P/86/99498c01.jpg
58121c01;Electric, Motor 9V Power Functions XL with Dark Bluish Gray Bottom;http://img.bricklink.com/P/86/58121c01.gif
```
Remaining:
```
47154c01;Electric, Motor 9V 4 x 4 x 3 1/3;http://img.bricklink.com/P/12/47154c01.gif
bb129;Electric, Motor 9V 4 x 10 x 3 1/3;http://img.bricklink.com/P/11/bb129.gif
2838c01;Electric, Motor 9V 5 x 4 x 2 1/3;http://img.bricklink.com/P/9/2838c01.gif
2986;Electric, Motor 9V Micromotor 2 x 2;http://img.bricklink.com/P/5/2986.gif
43362c01;Electric, Motor 9V Mini-Motor, newer lighter weight;http://img.bricklink.com/P/9/43362c01.gif
71427c01;Electric, Motor 9V Mini-Motor, older heavier weight;http://www.bricklink.com/catalogItemPic.asp?P=71427c01
87577c01;Electric, Motor 9V Power Functions with Gear Reduction Ratio 9.5 : 1;http://img.bricklink.com/P/86/87577c01.jpg
590;Electric, Train Motor 9V Modern;http://img.bricklink.com/P/11/590.gif
x1688;Electric, Train Motor 9V RC Train (motor only, no attached wheels);http://img.bricklink.com/P/11/x1688c01.gif 
```

Not sure if the train motor 590 here  is the same as the train motor already in the file.

And the naming convention.
E.g. the Power Functions with Gear Reduction Ratio 9.5 might be named
`LEGO_PF_GEAR_REDUCTION_MOTOR` ?

dlech, 2015-11-03:
> E.g. the Power Functions with Gear Reduction Ratio 9.5 might be named

This one is actually on th LEGO Education website: <https://education.lego.com/en-us/lego-education-product-database/machines-and-mechanisms/9670-e-motor>, so I would like to use that URL and the name from that page, "9670 E-Motor". So `LEGO_PF_E_MOTOR` sounds right.

The other motors, I would call something like `LEGO_9V_SOMETHING`

Also, I think this page will be useful: <http://philohome.com/motors/motorcomp.htm>

The train motors that get power through metal wheels are not directly connectable to the EV3, so let's leave them off the list. That would be 10153-1 and 590 from looking at the picture. And it looks like 87574c01 is the power functions train motor.

BertLindeman, 2015-11-03:
Thanks, will dig a bit deeper and try to get an update for this.

BertLindeman, 2015-11-03:
Loosing track ;-(  Sorry..

In `http://www.ev3dev.org/docs/motors/` there *is* already an `9670 	Power Functions E-Motor`
but I do not see it in the source
```
https://github.com/ev3dev/lego-linux-drivers/blob/master/motors/rcx_motor_defs.c
```
Am I looking in the wrong repository (master) or are more definition files involved.

dlech, 2015-11-03:
Don't look at the existing `http://www.ev3dev.org/docs/motors/`. It is being completely replaced.

https://github.com/ev3dev/lego-linux-drivers/blob/master/motors/rcx_motor_defs.c is the file we want to edit. The motors page will be automatically generated from this information.

BertLindeman, 2015-11-03:
Ah, right, thanks.

dlech, 2015-11-04:
Thanks for your help. :+1:

closed 2015-11-04 by dlech