ev3dev/ev3dev #423: Mono installation on ev3

pgrudzien12, 2015-11-07:
I'm trying to install mono-complete package on EV3. I tried same thing on RaspberryPi 2 rpi2-ev3dev-jessie-2015-09-15.img and it was ok. Then tried on EV3 with rpi-ev3dev-jessie-2015-09-15.img. 

I'm stuck with resolving dependencies. Mono-complete needs mono-devel and it in turn needs libgdiplus whitch needs libjpeg8 >= 8c.

When tried to install libjpeg8 I got this:
```
root@ev3dev:~# apt-get install libgdiplus
Reading package lists... Done
Building dependency tree
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libgdiplus : Depends: libjpeg8 (>= 8c) but it is not installable
E: Unable to correct problems, you have held broken packages.
```

Do you have any idea why on rpi2 it is available but on ev3 it is not? I have to say that I am not a linux ninja. Maybe you know some source where I can get package that is needed?

pgrudzien12, 2015-11-07:
Never mind. I was able to work around this particular problem by installing mono 3.2 and then upgrade.

dlech, 2015-11-08:
I don't think Xamarin builds all of the packages for the armel architecture, which is what the EV3 is. Raspberry Pi 2 is armhf, which Xamarin does support.

So, on the EV3 you probably have a mix of 3.2 packages from debian and and 4.x from Xamarin.

pgrudzien12, 2015-11-08:
Thanks @dlech for your support. Although I did not succeed I decided to publish description of my attempt on blog (if you are interested) http://bleedingnedge.com/2015/11/08/asp-net-5-on-lego-mindstorms-ev3-using-ev3dev/ And moreover I opened discussion at DNX repo here: https://github.com/aspnet/dnx/issues/3145

closed 2015-11-08 by pgrudzien12

pgrudzien12, 2015-11-09:
Awesome ! I have true solution. Seems that xamarin have full support for armel: http://origin-download.mono-project.com/repo/debian/dists/wheezy/main/binary-armel/

The problem was I was missing repository for `libgdiplus`:

    echo "deb http://download.mono-project.com/repo/debian wheezy-libjpeg62-compat main" | sudo tee -a /etc/apt/sources.list.d/mono-xamarin.list

Seemingly RPI2 has libgdiplus alredy installed? Nevertheless I am going to have mono v4 on my ev3 tomorrow and if everything goes well DNX as well.