ev3dev/ev3dev #425: Some I2C sensors don't work on PiStorms

dlech, 2015-11-09:
There is a hardware bug in the I2C on Raspberry Pi that causes data corruption on devices that use I2C clock stretching.

Hardware bug references:
* https://github.com/raspberrypi/linux/issues/254
* http://www.advamation.com/knowhow/raspberrypi/rpi-i2c-bug.html

In the kernel that was just released (4.1.12-8-ev3dev-rpi), there is a gpio bit banging I2C driver that can be used instead of the hardware I2C. I think we should experiment with that and see how it does.

dlech, 2015-11-09:
List of known sensors that don't work:

* mindsensors.com NxtMMX

If you find more, leave a comment and I will add them to this list.

micgil, 2016-03-17:
Try changing the I2C speed to 40K or 35K and checking the NXTMMX. This has worked for me,

dlech, 2016-03-17:
Thanks for the tip. I assume you did this using `dtparam=` in `config.txt`. Can you please post an example for others that might want to try this?

micgil, 2016-03-25:
There is a very brief description of the process at [](http://www.mindsensors.com/blog/how-to/change-i2c-speed-with-raspberry-pi)

micgil, 2016-03-25:
Not sure if the link went through. 
http://www.mindsensors.com/blog/how-to/change-i2c-speed-with-raspberry-pi