ev3dev/ev3dev #426: Touchscreen support for brickman

dlech, 2015-11-09:
The PiStorms only has a GO button, which is mapped to ENTER. There is no way to navigate in brickman.

As a workaround, you can attach a keyboard or even just a numeric keypad (if it has a backspace button) via USB or Bluetooth or use `sudo conspy` from a remote terminal.

dwalton76, 2016-05-22:
@dlech is this one on the roadmap?  If not I'm thinking why not use a REST API so I can launch a web page on my smart phone to send up, down, left, right, enter and backspace commands.  It wouldn't be as nice as native touchscreen support but would be better than tethering a keyboard.

dlech, 2016-05-22:
I'm actually working on this. Of course, that doesn't mean that I will finish before I get distracted with something else.

If you would like to help, we need to come up with a design of how navigation should work in Brickman.

Do we make an on-screen keyboard with the six navigation buttons? 

Or do we make it more like a smartphone? For example, if we touch something that is focusable, like a button, it should focus the button and trigger the same signal as the ENTER button. But how do we go back? Do we add a touchable back button to each screen (and where on the screen)? I don't think that gestures like a smartphone will be an option (although I haven't got that far yet, so I could be wrong).

Is there a "magic" place to touch that will end a user program the same way that holding down the back button on the EV3 works?

Any sketches of what it might look like would be helpful.

WasabiFan, 2016-05-22:
Design-wise, I think you should keep in mind that you can detect the duration of touch events (at least, I assume you can). So, maybe holding down on the screen is what you are looking for: when you are in the menus, long-pressing goes up a level, same as the back button. When you are running a user app, long-pressing kills the program. You can also theoretically use double-tapping.

dwalton76, 2016-05-22:
What if the Go button were remapped to Backspace? Then the touchscreen could act like a smartphone where a touch focuses and presses enter....would save you from having to change the brickman display.

dlech, 2016-05-22:
Maybe. It's a bit counter-intuitive have a "go" button go back instead of go forward. Also, it means that I would have to add a button to the [display I added to my BrickPi](http://lechnology.com/2016/05/adding-a-display-to-brickpi/). It's doable, but I was hoping to get something that works with touchscreen only.

dwalton76, 2016-05-22:
I could get used to "go" meaning "back" but yeah that approach wouldn't work for your brickpi.  I have not touched brickman at all but from a UI perspective having a "Back" button in the corner of every screen seems fine.

dlech, 2018-01-03:
suggestion from @ndward:

> In the Brickman file browser the left and right buttons seem to be 'wasted', duplicating the functionality of the up button. To make these buttons more useful I suggest that the left button could be used to move to the parent folder (unless already in home/robot) and the right button could be used to move to the first file in the list. Also, I would expect that if I hold down the up or down buttons then I should automatically scroll slowly through the files, which currently does not happen. Currently, I sometimes have to press the up or down buttons dozens of times to reach the fie I want, if the folder contains dozens of files, which is a drag.

R2D2Hud, 2020-05-06:
Swipe left to go back?

cospan, 2020-09-02:
Hi,

I was working on adding touch support to my PiStorm V2 with some results.

[Fork with Touch Event](https://github.com/cospan/ev3devKit/tree/feature/add-touch-event)

I put a lot of details in the commit message but here is the summary

**Status**

Things sort of work, you can touch the screen and activate buttons, more work is required to support moving though. There are two issues.

**Issues**

1. The 'TOUCH_UP' event is never fired off, I printed out the enumerated touch events, '10' is 'TOUCH_DOWN' and '11' is 'TOUCH_MOVE' but I never see '12' which, I believe, should be 'TOUCH_UP'.
2. The touch screen calibration seems to be off, perhaps I'm not saying this right but only the bottom left of the screen detects button presses. Basically the bottom half of the screen to about two thirds across the screen registers touches.

**Kernel: Go Button/Touch Screen**

I did make a small modification to the kernel source, or actually the git submodule, to make the 'Go' button a back button and that worked out well. The change was so small I didn't really see a need to fork the repo.

I initially thought that the touch screen calibration might have been a kernel configuration thing that I could fix with overlays or DTS but since those drivers are not controlled by the DTS I couldn't easily modify them.

I did find the drivers and it seems that the screen is 'rotated' within the driver by '90' degrees which does explain why the bottom left works but does not explain why the entire touch screen doesn't work. Is it possible the 'X' 'Y' screen flip within the driver is incorrect? I should try this.

**Next Steps**

There are a couple of paths to take from here.
If we can get the 'TOUCH_UP' to be detected we can use the touch screen as we would with other devices, pressing on a 'button' would activate it, using the 'TOUCH_MOVE' would allow us to use scroll and a quick 'TOUCH_DOWN' and 'TOUCH_UP' could be used to activate buttons but as it is I don't see how we can discern a button press from a scroll without knowing if the user has lifted the finger, perhaps I'm thinking of this wrong, let me know your thoughts.

 If we can't easily find a 'TOUCH_UP' solution then we can forego all motions and just add buttons to scroll up and down. I personally would like the 'TOUCH_UP' to work but I dug into the GRX code and couldn't find a place where 'TOUCH_UP' is generated, in fact it seemed as though GRX was simply a conduit for 'glib' and GRX was simply translating the glib values to GRX so it seems as though the 'TOUCH_UP' is within glib.h. Thoughts?

dlech, 2020-09-02:
Nice to see someone taking an interest in this. 

Did you [calibrate the touchscreen](https://github.com/ev3dev/grx/wiki/Touchscreen-Support)? That should fix some of the issues you are seeing.

I think some devices give `TOUCH_CANCEL` instead of `TOUCH_UP` so it is probably best to check for both and treat them as the same event.

A few years ago, I tried to split out the widgets from ev3devkit into a separate library and add touch support. But I think I tried to change too much at once and never finished. You can find that effort here: https://github.com/ev3dev/grx-widgets