ev3dev/ev3dev #427: EV3 not connecting to Computer with USB

bcorey, 2015-11-09:
I followed the "Getting Started" instructions from the ev3dev website and everything worked well until I turned it off. Now the EV3 won't reconnect to the computer through USB. I went through the whole setup process a second time, but have the same issue after turning the EV3 off. The disc image file I put on the micro SD was "ev3-ev3dev-jessie-2015-09-13.img".
The computer recognizes that a USB device called "EV3+ev3dev" is plugged in, but the "Hardware" tab inside the device's "Properties" window says "Device status: this device cannot start". Pinging the ev3 from the command prompt times out. The "Wired" section under "All Network Connections" does not show up in Brickman.

dlech, 2015-11-09:
Do you have another way to connected to the EV3? Bluetooth, Wi-Fi, wired USB Ethernet?

bcorey, 2015-11-09:
My computer doesn't have Bluetooth and I don't have a Wi-Fi dongle for the EV3. Isn't wired USB Ethernet what I'm trying to do, or is that something else?

dlech, 2015-11-09:
You are trying to do Ethernet over USB, which is different from a [wired USB adapter/dongle](http://www.amazon.com/s?ie=UTF8&page=1&rh=i%3Aaps%2Ck%3Ausb%20wired%20network%20adapter).

If you don't have another way to connect, you have a couple options.

1. You can connect a keyboard to the EV3 and login via the screen on the EV3 and we can troubleshoot further.
2. You can try flashing the image again and this time, don't upgrade any packages.
3. Wait a few weeks for a new image release.

bcorey, 2015-11-09:
I'll try flashing the image again without upgrades, and if that doesn't work I'll wait for the next release since I don't have any immediate need for ev3dev (I need to brush up on Python anyway).
Thanks for your help, things make more sense now. :)

bcorey, 2015-12-01:
Flashed the image without upgrades and the connectivity problem is gone. :D

closed 2015-12-12 by dlech