ev3dev/ev3dev #428: Updating regular FW using ev3dev

laurensvalk, 2015-11-11:
Is it possible to flash the [LEGO FW bin file] (http://esd.lego.com.edgesuite.net/digitaldelivery/mindstorms/57cd7e71-5c65-4651-a47e-29d6a83f8980/public/EV3%20Firmware%20V1.08H.bin) while running ev3dev? If so, how?

I'd like to do this because the normal update method doesn't work for me. When the EV3 is in update mode, the USB device won't enumerate on my host PC. This is Ubuntu, but I'm hearing some Mac users have a similar problem. 

Clearly that's not an ev3dev issue, but I thought maybe ev3dev has a workaround.

Perhaps not really a typical use case now, but maybe it will be at some point for ev3devLite, so it can update itself.

dlech, 2015-11-11:
Not currently. In fact we don't even have the spi driver to access the flash setup in the board configuration file. This is something that we will have to get working for ev3devLite though.

laurensvalk, 2015-11-11:
Okay, thanks. I'll close this issue for now then.

closed 2015-11-11 by laurensvalk