ev3dev/ev3dev #430: ssh connection closed

osmado, 2015-11-12:
Hi,

I am new user.  I have a ev3 mindstorm and I am using your ev3dev software.  You are making a great job.  I followed the "Getting started" section and all run fine (use bluetooh to connect my computer to the EV3) whit ev3-ev3dev-jessie-2015-07-08 image. I would like to continue the "ros-project" and I followed the instructions described in this project to compile ros: : https://github.com/moriarty/ros-ev3/blob/master/brickstrap-build-status.md

After that, I made a new image (also I followed the project instruction).  Then I wrote the image into de SD Card and plug into the EV3.  I setup the bluetooth connection and that was OK.  But when I try to make an ssh connection the EV3 close it:

oscar@monasterio:~/otro_brickstrap_ws$ ssh -vvv root@192.168.3.2
OpenSSH_6.6.1, OpenSSL 1.0.1f 6 Jan 2014
debug1: Reading configuration data /etc/ssh/ssh_config
debug1: /etc/ssh/ssh_config line 19: Applying options for *
debug2: ssh_connect: needpriv 0
debug1: Connecting to 192.168.3.2 [192.168.3.2] port 22.
debug1: Connection established.
debug1: identity file /home/oscar/.ssh/id_rsa type -1
debug1: identity file /home/oscar/.ssh/id_rsa-cert type -1
debug1: identity file /home/oscar/.ssh/id_dsa type -1
debug1: identity file /home/oscar/.ssh/id_dsa-cert type -1
debug1: identity file /home/oscar/.ssh/id_ecdsa type -1
debug1: identity file /home/oscar/.ssh/id_ecdsa-cert type -1
debug1: identity file /home/oscar/.ssh/id_ed25519 type -1
debug1: identity file /home/oscar/.ssh/id_ed25519-cert type -1
debug1: Enabling compatibility mode for protocol 2.0
debug1: Local version string SSH-2.0-OpenSSH_6.6.1p1 Ubuntu-2ubuntu2.3
debug1: Remote protocol version 2.0, remote software version OpenSSH_6.7p1 Debian-5
debug1: match: OpenSSH_6.7p1 Debian-5 pat OpenSSH* compat 0x04000000
debug2: fd 3 setting O_NONBLOCK
debug3: load_hostkeys: loading entries for host "192.168.3.2" from file "/home/oscar/.ssh/known_hosts"
debug3: load_hostkeys: loaded 0 keys
debug1: SSH2_MSG_KEXINIT sent
Connection closed by 192.168.3.2
oscar@monasterio:~/otro_brickstrap_ws$ 

Thanks in advance.

dlech, 2015-11-14:
The ros project has not been updated in over a year and ev3dev has changed much since then. Are you able to connect using the latest image?

osmado, 2015-11-14:
I think that I am using an updated version (kernel 3.16.7 ctk16-7 ev3dev.ev3).   Due to ros image did not work I made another one, but in this case I made a basic image (no ros software). It did not work too. Thanks.

dlech, 2015-11-14:
Can you ping the EV3? What size of SD card are you using?

osmado, 2015-11-15:
Yes, I can ping the EV3 and my image size is:
IMAGE_FILE_SIZE=${IMAGE_FILE_SIZE:-"2700M"}

dlech, 2015-11-15:
Did you install brickstrap via apt or are you using the latest version from git?

osmado, 2015-11-15:
via apt (synaptic shows v 0.3.3)

dlech, 2015-11-15:
It sounds like you are doing everything correctly, so I am not sure what the problem could be. We occasionally have a report that the SSH host keys are not generated correctly during the first boot, but this is usually fixed by re-flashing and trying again.

It is also possible that the SSH configuration on your host computer may set to now allow password authentication or something like that.

There are also some significant changes in brickstrap since the last release. You could try building an image using brickstrap from git. If you do that though, you won't be able to login as root. The new default login will be `robot` with password `ev3dev`.

osmado, 2015-11-15:
Ok, I will install brickstrap from git and try again. I will tell you the result. Thanks.

osmado, 2015-11-24:
Hi again,
After several tries, I have been able to setup the ssh connection.  Yesterday I read the brickstrap issue#9 and included ev3-systemd into preinst.blacklist file.  After that, all worked fine.  Maybe in previous tries I did not pay attention to the modprobe error, I do not remember it.
I will continue with ROS integration. I would like to make some type of contribution in a short period of time. 
Thanks for your time.

dlech, 2015-11-24:
Glad you got it sorted.

closed 2015-11-24 by dlech