ev3dev/ev3dev #431: Use Logo for a graphics library.

dlech, 2015-11-16:
It has been mentioned a few times before that we need a way to do graphics on the screen that is as easy to use as the other hardware drivers. The idea is that you have something like a sysfs attribute that you write commands to, like `rectangle 10, 10, 20, 20` or something like that. So I was thinking about it and it occurred to me, why not use Logo? It's super simple (it was designed for teaching kids to program) and it means we don't have to come up with our own graphics "language".

Obviously, for languages where there is already a graphics library (like PIL/Pillow for python), that should be used instead. But, I think this would be a great way to fill in the gaps for languages that don't have such libraries. And it can be useful for remote applications as well.

It seems like [UCBLogo](http://www.cs.berkeley.edu/~bh/logo.html) is the flagship open source implementation. There isn't a Linux framebuffer implementation, so we would need to add that. But it would be pretty simple using our [GRX](https://github.com/ev3dev/grx) graphics library.

I found someone who has taken a recent interest in UCBLogo and they have a [fork on GithHub](https://github.com/grinner/ucblogo). We just need to fork it and implement the functions in <https://github.com/grinner/ucblogo/blob/master/nographics.h>.

I think the interpreter part is terminal aware, so we would need to find a way to pipe it to stdin/stdout so that it could be used programmatically in addition to interactively.

dlech, 2015-11-16:
<https://github.com/Distrotech/ucblogo> looks like an active version as well.

rhempel, 2015-11-16:
Hmmmmm - think about this I must.

dlech, 2015-11-16:
To give an example of how powerful logo is, here is a "paint" program that I just wrote for the touchscreen on PiStorms. It just draws wherever you touch.

    while [ 1 = 1 ] [ ifelse [button?] [pendown] [penup] setpos mousepos]

Yeah, one line.

rhempel, 2015-11-16:
I know LOGO is powerful - it's one of those things I played with as a kid! Of course, we'll need a LOGO binding now too :-) 

So we can draw lines, does it do arcs, circles, fills, and text?

dlech, 2015-11-16:
There is no built-in arcs/circles, but it is easy enough to write a procedure. And you can load procedures and macros from files. Fills and text are built-in. And is has a shell command for everything else that is not built-in.

rhempel, 2015-11-16:
So how does LOGO help us more than just making a parser for commands that get passed to grx directly?

dlech, 2015-11-16:
we don't have to write the parser.

dlech, 2015-11-16:
and it has a :turtle:

rhempel, 2015-11-16:
As much as I like the idea of LOGO, I can argue that we can make a nice little DSL (Domain Specific Languge) in Lua too.
In both cases, we'd have to bind the language to a grx.so (or similar) library - I know how to do that for Lua, how much work is it for LOGO?

dlech, 2015-11-16:
Logo and Grx are both written in C. There is a link in my first comment to a file of the functions we have to implement. This list may seem a bit long, but each one can probably be implemented in one or just a few lines.

dlech, 2015-11-16:
There is some extra work involved in modifying the input and output to use stdin/stdout.

dlech, 2017-10-03:
Interesting idea, but probably not going to happen.

closed 2017-10-03 by dlech