ev3dev/ev3dev #432: Documentation: iPython setup guide

laszewsk, 2015-11-21:
Not sure if this is helpful for others, but I wrote down the steps on how i did iPython so others can replicate. 

https://github.com/laszewsk/mindstorm/blob/master/ev3dev/bin/README.rst

As I  do not think it belongs in the project category, I just let you know the link. I can clean out the document (such as remove my home made casters) if there is interest in hosting this is some form on the Web page or make a link to it.

dlech, 2015-11-21:
I think others would appreciate this. I think the best place for it would be <https://github.com/ev3dev/ev3dev.github.io/tree/master/docs/tutorials>.

Also, we could consider adding something to brickman to start and stop the iPython notebook similar to what we are working on for OpenRoberta.

laurensvalk, 2015-11-22:
Nice. How well is iPython working for you? I tried it [a while ago] (https://www.youtube.com/watch?v=vJHr6vIZk0k) and it was all a little too slow. I had to extend some of the time out settings in the config file to ensure it wouldn't just give up occasionally and shut down. But maybe I had just configured it the wrong way.

By the way, I came across one small typo in the instructions:

> apt-get install ipython-noxtebook

@dlech wrote:
> Also, we could consider adding something to brickman to start and stop the iPython notebook similar to what we are working on for OpenRoberta.

Having something like iPython Notebook (now called [Jupyter] (https://jupyter.org/)) preconfigured by default and easy to launch from brickman would be very cool. Since it has a terminal option as well (also in the browser), it could be a neat cross platform solution for those getting started.

psychemedia, 2016-04-15:
Running the notebook server from the brick seems quite slow to me, but it is possible to have a Jupyter server running on a desktop machine and then spawn a remote IPython kernel on the brick. I've done a proof of concept here: https://blog.ouseful.info/2016/04/15/thinking-around-the-edges-lego-ev3-robots-running-remote-jupyter-kernels/

psychemedia, 2016-05-12:
I just posted a how0to on getting up and running with IPython on Ev3; there's also a demo of how to use connect to IPython as a remote kernel from a Jupyter notebook server running elsewhere (as in, on my Mac): [Using IPython on Lego EV3 Robots Running Ev3Dev](https://blog.ouseful.info/2016/05/12/using-ipython-on-lego-ev3-robots-running-ev3dev/)

FWIW, I also found I could use an old Samsung Galaxy S3 phone without a sim-card and running the Hotspot Control app to act as a wifi router to connect my laptop to the brick (only issue was it was assigned a dynamic IP).

psychemedia, 2016-05-14:
Further to the previous comment, here are a couple of (work in progress) [example notebooks](https://gist.github.com/psychemedia/1e2282149a17f5c16252c1ed7c8c6ce6) (using old ev3dev Python bindings) that present one take on how we could use notebooks running on one server to connect a remote IPython kernel running on the brick.

I also managed to get *ipywidgets* installed on the brick so you can use them in the notebook.

[image]

To make things even easier on the host desktop/laptop/server side (where the ev3 is the client), I think it might be useful to have a Docker container with the Jupyter notebook and *irkernel* package installed, along with a script/dashboard to help set up the connection definition file the first time. Two issues with this: a) I'm not sure how this would work in terms of setting up ports for the Jupyter server to tunnel to the remote ev3 IPython kernel? b) the brick would have to have  a public / internet observable IP address if the container was running in the cloud, for example?

laszewsk, 2016-05-14:
I am running docker usually on my local machine and tunnel via the local network should be possible directly to the brick while using the assigned ipadress (for other projects). So one thing one could do is if you run this in the cloud. is to do a tunnel form the laptop to the cloud and vice versa

e.g.

brick   <-  tunnel -> laptop <-  tunnel -> vm on cloud

Also if Tony or others that want to work on this needs some vms to try things out for this project we  could spawn them up on our cluster (after we vet you and get approved). However they sit in the US so if you are in Europe it may be a bit slow to delegate the commands. 

I also wrote a wrapper around vagrant to manage some vms via virtualbox to simulate a cloud. As we  found vagrant is just to complex for the students to work with as yet another prg language is introduced, I wrote a small python wrapper around vagrant. So this way you have an easy way to manage the vms in python on virtualbox but still have the ability to customize with vagrant if needed. Hence, In contrast to managing multiple vms in one vagrant file this tool creates a vagrant file for each vm automatically simplifying management of tons of vms.

https://github.com/cloudmesh/vagrant <https://github.com/cloudmesh/vagrant>

Also this comes with a simple API that you could even use directly in ipython to start the vms and than do the connection.

laszewsk, 2016-05-14:
This also reminds me about another important point not really related to ev3dev, but in general how we use ev3dev in the field.

The issue is actually that due to regulations the internet at schools is often not public. So what we need to do is to set up a local completely from the internet isolated network. I tried some of my old routers, but it seems that they shut themselves down if they do not detect a connection to internet.  So what we need is actually a “working” hardware setup with specific routers and so on to set up a completely isolated network just for the bricks and the laptops.

brick 1          < —— ISOLATED NETWORK —>  laptop 1
brick N                                                                  laptop 2

We tried using bluetooth for the bricks but encountered SIGNIFICANT issues with connection reliability and startup. Students spend more time trying to connect to the brick via bluetooth than to actually program. However I have the edimax wifi usb plugin which works great. But as i do not have a router that works i can not utilize it in such isolated environments. 

Anyone having a working solution for this with specific hardware that has proven to work?

psychemedia, 2016-05-14:
@laszewsk thanks for sharing that (we're using VMs to ship s/w for a distance ed course, though in that case students install the VM using Virtualbox on their own machines; I had a dockerised solution but we canned it because of low spec requirement to support 32 bit Windows machines).

Back on the remote kernel ev3 stuff, looking at the Jupyter logs, there seems to be some tunnelling going on.

I don't know much about networking, but if the tunnels go through arbitrary ports, how do you work out what ports to expose when defining the container ?

[image]

Or can Jupyter hub / IPython remote kernel connection file be configured to tunnel out through specified container ports? Or do port requests result from a negotiation with the remote kernel? (You can probably tell I'm way out of my depth and have no idea what I'm talking about here!;-)

psychemedia, 2016-05-14:
Just in terms of the architecture, I was thinking:

1) Fully remote robot lab:

laptop <-> cloud container running Jupyter
Jupyter <-> ev3

2) Self-containe remote robot lab:
user BYOD browser <-> lab network Jupyter server
lab network Jupyter server <-> lab network ev3

3) at home, the test case I'm running is:
laptop desktop jupyter <-> home router
home router <->  ev3

My current use case is setting up a room for a residential school.

My [original thinking](https://blog.ouseful.info/2016/04/15/thinking-around-the-edges-lego-ev3-robots-running-remote-jupyter-kernels/) was as follows, though I may need to revise this in light of recent experiments:

*We need 8 desktop computers and 8 EV3s.*

*One thing we’d need to do is find a way of marrying computers and EV3s. Specifically, we need to have known IP addresses associated with specific bricks (can we assign a known IP based on MAC address of the wifi dongle attached to the bricks?) and we need to make sure that passwordless ssh works between the computers and the EV3s. The easiest way of doing the latter would be to use the same ssh keypair in every machine, but this would mean student group A might be able to mistakenly connect to the Student Group B’s machine.*

*One way round it might be to set up a Jupyterhub server that is responsible for managing connections. The Jupyterhub server is a multiuser hub that will set up a Jupyter server for each logged in user. If we set up the Jupyterhub attached the same wifi network as the EV3 bricks, with 8 users, one per student group, each user/student group could then presumably be assigned to one of the bricks. Students would login to the Jupyterhub, and their Jupyter server would be associated with the remote_ikernel kernel.json file for one of the bricks. Anyone who can connect to the local wifi network can then login to the Jupyterhub as one of the group users, launch a notebook, and run code on one of the bricks. This means that students could write notebooks from their own laptops, connected to the network. Wifi-less desktop machines provided for the residential school could presumably be added into the network via a local ethernet cable?*

*So we’d have something like this:*

[image]

There are some more sketches here: [Robotlab thinkses](http://www.slideshare.net/psychemedia/robotlab-jupyter)

*(I have to go out for the day now...)*

laszewsk, 2016-05-14:
ok. The network ip if run on regular virtualbox can be found via two commands 

a) first you need to identify the “real” name of the vbox. In your case you probably know it. But just in case you can do 

vboxmanage list runningvms

b) than you can find the details including the ip with

vboxmanage showvminfo <vmname> --details
vboxmanage guestproperty enumerate <vmname>

The last command will have the network information and one can fgrep to it.

I think this should work.

PS: Probably sticking with the vms for now is better till MS and Apple support containers natively. I have beta docker, but this is not yet public so i would wait till this all becoms available. THe current docker runs anyways in a vbox so ther is no real benefit at this time as you already have a vm

laszewsk, 2016-05-14:
Our issue is that this will not work:

"Wifi-less desktop machines provided for the residential school could presumably be added into the network via a local ethernet cable?”

As NO laptop can be added to the local network. So we need to run our own network completely separate from the school network :(

rhempel, 2016-05-14:
Consider using Bluetooth networking, you can easily fix the IP over Bluetooth address

psychemedia, 2016-05-14:
@rhempel I've not had much success with bluetooth and have abandoned it as an approach.

laszewsk, 2016-05-14:
Same here. As said when we do bluetooth we spend a lot of time to get it to work. We abandoned it also.

rhempel, 2016-05-14:
Too bad, it's what I use for connecting with the bricks, doing remote editing with FileZilla - works great here :-)

rhempel, 2016-05-14:
But seriously, did you try Bluetooth recently, and what is your host OS

laszewsk, 2016-05-14:
I saw significant differences in reliability of Bluetooth setup on OSX vs Windows by the way.
I have OSX and its really a pain to get it working for some reason. Users with WIndows laptops had a much easier time.

laurensvalk, 2016-05-15:
> Same here. As said when we do bluetooth we spend a lot of time to get it to work. We abandoned it also.

Is this in an (classroom) environment with multiple EV3's or bluetooth devices?

My ev3dev bluetooth connection works great at home, but it wouldn't connect at all when I attended an EV3 workshop with many laptops and EV3's around the room.

mpasternak, 2017-12-20:
Hi.

I came here to comment on this issue. Running IPython was kind of tempting for m.e 

I have run IPython on recent ev3dev-stretch. 

I **DO NOT RECOMMEND** running IPython on the standard EV3 brick as it is simply unusable. As in "way too slow". 

First, to get IPython up and running, you need to compile Tornado and PyZMQ packages. They both say their C-optimalizations are optional, but for machine as slow as EV3 brick every cycle counts. So, I used Docker images to cross-compile those. I had to manually rename some files after installation, as it looks like distutils and/or Cython don't actually support cross-compiling too much. 

Pip3, the command to install Python packages, in Debian Stretch... it looks like for Debian Stretch they put some WHL packages to diet-packages directory and pip3 command wants to import stuff from there. Every single run of "pip3" command takes AGES and I mean it. You wait a minute or two just to get a message that you typed command-line wrong. This needs fixing, perhaps, maybe it would be better to simplify that Python installation and just blatantly extract those files or whatever. 

I was not able to setup and run remote ipykernel with remote_ikernel command, as running ipython this way produced mass flood of journal messages (as in systemd-journal taking up to 50% CPU). But I ran ipython kernel by hand, copy-pasted json file to my MacBook and started typing. 

Every single command has a 1-3 seconds latency. And I mean "command" as in "import os" or "os.getlogin()". Or "sys.platform". 

Currently IPython with/without Jupyter Notebook seems **TOTALLY UNUSABLE** on EV3. 

Hints for future improvement: don't. I'm getting a PiStorms with quad-core RaspberryPi on board. I think that the maximum usable configuration with EV3 you can achieve is either typing Python by hand on the box itself in some minimally invasive editor, using RPyC (latency!) or playing with OpenRoberta. As for now, EV3 seems to be not exactly usable for on-box Python development. OpenRoberta with the simulator and code generation seems very nice solution. IPython? Maybe in the past. I was not able to run it in a comfortable manner. 

Just my $.25