ev3dev/ev3dev #433: Major breaking kernel change - rename port_name to address

dlech, 2015-11-24:
I've just pushed <https://github.com/ev3dev/lego-linux-drivers/commit/37ad261d25ccab7ee24276d16fc83cc38b85fb65> which I plan on including in the next kernel release.

This will break pretty much every ev3dev library out there, so I want to get the word out early so that library maintainers can start preparing for it (or try to change my mind).

pgrudzien12, 2015-11-24:
Will you follow up with changes to https://github.com/ev3dev/ev3dev-lang repo? I'm thinking of spec.json file to generate libs.

dlech, 2015-11-24:
These changes have not been published yet. I imagine the ev3dev-lang maintainers will be updating spec.json soon though

dlech, 2015-12-16:
[I have just released kernel "level 9"](http://www.ev3dev.org/news/2015/12/15/Kernel-Release-Cycle-9/) which includes these change.

As soon as @rhempel fixes up ev3dev-lang-python, I think we will be ready for an image release, so you should fix your libraries soon or update your readme to say that they are broken.

cc: @ev3dev/language-bindings

WasabiFan, 2015-12-16:
I will be taking a look at this shortly to get my stuff in line. Good to know that we are getting this released!

dlech, 2015-12-30:
All of the packages included in the 2015-Q4 image release have been updated, so closing.

closed 2015-12-30 by dlech