ev3dev/ev3dev #435: NXT light sensor docs unclear about range of values

WasabiFan, 2015-11-26:
On the NXT color sensor [docs page](http://www.ev3dev.org/docs/sensors/lego-nxt-light-sensor/), it says that the values are percentages. In other places when it has said this, the given range for `value0` has been 0-1000 and the `decimals` has been `1`. On this page, however, the decimals are also `1` but the  range is given as 0-100. Does this mean that the value is from 0-10 and not a percentage, or should the given range be 0-1000?

dlech, 2015-11-26:
The NXT Light Sensor != the NXT Color Sensor.

It looks like it should say 0-1000 like the other similar values.

https://github.com/ev3dev/lego-linux-drivers/blob/master/sensors/nxt_analog_sensor_defs.c#L195

WasabiFan, 2015-11-26:
> The NXT Light Sensor != the NXT Color Sensor.

Sorry, I meant light sensor :wink: I had at least 20 docs tabs open at the time so it's a wonder that I got the right link! The proper text is just a bonus.

dlech, 2015-12-16:
published

closed 2015-12-16 by dlech