ev3dev/ev3dev #436: Mindstorms Pixy Adapter

ddemidov, 2015-11-27:
The [Pixy Adapter](http://www.ev3dev.org/docs/sensors/mindsensors.com-pixy-adapter-for-mindstorms-ev3-or-nxt/) is recognized automatically, but values 0 to 5 are always zero. The PixyCam is working fine with PixyMan when connected to a desktop computer via USB cable. Also, the led on the cam indicates that it recognizes and tracks an object when connected to ev3 through the adapter.

@dlech, I've seen in the [commit message](https://github.com/ev3dev/lego-linux-drivers/commit/e6c0dbdb0a3c9217448442d220b3f5ee7c28206f) that you don't own the adapter. Is there anything I could help you with debugging this?

rhempel, 2015-11-27:
I have a PixyCam - I was busy with getting the Lua bindings and `event2sensor` programs working. I'll see what I can do this weekend

dlech, 2015-12-01:
Did you try sending commands to the sensor?

ddemidov, 2015-12-01:
I did. After both `echo SORT-COL > command` and `echo SORT-SIZE > command` the values are still zero.

ddemidov, 2015-12-01:
The [manual](http://www.mindsensors.com/index.php?controller=attachment&id_attachment=260) from mindsensors.com says that `Data out port` on the `Interface` tab of PixyMon should be set to 3 (UART). After doing that, the adapter works with ev3dev.

closed 2015-12-01 by ddemidov