ev3dev/ev3dev #437: Table styling issues with the website.

dlech, 2015-12-01:
- [ ] Table needs styling at <http://www.ev3dev.org/docs/sensors/i2c-sensor-addressing/>.
- [ ] The general info table should not span the full page width. Example: <http://www.ev3dev.org/docs/ports/brickpi-in-port/>. Applies ports, sensors and motors.
- [ ] The general info table in motors is not properly styled. Example: <http://www.ev3dev.org/docs/motors/firgelli-l12-ev3-100mm/>
- [ ] Background of columns that span multiple rows looks wrong. Example: <http://www.ev3dev.org/docs/sensors/>. I like having the alternating background colors, but the first column should be different. It looks kind of random since the background color depends on if it starts on an odd or even row.

Note: It would be nice if all tables used the same styling without relying on a liquid template to provide the alternating row backgrounds.

cc: @WasabiFan

WasabiFan, 2015-12-01:
> Note: It would be nice if all tables used the same styling without relying on a liquid template to provide the alternating row backgrounds.

As far as I know, we don't rely on any templates for the table styling -- any table should style correctly as long as it has the `table table-striped table-bordered` classes. If there is any alternation code, I don't think we use it any more.

> The general info table should not span the full page width. Example: 

Honestly, I like it full-width -- or at least I like it much more than the alternative.

As for the others, they should be easy to fix -- I can take a look. There _might_ be a built-in way to style the first column differently, but if not I'll have to cook up a custom solution.

dlech, 2015-12-01:
> As far as I know, we don't rely on any templates for the table styling

You are right, they can't be used anymore since the css for it is gone, but the code is still there. e.g. `<tr class="{% cycle 'info': 'd0', 'd1' %}">`. The `class="{% cycle 'info': 'd0', 'd1' %}` can be deleted. This is in a number of files.

> Honestly, I like it full-width -- or at least I like it much more than the alternative.

I find it hard to read. Someday I may add some actual text content to the general info section in which case it would work to have the table as sort of a side bar. We can leave it for now though. Maybe I will get used to it.

> I can take a look

Go for it. :smile:

WasabiFan, 2015-12-02:
Note to self: The `sensor-connection` include template should be replaced with a YAML data file at some point

WasabiFan, 2015-12-02:
@dlech The I2C Sensor Addressing page uses a markdown table. Should I convert it to HTML and add the styling that way or is there more kramdown magic for this?

dlech, 2015-12-02:
If the table classes don't have to be on the actual table element, you could put a `<div markdown="1" class="table ...">` around the whole thing.

You could also try using a block inline attribute list after the table. `{: class="table ..."}`  I don't know if kramdown considers a table a block or not though.

If neither of those work, I guess we will have to convert to html.

WasabiFan, 2015-12-02:
@dlech I just pushed some updates to a new branch on my fork and (hopefully) my gh-pages branch. I fixed most of the issues, but as far as I can tell there isn't any CSS-only way to alternate colors but skip based on row span. To that end, I modified the table styling and updated the sensor page as a test to make the whole first column white/lighter. It gets rid of the irregularity... as for whether it is _better_, I'll leave that up to you to decide.

The page in question should be [here](http://wasabifan.github.io/ev3dev.github.io/docs/sensors/), although as of now it doesn't look like GH pages has updated yet.

WasabiFan, 2015-12-02:
@dlech I just committed and pushed those stray changes to make the table column light -- you should be able to see it live now.

dlech, 2015-12-02:
I think it looks better that way. Although I wonder if it would be even better if that column was a 3rd color that is half way between the other two row colors instead (i.e. `#2f2f2f`).

closed 2015-12-04 by dlech