ev3dev/ev3dev #438: Uploading Files to the EV3

bcorey, 2015-12-01:
This may be a stupid question, as I don't get many Google results for this issue. I've followed the entire setup guide, but how exactly do I upload a program to the EV3?

aleh-arol, 2015-12-01:
ssh?

bcorey, 2015-12-01:
Yes, SSH over USB Ethernet.

WasabiFan, 2015-12-01:
If you are on Windows, I have found that WinSCP is great for this purpose.

P.S. @dlech do we have tutorials for this (file transfer)? I remember having a discussion about this probably over a year ago but I don't remember what we decided on or what we created.

bcorey, 2015-12-01:
I found a workaround (Copying the file to a new file created in SSH), but it keeps saying that the file is non-executable. The file includes ```#!/usr/bin/python``` at the top.

WasabiFan, 2015-12-01:
Workaround?

The simplest way to copy files over is using an scp client. You can use any method that you'd normally use for accessing a Linux machine. I can't give much detail for platforms other than Windows, so if you are on Mac or Linux someone else will need to help you out.

As for executing the file, you need to add execute permissions to make the file runnable. `chmod 777 file.py`should work.

dlech, 2015-12-01:
You can set permission bits in WinSCP: <http://winscp.net/eng/docs/ui_permissions>.

bcorey, 2015-12-01:
I ran the following program (With all permissions granted) on the EV3 brick, but got an error message that disappeared to quickly to read. Any idea why? ```#!/usr/bin/python
from time import sleep
print("Hello!")
sleep(5)```

WasabiFan, 2015-12-01:
if you are seeing this error while running it from brickman, try from an SSH session instead -- see #418. My guess would be that it's a permissions issue.

bcorey, 2015-12-01:
I'm pretty sure it's not permissions-related (I checked off all of the permissions in WinSCP). I also don't think it was my program, I get the same problem when running [this demo](https://github.com/ddemidov/ev3dev-lang-python/blob/master/demo/motor-control.py).

rhempel, 2015-12-01:
Please open up a separate issue and close this issue if you have managed to upload a file to the EV3.

closed 2015-12-01 by bcorey

WasabiFan, 2015-12-01:
Before opening a new issue, make sure to confirm that it isn't a duplicate -- see my link above.

stefantasticool, 2025-02-02:
Hey there!
I have installed DOOM (https://github.com/Seva167/ev3doom?tab=readme-ov-file) on EV3Dev. I have did exactly as you said, but the DOOM file keep saying that this is not an executable. How do I run it?
Please help!

dlech, 2025-02-02:
If you downloaded a pre-built binary file, you can set the executable bit with `chmod +x ev3doom`.