ev3dev/ev3dev #439: Running Files in Brickman

bcorey, 2015-12-01:
Whenever I try to run a .py file in brickman, the screen goes blank except for a cursor in the top left corner then after about a second an error message shows up for a fraction of second after which the screen goes back to the brickman file browser. [This](https://github.com/ddemidov/ev3dev-lang-python/blob/master/demo/motor-control.py) is the program I am trying to run. I don't think it's permissions-related because I checked all of the permissions off when I uploaded the file via WinSCP.

dlech, 2015-12-01:
Which version of `python-ev3dev` did you install? Also, <https://github.com/ddemidov/ev3dev-lang-python> has been deprecated and replaced by <https://github.com/rhempel/ev3dev-lang-python>.

bcorey, 2015-12-01:
Was unaware I had to install anything. I guess this would explain the error message.
I'm not able to find the python-ev3dev download, could you post a link?

rhempel, 2015-12-01:
I would try to do the simplest possible thing first - just print "Hello World" and wait 5 or 10 seconds and report back to us if you get that working.

Then move on to ev3dev-python - and yes, you'll need the new ev3dev-python package that @dlech pointed you to. Again, do the simplest possible thing, like change an LED from red to green and then turn it off - then let's get more complicated by adding motors...

dlech, 2015-12-01:
@bcorey, `sudo apt-get install python-ev3dev` should work.

@rhempel, your readme could be a bit more informative. :wink:

WasabiFan, 2015-12-01:
@rhempel Maybe the readme should be more than a one-line poetic epithet? :wink:

bcorey, 2015-12-01:
Okay, here's what I got when I tried installing ```python-ev3dev``` through putty.
```
root@ev3dev:~# sudo apt-get install python-ev3dev
Reading package lists... Done
Building dependency tree... Done
E: Unable to locate package python-ev3dev
```

G33kDude, 2015-12-02:
It sounds like your brick might not have a solid internet connection. If you're using a USB connection to the brick, you'll have to set it up as per this guide http://www.ev3dev.org/docs/tutorials/connecting-to-the-internet-via-usb

WasabiFan, 2015-12-02:
Actually, that shouldn't use internet... I suspect it's just a matter of `sudo apt-get update` before trying again. The current image release was released before the new python binding from what I remember.

G33kDude, 2015-12-02:
Ah, you're right. A network connection problem would show up a few steps later in the apt-get. He will however still need an internet connection to `apt-get update` and `apt-get install` the package, right?

bcorey, 2015-12-02:
@G33kDude: I've definitely got a solid connection, as I can ping the EV3 from a command prompt (Also, I've already done the USB over Ethernet tutorial a few times).

WasabiFan, 2015-12-02:
> He will however still need an internet connection to  apt-get update  and  apt-get install  the package, right?

Yes, the update should require an internet connection.

@bcorey Try running `sudo apt-get update` and try again.

bcorey, 2015-12-02:
Got this after ```sudo apt-get update```. Still can't find the python-ev3dev package.
```
root@ev3dev:~# sudo apt-get update
Err http://ftp.debian.org jessie InRelease

Err http://ev3dev.org jessie InRelease

Err http://ftp.debian.org jessie Release.gpg
  Temporary failure resolving 'ftp.debian.org'
Err http://ev3dev.org jessie Release.gpg
  Temporary failure resolving 'ev3dev.org'
Reading package lists... Done
W: Failed to fetch http://ftp.debian.org/debian/dists/jessie/InRelease

W: Failed to fetch http://ev3dev.org/debian/dists/jessie/InRelease

W: Failed to fetch http://ftp.debian.org/debian/dists/jessie/Release.gpg  Temporary failure resolving 'ftp.debian.org'

W: Failed to fetch http://ev3dev.org/debian/dists/jessie/Release.gpg  Temporary failure resolving 'ev3dev.org'

W: Some index files failed to download. They have been ignored, or old ones used instead.
```

G33kDude, 2015-12-02:
Just a side note here: Being able to ping the EV3 from your computer doesn't necessarily imply the brick is connected to the internet, just that it's connected to your computer. To check that it can see the internet as well as your computer you'd need to ping something on the internet from the brick. For example, `ping google.com` (using a domain instead of an IP also tests that DNS is working properly).

bcorey, 2015-12-02:
@G33kDude Oh, okay.

rhempel, 2015-12-02:
Yes, the readme should be better - and eventually the pure Python binding will be part of the standard image so users won't need to install it.

bcorey, 2015-12-02:
So I think the problem now is that putty can't find the python packages. Should I open a new issue for that?

cavenel, 2015-12-02:
These aren't really issues (nothing is wrong with ev3dev), they are more like questions. Maybe it would be better to ask for help on irc? http://www.ev3dev.org/support/#irc

dlech, 2015-12-02:
As we have already said, your EV3 is not connected to the Internet. It sounds like you have not configured Internet Connection Sharing in Windows. Sometimes Internet Connection Sharing in Windows gets in a bad state and you have to turn it off and back on.

bcorey, 2015-12-03:
Okay, I've done the "Connecting to the Internet with USB" tutorial again. Something I've noticed: the status on the brick shows as "Connected", not "Online". Also, when I double-click on the EV3 inside "Network Connections", "Ipv4 connectivity" is listed as

rhempel, 2015-12-03:
Here's a quick test - ssh onto the brick. Now
```
ping ev3dev.org
```
If you get no reply, you are not connected to the Internet and you will have to go through setting up Internet connection sharing on your Windows machine again.

bcorey, 2015-12-03:
Here's what I got:
```
root@ev3dev:~# ping ev3dev.org
ping: unknown host ev3dev.org
```
So I don't think there's any internet connection, but I can't see how to do anything differently from the last few times.

rhempel, 2015-12-03:
Welcome to learning about Linux! Take a step back and consider that quite a few people have got this working - so let's assume that there's something about your setup or procedure that's odd. Are you using Windows 7/8 or 10 - check the Windows tab here [http://www.ev3dev.org/docs/tutorials/connecting-to-the-internet-via-usb/] for instructions and follow them all.

Obviously, you can `ssh` onto the brick via USB and you are logged in as `root`.

Can you ping `ev3dev.org` from the Windows command line?

rhempel, 2015-12-03:
And maybe close this issue and get help on the IRC channel - this has nothing to do with running files under Brickman

WasabiFan, 2015-12-03:
I have found that using networking over USB is much more trouble than it's worth. If you have a Wi-Fi dongle, use that and you should be fine.

But either way, having a discussion about networking in an issue about brickman file execution isn't going to make it useful to others, and this is most likely a problem specific to your setup. So you should participate in an extended discussion over IRC (you can access it from the "help" page on our website) instead.

closed 2015-12-03 by bcorey