ev3dev/ev3dev #440: Does EV3Dev any LCP support?

jabrena, 2015-12-02:
Hi mates,

Currently i am playing with motors and some sensors, but I don't know if EV3Dev has some support for LCP.

a solution for node from a laptop.
https://github.com/wenchaojiang/ev3-Nodejs-bluetooth-Api

Cheers

Juan Antonio

dlech, 2015-12-02:
By LCP, do you mean Link Control Protocol? If so, no we don't have anything like that yet.

jabrena, 2015-12-02:
No, I said the implementation about LEGO Communication protocol.
http://www.tau.ac.il/~stoledo/lego/btperformance.html

Some hours ago, I was thinking about this issue and I remember that exist another issue about Lego firmware, so I supposed that we don't have that support yet.

Oki. :D

dlech, 2015-12-02:
OK, you want issue #329 then.

closed 2015-12-02 by jabrena