ev3dev/ev3dev #441: Interfacing XV11 LIDAR to EV3/ev3dev

bmegli, 2015-12-02:
First, I am sorry for abandoning the project for a longer time. It seemed that I was the only person using R bindings so there was no reason in putting that much effort in them.

Anyway...

I have just finished interfacing XV11 LIDAR to EV3/ev3dev, here is 5 min. tutorial:

http://www.youtube.com/watch?v=G6uVg34VzHw

XV Lidars (those ripped off Neato) are probably the cheapest reasonable LIDARs around. You can get one at around 70$ in US, somewhat worse in Europe (duty and shipping fees).

EV3 has all the hardware needed to get the LIDAR working. Ev3dev has all the software. It's only a meter of soldering the connectors.

So for around 70$ + sacrificing EV3 cable + sacrificing USB cable + some work we get fully working LIDAR. Not perfect but it is the best sensor for EV3 I have seen.

Now - we could write a driver but I am not sure we would want one.

Reasons:
1. It's a part of commercial robot.
2. If one is not counting the money there is USB only XV Lidar Controller v1.2 at GetSurreal.
3. There is already a driver for ROS. From what I see in the code it returns full 360 degree scans. It is okay for a robot that scans stationary but for XV11 LIDAR 360 degree scan takes around 200 ms. This needs correction for movement and rotation for moving robot.
4. It seems difficult to write a driver reasonably. When using LIDAR you usually want to synchronize its readings with other sensors and odometry (in time). LIDAR spits around 10 kB/s in packets throgh UART. For the robot in tutorial, while getting data from the LIDAR I also get data from gyro and odometry. Then I interpolate the odometry and gyro data for LIDAR data. In a driver we could probably timestamp the readings (?) with more granularity then once per 360 degrees.
5. The way I have connected the LIDAR is the cheapest but it takes up 1 motor port (LIDAR motor), 1 sensor port (UART data), USB port (Laser power) so most likely one needs USB hub additionaly (hub would be also needed for other methods of connnecting LIDAR).

To consider.

There are other things that come up to me during this project.

I ended up disabling my driver for CruizCore gyroscope and reading from I2C directly to get tighter timings. CruizCore works at 100 Hz (once per 10 ms). Polling it through I2C added a lot of delay to the readings (and polling more often seems to be stressing CPU). 

Idea to consider: when we disable I2C polling (polling_ms to 0) when the user wants device reading we could read directly from I2C. This would put less stress on CPU and give better timings. I could implement it for the CruizCore for the start. Or we can leave it as is for advanced scenarios. They can disable polling and use the direct attribute anyway.

Just to consider.

Kind regards

dlech, 2015-12-02:
Nice to hear from you again. :smiley: 

This video is absolutely excellent. I do have a couple of suggestions though. 1) With regard to the "?" about the motor connection, you can manually put the output ports in DC motor mode (might be called rcx-motor) and then you don't need the resistors. 2) When there is text at the bottom of the screen, you can't read it when the video is paused because the youtube controls are overlayed on top of it.

I think this would be a good thing to add to the [tutorials page](http://www.ev3dev.org/docs/tutorials/). Would you like to do this?

I'm with you on not writing a kernel driver for this. I have found that writing kernel drivers for UART to be a pain because you have to use what is called a "line discipline". This isn't actually so bad, but the catch is that there are a fixed number of IDs for line disciplines and there are about 5 free ones (we are already using 2 in ev3dev).

> Polling it through I2C added a lot of delay to the readings (and polling more often seems to be stressing CPU).

I'm not sure exactly what you mean here. Could you please explain a bit more?

> hen we disable I2C polling (polling_ms to 0) when the user wants device reading we could read directly from I2C. This would put less stress on CPU and give better timings. I could implement it for the CruizCore for the start.

Again, I'm not exactly what you mean here. Can you give an example of how this would work?

bmegli, 2015-12-02:
Hi dlech!

Thanks for the comments.

1. Motor - yes, I thought so but wasn't sure. Thanks. Anyway, I like my non-stock motors to be plug and play. I do the same with diodes, etc. That way they work with all the platforms (be it ev3dev or original LabView or EV3 firmware on brick).

2. Yes. Now that you have pointed it out ;-) It seems that I just haven't been stopping the tutorial.

3. Tutorial - ok, maybe. I would say - if there is demand. The video is ok to get one started and hopefully inspire.

4. Driver - I know. I have read your kernel driver code for sensors following Lego line discipline. Also connecting through motor port/sensor port/usb port seems to be too "ad hoc" for a driver ;-) And it would need to work on tty/motor port pair (so that it works on RPi through gpio, EV3 through sensor port etc). Not a good direction.

5. I2C - hmm.

The short version: CruizCore works at 100 Hz (10 ms delay).
Polling it at 50 ms gives readings at worst 60 ms off. 
LIDAR (5 HZ) takes around 200 ms for 360 degree scan in 90 packets, so data arrives at about 2.22 ms. 

Reading it directly was not an option. So I ended up configuring UART to spit 15 packets at once (60 degree), about 30 ms each and synchronizing it with gyroscope and odometry. 60 ms out of time gyroscope readings were not an option. This would mean error of around 7 degree (while rotating) for the robot.

I have tried going down with polling time of I2C sensor. Just experimenting. Below 20 ms the CPU load seemed to be increased (just an impression, I would have to test that). I am not even sure one can go as low as 20 ms with current config hz. Setting it below 20 ms gave no visible difference.

I ended up disabling polling and reading from the direct attribute. I have written the driver for CruizCore so using direct was an easy option. That way I got the best timings (around 10 ms) and less CPU stress (impression, I would have to test that).

Ugh. Overly long. 

Why 30 ms - for LIDAR readings. I am mapping while moving, correcting LIDAR readings for robot rotation, movement, estimating robot position/path and sending through UDP to PC for further processing. This is just the lowest layer.

6. Not sure about this one. Such behaviour of I2C drivers could be automatic. When polling is disabled read on demand. Probably too uncommon scenario.  How this would work - in theory - reading I2C sensor data with polling disabled would trigger I2C reading from the correct register and I mean here classic blocking read. Some issues here like error reporting etc. This is just an idea. Probably not the best one ;-)

dlech, 2015-12-02:
> When polling is disabled read on demand

I think this should already work. Each time you set the mode, it should read from the sensor and update the `valueX` attributes.

dlech, 2015-12-02:
> Tutorial - ok, maybe. I would say - if there is demand.

I was thinking the tutorial would just be a link to the video. This would just be a way for people to find your video.

dlech, 2015-12-02:
> I am not even sure one can go as low as 20 ms with current config hz. Setting it below 20 ms gave no visible difference.

This would be interesting to plot out as in #324 to see what is going on. We found that changing config hz made *everything* really slow, so that is not really an option. It is possible though that the current driver is affected by config hz because of using the kernel work queue.

bmegli, 2015-12-02:
Oh, ok.

Tutorial - yes.
Polling 1 - rigggght, I didn't remember this else in nxt_i2c_sensor_set_mode. Silly me! Thanks. 
Polling 2 - from what I remember setting polling time below 20 ms made the short term average load (top command) go above 1. Not entirely sure. I will investigate soon.

jabrena, 2015-12-02:
Hi @bmegli, I have a unit at home. I don't have many time at the moment, but I will experiment with the LIDAR soon. With LeJOS, I used the another device in the market RPLidar (https://www.youtube.com/watch?v=5j82Y4aRbuA), but I had to use an Arduino one unit to run the logic. I will use your notes. 

Nice progress with Mindstorms!!!

Doubt: What tool did you use for mapping?

Cheers

bmegli, 2015-12-02:
Hi @jabrena. 

Great to hear.

Time... - yes, same pain everywhere.

RPLidar - I haven't heard about that one, thanks!

Mapping Tool - own quick and dirty code. What you see in the movie is *just* odometry + gyroscope + LIDAR. Just the lowest layer, more to be build atop.

For short:
-robot scans with laser, gyroscope, odometry
-sends this data through UDP
-quick and dirty app (.NET, WPF) receives on UDP, interpolates odometry & gyroscope for laser data, corrects for movement/rotation, estimates path/position, adds readings to map. Map is just a big bitmap updated through backbuffer in realtime ( ~30 FPS).

Nothing more now.

bmegli, 2015-12-02:
UGH - RPLidar seems great but costs one 0 too much ;-)

bmegli, 2015-12-03:
Ok  - tutorial page for XV11 pull request initiated

bmegli, 2015-12-03:
As for the I2C poll_ms (experiments for CruizCore) and the load:

Polling frequency/Load
100 ms - below 0.20
15 ms - somewhat above 1 
20 ms - somewhere around 0.6

System was up for about an hour in each case. 
I am not quite sure if it should be like that. 

Anyway, seeing it in top was enough for me to seek alternative aproach to polling for reading from gyroscope.

[image]

jabrena, 2015-12-03:
Did you test with Arduino to help EV3 CPU?
Arduino X process the sensor and returns to EV3 the values processed. (It is an idea :) )

bmegli, 2015-12-03:
@jabrena 

Thanks for ideas. We don't have a problem with reading from the LIDAR. 

The load gets high when polling I2C bus sensor (gyroscope) with high polling rate.

From my perspective it's not a problem. I can just read data from I2C when I need it instead of polling.

If you look closely at top output you can see that the CPU is not stressed (88,4 % idle in the worst case).

The load above 1 for polling at 15 ms means only that there was more than 1 item waiting in the workqueue. It wasn't there due to lack of CPU resources (88,4 % idle).

Wild guess - the kernel worker thread wakes up, does its work quickly, goes asleep. By the time it wakes up next time new work is already waiting. So maybe the "granularity" of waking up.

I am definitely not an expert on that. The above may be nonsense.

I haven't seen high loads when reading i2c from userspace (just an impression) but I was reading at 30 ms. I would have to code an experiment.

@dlech will probably know what's going on anyway.

bmegli, 2015-12-03:
Alternative ideas:
- tasks in the workqueue wait for I2C IO? The wa=0 contradict this. I2C read should be rather quick.

ugh

dlech, 2015-12-03:
I'm moving the discussion on polling to #445 so that it doesn't get lost.

bmegli, 2015-12-03:
Ok, everything clear here then!

closed 2015-12-03 by bmegli

bmegli, 2015-12-06:
@jabrena 

"Doubt: What tool did you use for mapping?"

Ok. I believe I didn't understand the question. I kind of answered it but since I did not understand the question you might have overlooked the answer ;-)

Now - knowing what you do put me in the right context.

If you are asking for a tool like feature extraction (e.g. RANSAC), particlule filters, grid maps, Kalman Filters and EKF, Fast SLAM then the answer is still the same.

None.

I was working on the lowest layer now which is gyroscope enchanced odometry and corrections for movement & rotations - laser readings of moving robot.

What you see in the video are the readings cast on a bitmap in realtime. The accuracy is just good so it seems something more.

If I can hint you - always visualize what the robot sees/knows, best in realtime. Layer by layer. And only move to the upper layer when you are sure that previous is 100% ok.

Visualisation helped me catch errors in my code like interpolating between gyroscope readings from -175 to 175. This obviously should go only through 10 degrees but was going through 350 in the code and so on. Visualize.

jabrena, 2015-12-06:
Interesting comment.

I think that if you connect with ROS RViz, you could add SLAM features. I send you some links about it:
http://www.hessmer.org/blog/2011/03/13/ardros-dead-reckoning/
http://wiki.ros.org/turtlebot_navigation/Tutorials/Build%20a%20map%20with%20SLAM
If you decide to explore this way, I could help you in this process. I think that ROS could offer nice features.

What programming language are you using to get odometry from regulated motors? I read that you use gyro sensor.

``` bash
I was working on the lowest layer now which is gyroscope enchanced odometry and corrections for movement & rotations - laser readings of moving robot.
```

Do you have the example in some repo?

bmegli, 2015-12-07:
#jabrena

Thanks for interesting links, especially the second one.

> If you decide to explore this way, I could help you in this process. I think that ROS could offer nice features

Thank you, maybe much later on. This is mostly "for fun" project so I am going to write the SLAM  part myself.

> What programming language are you using to get odometry from regulated motors? I read that you use gyro sensor.

C (in spirit) with ev3dev C++ bindings to get motor encoder positions, gyro readings, laser data then forwared through UDP to PC. PC side processed in C#, .Net, WPF for frontend.

But the great side of ev3dev is that you can use any language you like.  There are numerous bindings.
The design of driver system is truly great.

If drivers for your faviourite language don't exist it's a matter of writing 3 short functions (!):
- connect (the hardest, that finds the device in file system but also a few liner)
- read_attriubute
- write_attribute

Really. This is enough to get thin bindings system for any language. Without fancy stuff. I use it with R sometimes to calibrate the sensors/motors etc. I stopped developing R bindings and use 3 simple functions for everything. 

> Do you have the example in some repo?

I have just setup one and unlicensed the code:

https://github.com/bmegli/ev3dev-mapping.git

But be aware that I never meant to use this code 'in production' or make it public. 
It's not a good code. I had a small battle with myself whether I should make it available in such state.
EV3 side code is quite ok, PC side is not.

> I was reading that CruizCore sensor is very accuracy.

Yes. Incredibly. From specs and empirically.
But be aware of limitations. It works internally at 100 Hz (10 ms). So the readings can be 10 ms off for the start. Assuming its readings are perfect this still leaves around 1 degree possible error for quickly rotating robot (I calculated this for mine at some point).  

If you are not carefull you will add more delay from polling scheme so I prefer to read from gyro exactly when I need it. Or even (!) 5 ms later so that you center the error.

In the repository code I use really hard way to do it (using the direct attriubute read from I2C bus). From what @dlech said you can have the same effect by disabling polling, changing the mode (to the same) and reading the value from sensor.

I am 100% sure this will work for CruizCore because I have written the driver (ok, it felt more like filling the template @dlech written, the easiest driver I have ever had).

CruizCore doesn't really have modes. I only made kind of virtual modes so that we can document the units and readings it gives and make its use easy. Internally it does nothing when switching the mode.

Some other sensors do internall stuff when switching the mode and are not ready right away so the trick with mode switching may not work for everythihng.

@dlech Will the trick - disable polling, change mode, read value - work with every I2C sensor driver you have written? Do some of them make some time consuming work when switching the mode (to the same mode)?

> Are you using that sensor to get inertial navigation data?

No. Just the angle. Gyroscope enchanded odometry is kind of 'integrating the path'.
Dispacement from encoders along the angle from gyroscope. Easy and straightforward. Integrate as often as possible.

Good overview:

http://www.robotnav.com/position-estimation/

But obviously Lauro got the line that is interesting us wrong.

Instead of:

    POS_X = POS_X * cos(HEADING)
    POS_Y = POS_Y * sin(HEADING)

He should have:

    POS_X = POS_X  + displacment* cos(HEADING)
    POS_Y = POS_Y + dispacement* sin(HEADING)

bmegli, 2015-12-07:
@dlech Will the trick - disable polling, change mode, read value - work with every I2C sensor driver you have written? Do some of them make some time consuming work when switching the mode (to the same mode)?

rhempel, 2015-12-07:
@bmegli - thanks for supporting my original design decision to set the goal of allowing every device on the EV3 to be controlled by simple file I/O. This was (in my mind) the prime goal to allow as many scripting (and compiled) languages to run on the EV3 and therefore be available for experimentation by programmers.
Of course, the hard work of @dlech and many others that helped bring my clumsy first efforts to the polished state that we have today is the other half of the ev3dev success story - without their work ev3dev would not be as awesome as it is now.

bmegli, 2015-12-08:
Huh, I remember this right after EV3 was released, way back in 2013, before ev3dev existed (ok, maybe it was ev3dev but not by this name):

http://hempeldesigngroup.com/embedded/2013/09/25/making-hacking-accessible-lego-mindstorms-ev3/

I always kept my fingers crossed for you to succeded and it clearly helped ;-)