ev3dev/ev3dev #442: How to detect a RC Motor (8287-1)?

jabrena, 2015-12-02:
Hi,

I was tryting to detect a RC Motor as dc-motor
http://www.peeron.com/inv/sets/8287-1

but when y execute the following statement:

``` bash
root@ev3dev:/home# ls /sys/class/dc-motor/
```

I don't receive any response but if I want to detect my 2 tacho motors, I don't have any problem:

``` bash
root@ev3dev:/home# ls /sys/class/tacho-motor/
motor0  motor1
```

I connected the rc motors with EV3 Brick with this kind of cable:
http://www.ebay.es/itm/LEGO-Conversion-Cable-Mindstorms-NXT-RCX-Technic-Education-9797-USED-/321389175953

What is the problem?

Many thanks in advance.

Juan Antonio

dlech, 2015-12-02:
These motors cannot be automatically detected (because nothing is attached to pins 5 or 6). You have to tell the port to use the `dc-motor` mode. There will be a port in `/sys/class/lego-port/` that has the same port name as the port you are using. Write `dc-motor` to the `mode` attribute to change the mode, e.g. `echo dc-motor > /sys/class/lego-port/port4/mode`.

You can also change the port mode using the brickman user interface, but I assume you are wanting to do this programatically.

jabrena, 2015-12-02:
Many thanks @dlech, I go with the experiment! :D

jabrena, 2015-12-02:
What motors does support for /sys/class/dc-motor?

dlech, 2015-12-02:
Any type of DC motor. Basically, it is just a PWM output that you can control using duty cycle.

jabrena, 2015-12-02:
When I list this path:

``` bash
root@ev3dev:/home# ls /sys/class/lego-port
port0  port1  port2  port3  port4  port5  port6  port7
```

I see several ports. How to know the right port to write in the example that you say?

cheers

dlech, 2015-12-02:
same as sensors, look at the `port_name` attribute.

jabrena, 2015-12-05:
Hi @dlech 

I have the DC Motor connected in port C but I receive a rare output when I set the mode:

```
root@ev3dev:~# echo dc-motor > /sys/class/lego-port/port6/mode
-bash: echo: write error: Invalid argument
root@ev3dev:~# cat /sys/class/lego-port/port6/port_name 
outC
root@ev3dev:~# 
```

What is the problem?

G33kDude, 2015-12-05:
@jabrena Try quoting the parameter to echo, like this:
```
root@ev3dev:~# echo 'dc-motor' > /sys/class/lego-port/port6/mode'
```

ddemidov, 2015-12-05:
I think it should be 
```
root@ev3dev:~# echo dc-motor > /sys/class/lego-port/port6/set_device
```

jabrena, 2015-12-05:
I tried but the test failed: 

``` bash
root@ev3dev:~# echo dc-motor > /sys/class/lego-port/port6/set_device
-bash: echo: write error: Operation not supported
root@ev3dev:~# ls /sys/class/lego-port/port6
device       mode   port_name  set_device  subsystem
driver_name  modes  power      status      uevent
root@ev3dev:~# cat /sys/class/lego-port/port6/driver_name 
legoev3-output-port
root@ev3dev:~# cat /sys/class/lego-port/port6/port_name 
outC
```

jabrena, 2015-12-05:
hi @G33kDude,

I tryed your way but it failed too:

``` bash
root@ev3dev:~# echo 'dc-motor' > /sys/class/lego-port/port6/mode
-bash: echo: write error: Invalid argument
```

dlech, 2015-12-06:
It depends on which kernel you have. The name of the mode changed recently. Run `cat modes` to see which modes are actually available. If you have an older kernel, it might be `rcx-motor` instead of `dc-motor`.

jabrena, 2015-12-06:
Hi @dlech yes, you have reason, when I execute that statement I receive this result:

``` bash
root@ev3dev:/home# cat /sys/class/lego-port/port6/modes
auto legoev3-motor rcx-motor rcx-led raw
root@ev3dev:/home# 
```

How to get current EV3Dev version by command-line?

rhempel, 2015-12-06:
`uname -a`

jabrena, 2015-12-06:
Currently, I am using this version:

``` bash
root@ev3dev:~# uname -a
Linux ev3dev 3.16.7-ckt16-7-ev3dev-ev3 #1 PREEMPT Sun Sep 6 22:59:18 CDT 2015 armv5tejl GNU/Linux
```

dlech, 2015-12-06:
> How to get current EV3Dev version by command-line?

    sudo apt-get update
    sudo apt-get upgrade
    sudo apt-get dist-upgrade

jabrena, 2015-12-06:
Hi @dlech, 

hehehe, you did again! :)

I could manage the motor without any problem.
I did a mini example in shell script:
https://github.com/jabrena/ev3dev-lang-java/blob/master/examples/bash/motors/exampleDCMotor.sh

``` bash
root@ev3dev:/home# uname -a
Linux ev3dev 3.16.7-ckt19-8-ev3dev-ev3 #1 PREEMPT Mon Nov 9 11:03:41 CST 2015 armv5tejl GNU/Linux
root@ev3dev:/home# cat /sys/class/lego-port/port6/modes
auto tacho-motor dc-motor led raw
root@ev3dev:/home# echo dc-motor > /sys/class/lego-port/port6/mode
root@ev3dev:/home# ls /sys/class/dc-motor/
motor0
root@ev3dev:/home# ls /sys/class/dc-motor/motor0
command   driver_name    polarity   ramp_down_sp  stop_command   time_sp
commands  duty_cycle     port_name  ramp_up_sp    stop_commands  uevent
device    duty_cycle_sp  power      state         subsystem
````
Many thanks.

I continue with the development, tomorrow I will release v0.2.0 of ev3dev-lang-java with the following support:

* Regulated Motor Support
* Unregulated Motor Support
* Sensor Support (Few sensors)
* OpenCV (https://github.com/jabrena/ev3dev-lang-java/issues/33)
* Support for Java profiling tools as VisualVM & JConsole (https://github.com/jabrena/ev3dev-lang-java/issues/38)
* LeJOS filter Support

Cheers

closed 2015-12-06 by jabrena