ev3dev/ev3dev #443: Any way to speed up writing to devices?

aleh-arol, 2015-12-03:
Hi, I looked through all the issues related to the polling speed and looks like I need a clarification.
Typically I'm looking for a way to achieve higher sampling freqs and looks like my current bottle neck is writing to devices.

I'm making a line follower using python ev3 (this one - https://github.com/topikachu/python-ev3) and I add file handles cache there (just to see what's the improvement - and it's quite good) and I have ~30-40ms per loop iteration (I'm trying to sleep at least 10 ms to let other stuff work - as on top of this line follower there will be other logic including network communication). I checked for FP performance - not an issue (as per profiling), my loop without actual writing to devices  takes somewhere 2-3 ms.
Just for a note I'm doing excessive writes as I'm using 1 color sensor in 2 modes (i will optimize this probably in future) during a single loop iteration - total: writing to 2 motors, twice write the mode of the color sensor.

Finally I did profiling and found out that most of the time spent in file.write. I guess python is not an issue here and finally the questions are:
1) am I missing something in my suggestion regarding sysfs write is "expensive" operation (that's the driver code is working there)?
2) otherwise is there any way to speed it up?
3) there's no other interface from user space except sysfs, right?

rhempel, 2015-12-03:
We have a new pure Python library that has file caching and some other neat features like screen manipulation. 
The source is here: [https://github.com/rhempel/ev3dev-lang-python]
You can also install it as a package on your EV3 (assuming you have it connected to the Internet) using:
```
sudo apt-get-update
sudo apt-get install python-ev3dev
```

aleh-arol, 2015-12-03:
Thanks, I know. I've just started with the one I mentioned in the first post and don't like to rewrite (no much work but anyway) my program while it works. For the file caching - I'm using it - it does a good job.

I checked out driver sources and googled a bit on sysfs - so there's nothing (major) between driver store call and my python program write call. So I guess I can't do anything except to fix my own stuff.

Just want to be sure - as far as I've from other issues/posts - 100Hz sampling is a hard limit (sensors) and 20-50Hz is quite normal sampling rate for lego robot.

rhempel, 2015-12-03:
Yes, the relatively low sampling speed is good enough in most cases. Remember, the goal is to allow many different language bindings, and the easiest way to do that is through a file interface instead of a shared memory interface.
You can program robots in `bash` to get started :-)

aleh-arol, 2015-12-03:
Thanks, I'm closing the issue now.

closed 2015-12-03 by aleh-arol