ev3dev/ev3dev #444: kmod_search_moddep error when setting up ev3-ev3dev-jessie on Trusty

davidhaller, 2015-12-03:
Hello everyone,

I installed Ubuntu 14.04.3 Server on VirtualBox 5.0.10 and followed this guide: https://github.com/ev3dev/ev3dev/wiki/Using-brickstrap-to-cross-compile-and-debug

The installation fails after extracting the packages with this error:

>13:56:17 [ info] Copying root files from BOARDDIR definition...
>13:56:17 [ info] Configuring packages...
>13:56:17 [ info] preseed debconf
>13:56:17 [ info] running preinst scripts...
>13:56:18 [ info] running /alsa-utils.preinst
>13:56:18 [ info] running /apt.preinst
>13:56:18 [ info] running /base-passwd.preinst
>13:56:18 [ info] running /bash.preinst
>13:56:18 [ info] running /bluez.preinst
>13:56:18 [ info] running /dash.preinst
>Adding 'diversion of /bin/sh to /bin/sh.distrib by dash'
>Adding 'diversion of /usr/share/man/man1/sh.1.gz to /usr/share/man/man1/sh.distrib.1.gz by dash'
>13:56:19 [ info] running /debconf.preinst
>13:56:19 [ info] running /dh-python.preinst
>13:56:19 [ info] running /dpkg.preinst
>13:56:19 [ info] running /e2fsprogs.preinst
>13:56:19 [ info] running /ev3-systemd.preinst
>modprobe: ERROR: ../libkmod/libkmod.c:557 kmod_search_moddep() could not open moddep file '/lib/modules/3.19.0-37-generic/modules.dep.bin'

I already did sudo depmod -a, the modules.dep.bin files exists and is readable.

dlech, 2015-12-03:
See https://github.com/ev3dev/brickstrap/issues/9

davidhaller, 2015-12-04:
Ah, I'm one the wrong issue tracker. Thanks man, that worked.

closed 2015-12-04 by davidhaller