ev3dev/ev3dev #445: I2C sensor polling issues

dlech, 2015-12-03:
In https://github.com/ev3dev/ev3dev/issues/441#issuecomment-161562639 @bmegli said...

As for the I2C poll_ms (experiments for CruizCore) and the load:

Polling frequency/Load
100 ms - below 0.20
15 ms - somewhat above 1 
20 ms - somewhere around 0.6

System was up for about an hour in each case. 
I am not quite sure if it should be like that. 

Anyway, seeing it in top was enough for me to seek alternative aproach to polling for reading from gyroscope.

[image]

dlech, 2015-12-03:
More from @bmgeli...

The load gets high when polling I2C bus sensor (gyroscope) with high polling rate.

From my perspective it's not a problem. I can just read data from I2C when I need it instead of polling.

If you look closely at top output you can see that the CPU is not stressed (88,4 % idle in the worst case).

The load above 1 for polling at 15 ms means only that there was more than 1 item waiting in the workqueue. It wasn't there due to lack of CPU resources (88,4 % idle).

Wild guess - the kernel worker thread wakes up, does its work quickly, goes asleep. By the time it wakes up next time new work is already waiting. So maybe the "granularity" of waking up.

I am definitely not an expert on that. The above may be nonsense.

I haven't seen high loads when reading i2c from userspace (just an impression) but I was reading at 30 ms. I would have to code an experiment.

@dlech will probably know what's going on anyway.

dlech, 2015-12-03:
Your wild guess is pretty much correct, I think. I'll just add a few details. The I2C sensor driver uses a hrtimer (software interrupt) for the polling timer. Since the I2C read sleeps, the interrupt has to schedule work in the workqueue instead of starting the I2C message directly. Since we have HZ=100, the kernel only wakes up every 10ms to process the workqueue.

When initiating a message using the `direct` attribute, we are not in an interrupt context, so we can initiate the I2C message directly without queuing it, which is why it appears to work faster. The same effect can be achieved by setting the `mode`.

bmegli, 2015-12-03:
I would second you on that but aren't hrtimers supposed to work with resolution below that resulting from CONFIG HZ? At least with the correct patch set and hardware support. 

Off topic - I have just stumbled across that:

"A Platform for LEGO Mindstorms EV3 Based on an RTOS with MMU Support",  last year.

https://www.mpi-sws.org/~bbb/proceedings/ospert14-proceedings.pdf

dlech, 2015-12-03:
Yes, hrtimers work like that, but in the driver, the hrtimer is just scheduling work in a workqueue, which doesn't get handled until later. If the htimer interrupt happens more than once before the work in the workqueue is handled, then we are missing a poll sometimes.

> Off topic

That might be a good link to add to #324.

bmegli, 2015-12-07:
@dlech Will the trick - disable polling, change mode, read value - work with every I2C sensor driver you have written? Do some of them make some time consuming work when switching the mode (to the same mode)?

dlech, 2015-12-07:
Yes and yes. Most don't have extra work when setting the mode, but some do. You can tell which ones do by checking the source code. They will have `set_mode_reg` and/or `set_mode_{pre,post}_cb` set.

jdchristensen, 2016-02-15:
I'm using the trick of setting the mode on my Absolute IMU gyro to get samples at 32Hz instead of the 20Hz that I would get before with poll_ms set to 50.  I'm wondering if people expect that I would be able to get even better than 32Hz using the direct attribute, or by unbinding ev3dev and using a separate i2c library.  If so, can anyone point me to some sample code and/or instructions?  I'm completely new to i2c.  The sensor's attributes currently show up in a directory like

/sys/devices/platform/legoev3-ports/lego-port/port0/i2c-legoev3.3/i2c-3/3-0011/lego-sensor/sensorXX/

where XX varies.

Thanks!

dlech, 2016-02-15:
Why do you need to read the sensor so quickly? What are you trying to achieve?

There is some info about using the direct mode and unbinding the sensor at http://www.ev3dev.org/docs/sensors/using-i2c-sensors/

jdchristensen, 2016-02-15:
I'm trying to get a balancing robot to balance, and noticed that it did better when I increased the sample rate, so I thought I'd try increasing it even further.

Thanks, that link looks very helpful.  My initial test from a shell script doesn't look good, though.  The following takes 35ms per iteration, which roughly matches the 32Hz I've already achieved:

``time for i in `seq 10`; do echo ALL > mode ; done``

But reading the direct attribute is much slower, taking 126ms per iteration:

``time for i in `seq 10`; do hd -s $((0x53)) -n 1 direct > /dev/null ; done``

(These are both with poll_ms set to 0.)
Any idea why this happens?  Would unbinding and directly using an i2c library help?

dlech, 2016-02-15:
Using shell script will be slow because it is starting a new process for each command, which is what takes most of the time here. Try it using python instead.

Many have made balancing robots with slower polling rates, so there are probably other things you could tweak too. On the other hand, if you want to go for "absolute fastest possible" polling, then you must unbind the driver and use `/dev/i2c_inX` with an i2c library instead.

jdchristensen, 2016-02-15:
Yes, I suspect that the polling rate isn't the key problem, but it still would be good to know how to poll this sensor more quickly.

I tried reading from the `direct` file from python.  If I open it once and then repeatedly seek and read, the data is not refreshed.  If I open, seek, read and close in a loop, then most of the time the path to the direct file changes after just a few iterations of the loop:  the `sensorXX` part changes.  Occasionally I make it through 10 iterations, but my sample rate is only 11Hz in these cases, even if I only read one byte.  This is similar to the 8Hz I got with my test using the shell.  So something fishy is going on when reading from the `direct` file.

dlech, 2016-02-15:
The reason it is disappearing is that you are not polling fast enough and the auto-detection thinks the sensor is disconnected. You can put the port in `nxt-i2c` mode to prevent this. When using the `direct` attribute, yes, no other attributes will be updated.You read the value from the `direct` attribute. There could be a bug if seek and read does not work without opening and closing the file first (open a new issue for that).

In the shell example where you use `hd`, I have found that `hd` actually  reads every address up to the `-s` value, which is why it takes so long. It is doing 0x54 I2C messages.

In the mean time, give the python-smbus library a try.

jdchristensen, 2016-02-16:
I verified that if I do

`echo nxt-i2c > /sys/devices/platform/legoev3-ports/lego-port/port0/mode`

then the sensorXX number is stable.  I can get 13Hz if I repeatedly open and close the `direct` file.  If I don't open and close, the data doesn't change.  I created a separate issue for this: https://github.com/ev3dev/ev3dev/issues/552

jdchristensen, 2016-02-16:
The `python-smbus` library works pretty well for this, especially since I only need some of the data from the sensor.  Below, I'll give some comparisons, but first, here's the set-up.

Method 1 uses the `python-ev3dev` library and the `mode` trick.  First, turn off auto-detection to make the device path stable, and then turn off polling:

    echo nxt-i2c > /sys/devices/platform/legoev3-ports/lego-port/port0/mode
    echo 0 > /sys/devices/platform/legoev3-ports/lego-port/port0/i2c-legoev3.3/i2c-3/3-0011/lego-sensor/sensor0/poll_ms

(A few of the numbers there can vary.)  Then I repeatedly read `bin_data` after setting the `mode`.

Method 2 uses the `python-smbus` library.  I unbind the sensor with

    echo 3-0011 > /sys/bus/i2c/drivers/nxt-i2c-sensor/unbind

I set things up with

    from smbus import SMBus
    bus = SMBus(3)
    devaddress = 0x11

Then I read data with calls like

    d  = bus.read_i2c_block_data(devaddress, 0x45, 6)

where `6` is the number of bytes to read and `0x45` is the starting location.

My sensor has 23 bytes.  If I read them all (using `mode = 'ALL'` or by specifying 23 bytes in `read_i2c_block_data`), I get 32Hz with method 1 and 35Hz with method 2, not much different.

But if I read two non-consecutive groups of 6 bytes (by alternating `mode` between `ACCEL` and `GYRO`, or by calling `read_i2c_block_data` twice), I get 37Hz with method 1 and 48Hz with method 2.

If I read three consecutive bytes, I get 97Hz and 139Hz.

And if I read just one byte, I get 198Hz with method 2.  This can't be done with method 1.

For my application, I need 8 bytes, and can get 61Hz instead of 37Hz.  This improves my balancing robot noticeably!  Thanks for the help getting to this point!

jdchristensen, 2016-02-16:
With the correct use of the ev3dev `direct` attribute (see https://github.com/ev3dev/ev3dev/issues/552), I can now confirm that it works without needing to be closed and reopened each time, and is almost exactly the same speed as using the python-smbus library ("Method 2" in the above description).

bmegli, 2016-06-24:
Some more light on the issue. I will make math with CruizCore but this goes to any I2C device, just the number of bytes read changes.

EV3 I2C works at 10000 baudrate.
CruizCore doesn't really have modes and always reads 10 bytes, it's my implementation, can be improved.

I2C communication goes as follows:
- start (>= 1 clock)
- slave addres + R + ACK (9 clocs)
- register address + ACK (9 clocks)
- start (>= 1 clock)
- slave address + read + ACK (9 clocks)
- read data (10 bytes * 8 bits + 10 ACK, 90 clocs)
- stop (>= 1 clock)

Which gives at least 1+9+9+1+9+90+1=122 SCL clock pulses

Now with 10000 baudrate this gives:
`10000/1=122/x` and hence `x=0,122 s` which is more than 12 ms to just read the data (optmistically).

Setting poll ms below 12 ms is clearly not possible with such baudrate for CruizCore.

With other arbitrary I2C sensor reading at least 2 bytes this still will need at least 5 ms to just read the values. (9+9+9 +1 +1 +1 +  2*9 for data)