ev3dev/ev3dev #446: kernel compile 3.16.7-ckt19-9-rc1-ev3dev-ev3

meehien, 2015-12-08:
I have compiled the current git version of the kernel 3.16.7-ckt19-9-rc1-ev3dev-ev3 for the EV3 in order to add dualshock 4 support. The problem is that when I boot the new kernel I lose support for lego motors and sensors. On the brick the the device info says something like lego-l-motor attached to port (null) and sensors simply don't appear.

Does anyone have any ideas about why the ports are no longer available in my kernel?

dlech, 2015-12-08:
The latest git version won't work with most existing programs because of https://github.com/ev3dev/lego-linux-drivers/commit/37ad261d25ccab7ee24276d16fc83cc38b85fb65. Also see #433. Make sure this commit is not included when you compile your kernel.

meehien, 2015-12-08:
thanks. it works now. I didn't pull drivers/lego and seems to build the kernel properly now.

closed 2015-12-08 by meehien