ev3dev/ev3dev #447: How to wait a motor until it rotates from X degree to X+Y degrees?

jabrena, 2015-12-08:
Hi mates,

I am checking the methods:

``` bash
echo on > speed_regulation
echo 50 > speed_sp
cat speed_sp

cat position_sp
echo 100 > position_sp
echo run-to-abs-pos > command / echo run-to-rel-pos > command
``` 

In order to add some methods to rotate to some degrees in relative or absolute way.
https://github.com/jabrena/ev3dev-lang-java

Currently, when the you execute the method:

``` bash
echo run-to-abs-pos > command
```

The statement returns inmediatelly and my question is:

How to check that the rotate method is running?

Example:

initial position: Degree 30

``` bash
echo on > speed_regulation
echo 50 > speed_sp
cat speed_sp

cat position_sp
echo 90 > position_sp
echo run-to-abs-pos > command / echo run-to-rel-pos > command
``` 
In this example a Regulated Motor begin the process to rotate from degree 30 to degree 90. How to check in a loop that the statement " run-to-abs-pos > command" is changing the position_sp (31,32,33, ... 89,90)

I would like for example implement this sintax in Java:

``` bash
        EV3LargeRegulatedMotor mA = new EV3LargeRegulatedMotor(MotorPort.A);
        mA.setSpeed(100);

        Sound.beep();
        System.out.println(mA.getTachoCount());
        mA.rotate(90);
        Sound.beep();        
        System.out.println(mA.getTachoCount());
        mA.rotate(90);
        Sound.beep();  
        System.out.println(mA.getTachoCount());
        mA.rotate(90);
        Sound.beep();  
        System.out.println(mA.getTachoCount());
        mA.rotate(90);
        Sound.beep();
        System.out.println(mA.getTachoCount());
```

Many thanks in advance

This issue is a continuation from the previous one:
https://github.com/ev3dev/ev3dev-lang/issues/130

ddemidov, 2015-12-08:
There is `state` attribute (described in http://www.ev3dev.org/docs/drivers/tacho-motor-class/).

dlech, 2015-12-08:
> "run-to-abs-pos > command" is changing the position_sp

If starting the `run-to-abs-pos` is causing the value returned by the `position_sp` attribute to change, this is a bug.

To answer the main question though, right now, there is really not a good way to do this and it is something we need to think about.

rhempel, 2015-12-08:
I thought I implemented a `sys_notify` for when the current operation is complete and the motor is in the `idle` state - I even wrote a little watcher but I probably did not package it for building for Debian release.
This is one of those little utilities that we should think of packaging into a ev3dev-tools bundle - it's where I'd like the `event2state` utility I'm working on to go as well.

jabrena, 2015-12-08:
Now, I  in the programming language side, I found a tactical solution:

``` java
    public void rotate(int angle, boolean immediateReturn) {
		final String attribute1 = "position_sp";
		this.setAttribute(attribute1, "" + angle);
    	final String attribute2 = "command";
		final String value2 = "run-to-rel-pos";
		this.setAttribute(attribute2, value2);
		
		if (!immediateReturn) {
			//TODO Check if it possible to improve this way.
			//Add a double check (Time control to avoid race condition problem)
			while (true) {
				if (this.getAttribute("state") != "running") {
					break;
				}
			}
		}

    }
```

and this way run.

jabrena, 2015-12-08:
Testing the method rotateTo, I think that exist an internal problem and I am going to show with an example:

``` java
package ev3dev.examples;

import lejos.utility.Delay;
import ev3dev.hardware.Battery;
import ev3dev.hardware.Shell;
import ev3dev.hardware.Sound;
import ev3dev.hardware.port.MotorPort;
import ev3dev.hardware.motor.EV3LargeRegulatedMotor;

//java -cp ev3-lang-java-0.2-SNAPSHOT.jar ev3dev.examples.Test4
public class Test4 {
	
    public static void main(String[] args) {
    	
        EV3LargeRegulatedMotor mA = new EV3LargeRegulatedMotor(MotorPort.A);
        mA.setSpeed(100);

        //mA.resetTachoCount();
        System.out.println(mA.getTachoCount());
        Sound.beep();
        System.out.println(mA.getTachoCount());
        mA.rotateTo(0);
        Sound.beep();
        Delay.msDelay(1000);
        System.out.println(mA.getTachoCount());
        mA.rotateTo(90);
        Sound.beep();  
        Delay.msDelay(1000);
        System.out.println(mA.getTachoCount());
        mA.rotateTo(180);
        Sound.beep();
        Delay.msDelay(1000);
        System.out.println(mA.getTachoCount());
        mA.rotateTo(270);
        Sound.beep();
        Delay.msDelay(1000);
        System.out.println(mA.getTachoCount());
        mA.rotateTo(0);
        Sound.beep();
        System.out.println(mA.getTachoCount());

        System.out.println(Battery.getVoltage());
    }
}
```

RotateTo method:

``` java
    public void rotateTo(int limitAngle, boolean immediateReturn) {
    	final String attribute1 = "position_sp";
    	this.setAttribute(attribute1, "" + limitAngle);
    	final String attribute2 = "command";
		final String value2 = "run-to-abs-pos";
		this.setAttribute(attribute2, value2);
		
		if (!immediateReturn) {
			//TODO Check if it possible to improve this way.
			//Add a double check (Time control to avoid race condition problem)
			while (true) {
				if (this.getAttribute("state") != "running") {
					break;
				}
			}
		}
    }
```

When I execute the method, I noticed that the motor goes in the same direction for degree 0, 90, 180 & 270 but suddlenly when the motor need to turn to 0 degree, the turn in the inverse direction. I think that in all cases, the motor should turn in the same direction.

Output:

```
root@ev3dev:/home# java -cp ev3-lang-java-0.2-SNAPSHOT.jar ev3dev.examples.Test4
100
0
0
on
0
on
90
on
180
on
270
on
0
6979666.0
root@ev3dev:/home# 
```

dlech, 2015-12-08:
> I thought I implemented a sys_notify for when the current operation is complete and the motor is in the idle state

You did, but the motor driver has changed significantly since then so I doubt that it works anymore.

Also, there are lots of race conditions to deal with here. If you start `poll` (or `select`) for the notification before you send a command to the motor, another command could be running and trigger it prematurely. If you start polling after you send the command, the command could finish before `poll` is started (e.g. `run-to-rel-pos` where `position_sp` is 0). In either case, your program will be deadlocked.

Also, for programming languages that don't have a `poll` or `select` function, you have to call an external program to do it for you, which will make your program slow.

Although it goes against the advice of "the right way" of other kernel experts, we could make `*-wait` versions of each motor command. These would cause the write to the sysfs attribute to not return until the command is complete. I have read in a couple places that long running synchronous operations in sysfs are not a good idea, so I am not sure what other sorts of implications this has.

dlech, 2015-12-08:
> When I execute the method, I noticed that the motor goes in the same direction for degree 0, 90, 180 & 270 but suddlenly when the motor need to turn to 0 degree, the turn in the inverse direction. I think that in all cases, the motor should turn in the same direction.

When using `run-to-abs-pos`, positions are absolute. This is the expected behavior. It sounds like you want `360`, not `0`. Or use relative positions.

WasabiFan, 2015-12-08:
@jabrena In your sample, the code uses the comparison

```java
this.getAttribute("state") != "running"
```
... to check if the state returns "running". You'll want to change this to the following instead:

```java
!this.getAttribute("state").equals("running")
```
Because Java compares objects with `==` by instance (and Strings are reference types), so that won't actually check for value equality.

dlech, 2015-12-08:
furthermore, `state` returns a space separated list of flags, so, for example, it could return `running ramping` too.

bmegli, 2015-12-08:
@jabrena Good Job! You have made everybody think

In ev3dev past it was as easy as checking ``` run==1 ```, it seems it got a bit complicated along the way.

I suppose in java, without bindings you would check:

``` java
this.getAttribute("state").contains("running")
```

I would probably make a method of it:

``` java
boolean isRunning()
{
 return this.getAttribute("state").contains("running");
}
```

Then use along the way:

``` java
while(someMotor.isRunning())
{
   // do stuff or do nothing
   // possibly sleep for some short interval to not block
}
```

But this seems overly complicated and a bit inneficent for a simple thing to do. You got your point ;-)

Kind regards!

bmegli, 2015-12-08:
Ok Guys!

Let's make review:

LabView (original LEGO software)

```
blocking for rotations
blocking for counts 
blocking for seconds
non-blocking on
non-blocking off
```

LeJos
``` java
public void waitComplete()
// Wait until the current movement operation is complete (this can include the motor stalling).
```

MonoBrick

``` C#
   WaitHandle[] handles = new WaitHandle[2];
   handles[0] =  motorA.SpeedProfile(50, 200, 1600, 200,true);
   handles[1] = motorD.SpeedProfile(50,200,600,200,true);
   //you could do something else here
   LcdConsole.WriteLine("Waiting for both motors to stop");
   WaitHandle.WaitAll(handles);
```

RobotC

``` C
nMotorEncoder[motorB] = 0;                // reset the Motor Encoder of Motor B
nMotorEncoderTarget[motorB] = 360;        // set the  target for Motor Encoder of Motor B to 360
motor[motorB] = 75;                       // motor B is run at a power level of 75
motor[motorC] = 75;                       // motor C is run at a power level of 75
 
while(nMotorRunState[motorB] != runStateIdle)  // while Motor B is still running:
{
  // do not continue
}
motor[motorB] = 0;                       // motor B is stopped at a power level of 0
motor[motorC] = 0;                       // motor C is stopped at a power level of 0
```

Just to think over, ranging from the simplest to most complicated like in ev3dev.

But people usually expect simple things to be simple. 

That's @jabrena point.

bmegli, 2015-12-08:
Another point.

I program industrial hard real time systems in the course of my work.

Now - in the main controller there is *strictly* no possibility of doing anything synchronous.

Programming model: cycle with few ms, check states, set outputs, process asynchronous communication, done (next cycle).

Waiting for something to finish too long actually stops the controller with critical error.

More to say on that but I have to get the kids to bed.

rhempel, 2015-12-08:
Great conversation so far !

jabrena, 2015-12-09:
Good morning Team!

@WasabiFan 
Yes, Strings must be compared with .equals() always :p 
Current code was not released yet. I will refactor

@dlech 
I will refactor the method to process the array. (running ramping)

@bmegli 
Many thanks for the idea to refactor the ubly while(true) { }
I will follow that idea.

@dlech 
Yesterday, I had to stop testing the example, but I continue thinking that something goes rare, but I will try to check the code and try to record an internal video to check with this multi disciplinary team. :)

@bmegli 
Once, I finish v0.2.0 I will adquire a Cruizcore XG1300L to explore Gyro world.
http://www.minfinity.com/eng/page.php?Main=1&sub=1&tab=5

I think that mixing odometry fron EV3Dev regulated motors + Gyro data using a Kalman filter could be great to have a local navigation feature.

Well, continue with the day.

Cheers

bmegli, 2015-12-09:
Guys,

First, I think that we lack the correct context for this discussion.

@jabrena has practically made a port of leJOS to ev3dev. This is almost complete compatibility layer so that leJOS programs will be able to run on ev3dev in near future hopefully.

See here:

https://github.com/jabrena/ev3dev-lang-java

This is great job @jabrena.

Now if he has doubts about something treat them extremely seriously.

Doubts mean that there is some easy way or abstraction to do things in leJOS which ev3dev currently lacks.

Now I believe that this compatibility layer with leJOS should become part of ev3dev at some point. Either in apt package like ev3dev-lejos or some other way (if java is preinstalled on ev3dev).

While ev3dev is ahead in some ways of other EV3 frameworks we are also behind in other.

I anticipate similiar problems soon with motor synchronization. This is big thing that still needs to be done.

bmegli, 2015-12-09:
There are several issues here so let's see what we have:

1. An ad-hoc solution for @jabrena for leJOS waitComplete()
2. An ad-hoc getting started documentation for waiting for motor to complete
3. A decission whether synchronous kind of framework should become part of ev3dev

If 3 == true

4. A user filesystem (?) solution for waiting for the motor/sensor to complete, that is not busy waiting or polling from the user space
5. A ev3dev-lang level solution for waiting for the motor/sensor to complete
6. An efficent kernel level solution for monitoring the motor and notyfing through fs (?) the upper layers

bmegli, 2015-12-09:
1. An ad-hoc solution for @jabrena for leJOS waitComplete()

In leJOS we have:

``` java
public void waitComplete()
// Wait until the current movement operation is complete (this can include the motor stalling).
``` 

So an ad-hoc in ev3dev could be something along the lines:

``` java
boolean isRunningOrStalled()
{
 return this.getAttribute("state").contains("running") || this.getAttribute("state").contains("stalled");
}
```

``` java
public void waitComplete()
{
   while(this.isRunningOrStalled())
   {
      // probably sleep for some short interval to not block
   }
}
// Wait until the current movement operation is complete (this can include the motor stalling).
``` 

This can be naturally optimized. E.g. instead of contains something else might be better.
If ev3dev has some framework for doing it better in the future than @jabrena can change his implementation details.

bmegli, 2015-12-09:
> 2. An ad-hoc getting started documentation for waiting for motor to complete

Just getting started example to add of how to do it through filesystem.

People really want to wait for the motor to complete so they should have a solution ready in getting started.

bmegli, 2015-12-09:
> <div> 3. A decission whether synchronous kind of framework should become part of ev3dev </div>

Needs discussion:
-if at all
-implementation layer (should it be kernel or bindings implementation only)
-scope (should it be motors only or sensors too?)

I kind of am impressed by MonoBrick aproach:

``` C#
   WaitHandle[] handles = new WaitHandle[2];
   handles[0] =  motorA.SpeedProfile(50, 200, 1600, 200,true);
   handles[1] = motorD.SpeedProfile(50,200,600,200,true);
   //you could do something else here
   LcdConsole.WriteLine("Waiting for both motors to stop");
   WaitHandle.WaitAll(handles);
```

I suppose one can wait there also for other events like sensors exceeding thresholds etc but I haven't checked.

jabrena, 2015-12-09:
Hi, 

at the moment, I can't consider the project:
https://github.com/jabrena/ev3dev-lang-java

as a LeJOS port to EV3Dev, because leJOS scope cover many aspects:

* Motors
* Sensors
* Remote Comms
* Bluetooth
* Local Navigation

Besides this projects add new features as the usage of EV3Dev TTS (eSpeak). At the moment, I think that it is a Java binding library for EV3Dev.

Well, returning to the doubt, I will try to refactor the loop to wait.

bmegli, 2015-12-09:
> <div> 4. A user filesystem (?) solution for waiting for the motor/sensor to complete, that is not busy waiting or polling from the user space </div>

Whether what we have now is enough or there should be some simple attribute like complete that only changes like from idle/running/stalled?

Maybe we could use the inotify framework.

bmegli, 2015-12-09:
@jabrena 

This is great start. Don't let anybody make you shift to ev3dev-lang syntax. leJOS compatibilty layer, even incomplete is a great thing.

bmegli, 2015-12-09:
> <div> 5. A ev3dev-lang level solution for waiting for the motor/sensor to complete </div>

Sincerely - no idea. Either synchronous versions of each function like @dlech said (```*-wait```) or a function leJOS like ``` waitComplete ``` or a framework like MonoBrick ``` WaitHandle ```. 

I like MonoBrick way. It's in spirit ``` select ```.  It's a most generic. It can have timeout also like select. Maybe we could just integrate select somehow to ev3dev framework so you could wait for motors/sensor/network writes and reads, filesystem I/O.

But then again - I am ``` select ``` lover. One process sleeping allmost all the time and doing things occasionaly.

bmegli, 2015-12-09:
> <div> 6. An efficent kernel level solution for monitoring the motor and notyfing through fs (?) the upper layers </div>

This one is the most tough.

This was I believe from @dlech to @rhempel about some framework they are cooking ;-)

> Also, there are lots of race conditions to deal with here. If you start poll (or select) for the notification  before you send a command to the motor, another command could be running and trigger it [...]

In electronics and industrial automation usually a concept of edges is used. E.g. a positive edge triggers positioning, some edge notifies that the setpoint was accepted. There are ``` in position ``` flags sometimes. Usually there is only one *master* controlling the device and this makes things easier. To think and elaborate.

> Also, for programming languages that don't have a poll or select function, you have to call an external program to do it for you, which will make your program slow.

There has to be better solution for that (better than external program). I believe it's not that bad:
- C/C++ obviusly has that
- python has that
- java, anybody?
- .Net has it at least for Sockets in ``` System.Net.Sockets.Select(IList checkRead, IList checkWrite, IList checkError, int microSeconds) ```, anybody knows something more?
- ...

dlech, 2015-12-09:
> > 3\. A decission whether synchronous kind of framework should become part of ev3dev

> -if at all

I say yes. This is what MINDSTORMS users are already used to in other programming platforms.

> -implementation layer (should it be kernel or bindings level implementation only)

The sample from leJOS is actually good for deciding what to do here. Does the synchronous command only stop when a position is reached or does it stop also when the motor is stalled? I think this is an argument for not including synchronous in the kernel drivers. There are too may use cases and it would make the driver overly complicated. I think at binding level makes more sense.

> -scope (should it be motors only or sensors too?)

Same with sensors. For example, programmable limit switches has been brought up several times before. These are very useful, but would be much easier to maintain in userspace rather than in kernel drivers.

dlech, 2015-12-09:
> Maybe we could use the inotify framework.

I don't think `inotify` works with sysfs.

dlech, 2015-12-09:
> There has to be better solution for that (better than external program)

We set the bar very low. We should be able to write a program in `bash` that works the same as any other language.

bmegli, 2015-12-09:
> The sample from leJOS is actually good for deciding what to do here. Does the synchronous command only stop when a position is reached or does it stop also when the motor is stalled?

``` java
public void waitComplete()
// Wait until the current movement operation is complete (this can include the motor stalling).
``` 

The comment above comes from leJOS documentation (so reached/stalled)

 I would say that synchronous command should finish if user intervention *may* be required. 

> I don't think inotify works with sysfs.

Yes, you are right. My lack of knowledge.

jabrena, 2015-12-09:
Hi,

I have just refactored the methods with the notes from this issue. :)

In relation to my doubt about the turn, I have created a mini video here:
https://drive.google.com/file/d/0B6G7VJZ3-nZCT0tidnM1YWlQWUU/view?usp=sharing

Code:
https://github.com/jabrena/ev3dev-lang-java/blob/master/library/src/main/java/ev3dev/examples/Test4.java

If you observe, when the motor is in 360 degree, the wheel turn in the contrary direction. This is my doubt.

@bmegli This week, I will use  "waitComplete()" in the methods. Oki Doki.

Cheers

dlech, 2015-12-09:
You are still telling your motor to go back to 90 after you are at 360. If you want to keep going in the same direction, then you have to add 360 + 90.

bmegli, 2015-12-09:
@jabrena 

Ok, I now see why you are confused.

The range of encodoers (motor position) is huge, around +-2^31 so it's around +-2147483648.

It's not only 0-360 as in degrees. But yes, one encoder increment is one degree of motor shaft.

Moving to absolute position means: move from current position to desired, it takes shortest way.

Imagine robot with hand. You could position hand with abosolute command. Hand up would mean some absolute position, hand down some other, etc.

Also you would need more then just 360 steps if there were gears for instance.

The range of encoders is so huge that you will probably never reach limit. 

With medium motor turning at full speed (1200 counts per rotation) you would need:

2147483648/1200 ~= 1789569 seconds ~= 29826 minutes ~= 497 hours of continious rotation at full speed

bmegli, 2015-12-09:
Ok, I think I finally grasped.

There were two reasons of confusion here:
- you set the motor with speed_sp positive or negative so you would expect it to rotate this way
- you probably thought that the range is 0-360

Range is not 360, it's so huge that going the other way is not an option. For position modes only absolute value of speed_sp counts.

rhempel, 2015-12-09:
+1 for implementing motor and sensor notifications in userspace

jabrena, 2015-12-10:
Good morning,

@dlech now, I have the idea clear. I will use the idea that you say.
Normally this kind of methods is used to move arms add degrees and subtract degrees.

In relation to the possibility to add a new parameter in some methods to wait.
At the moment, I have a way to handle this way but I recognize that if the low level handle this part, it could be great. Anyway, I think that we (the whole team) should priorize the list of tasks. 

Do you like if we close this issue and we create a new issue in order to discuss the option to add a parameter in some methods to wait or not the rotation?

In my case, I will continue checking some methods for regulated motors to close this release.

bmegli, 2015-12-10:
@jabrena 

> Normally this kind of methods is used to move arms add degrees and subtract degrees.

``` run-to-rel-pos ``` is used for that. Then the position is interpreted as relative. You can specify it negative or positve. 90 to position_sp and command run-to-rel-pos moves 90 CW. -90 to position_sp moves 90 CCW.

> At the moment, I have a way to handle this way but I recognize that if the low level handle this part, it could be great

I believe that both @dlech and @rhempel decided that this should be implemented in the user space (bindings level). 

I am not entirely sure about @rhempel opinion:
> +1 for implementing motor and sensor notifications in userspace
@rhempel - do you mean the same as the @dlech, synchronous api on the bindings level or some other notification system (e.g. async).

> Do you like if we close this issue and we create a new issue in order to discuss the option to add a parameter in some methods to wait or not the rotation?

Close the issue if you believe that your original doubt has been resolved.
As for the parameter - ev3dev-lang has its own language specification in ev3lang. The bindings are autogenereted for supported languages by autogen system.

Since you are following leJOS api in your bindings not ev3dev-lang you should do what's in leJOS.

Putting it simply - either follow leJOS api as you did so far or switch to ev3dev-lang specifcation and autogen system.

jabrena, 2015-12-10:
Yes, I am going to to close the issue.

For next minor version, v0.3.0, I will add some Java classes generated by autogen system.

closed 2015-12-10 by jabrena

bmegli, 2015-12-10:
There is one last thing we should add here.

ev3dev sysfs api (the attributes/paths/names of the EV3 devices in filesystem) still change sometimes. That is one of the reasons that autogen is used. If something changes new bindings are automatically genereted.

If you decide to implement leJOS api manually you will also have to make changes in your api manually.

There is always an option - autogen ev3dev-lang java bindings and then build leJOS api on top of that (if you want leJOS api). This way you are independent of changes to some extent.

jabrena, 2015-12-10:
Hi @bmegli,

In the next SCRUM, v0.3.0, I have planned to generate a class using templates (autogen) to generate a class with attributes to avoid the problem that you say, but I have to priorize the time, at the moment, I have 2 hands only, I suppose that in the time, we will have more contributors for this binding. :)
https://github.com/jabrena/ev3dev-lang-java/issues/6

rhempel, 2015-12-10:
Strongly suggest that language bindings follow the suggestion by @bmegli - use `autogen` to bet the care functionality assembled, then manually add a layer.

Different languages will have varying levels of support for what we are trying to do. For example, Python and Lua have support for `select` type operations, along with some form of co-operative mutitasking.

It's useful to be able to educate programmers with the different ways of handling the problem of detecting when a long-running process is complete, while still doing other useful work.

jabrena, 2015-12-13:
Hi @rhempel & @bmegli, 

I followed your advices to work in the issue to integrate the library with EV3Dev using autogen project.

Cheers

kortschak, 2016-05-29:
>Although it goes against the advice of "the right way" of other kernel experts, we could make *-wait versions of each motor command.

I've just come up against this in writing an example for ev3go. The example needs to (for pedagogic reasons) implement a "steering block"-like function. A set of `*-wait` commands can help here with careful ordering of the `run-to-*-pos > command` writes, but with any more than 2 motors with more complicated timing relationships it does not.

rhempel, 2016-05-29:
@kortschak - the "right" way to do it, if your language supports it, is to use the sysfs API and set up a handler so that when the motor finishes the move and is no longer running, a callback is triggered in your program.

Could you briefly describe your scenario so that we can consider other ways to accomplish the task?

kortschak, 2016-05-29:
Yeah, it's doable with the sysfs API, but it involves a busywait which is pretty horrible. No callback is needed.

The case I'm working on is mimicking the EV3-G steering block which runs two large motors at potentially different speeds, stopping the slower one when the faster one has reached a relative position set point. You could conceive of systems with three motors where the action of one is dependent on one and then the other of the other two (both active) having reached distinct set points. Here, as before a busywait polling the motors' states gets you what you want, but in an ugly way.

If there were a `wait` attribute that caused a blocking write with state flags, you would get the benefits of the `*-wait` approach above, but in a more generalisable way. With a separate `wait` attribute you decouple the initiation of a motor action from the wait for completion. This allows you to create arbitrarily complex wait groups on the motor actions. How the flags are indicated in the write may be an issue, though not huge.

rhempel, 2016-05-29:
Maybe I'm not understanding how `wait-*` is different than a busy wait loop - can you provide a code sample of what you have today and how you would like it to be?
The entire point of `sysfs` is to avoid busy waits - you register a callback to "do something when the attribute changes" and then go about your business looking for sensor input or button presses.
Only some languages support this without add-on libraries - and of course your language has to support multiple threads natively.

kortschak, 2016-05-29:
Sorry, I misunderstood.

A callback like that would be painful in Go (I don't see any pure Go libraries for it, so I'd have to do it via C breaking cross-compilation, which is crucial because of the weakness of the EV3).

The code I have is not ready (dealing with strange differences in behaviour between the EV3-G program and the Go program). I'll post it in the next few days.

dlech, 2016-05-29:
I think this issue, despite its length, it pretty cut and dry. The question is: how do you wait for a motor to stop. The answer is: if you want a userspace program to wait until a motor stops, you have to read the `status` attribute in a loop.

At one time, we had it working where you could use `poll()` or `select()` to be notified of a change in motor run status. This was never widely used and is certainly broken since the `10-ev3dev` kernel release. If you would like to propose restoring this functionality, you should open a new issue.

If neither of these options satisfy your needs, you should open a new issue to brainstorm what else is possible.

rhempel, 2016-05-29:
I think I'd like to bring the `poll()` or `select()` capability back, even if it was not widely used.
Reading `status` in a loop is a bit ugly, but many languages support threads that can be triggered at timed intervals, and that sure helps with CPU usage. The motors can run to a specified position and then stop - if you check every 50 msec, is that good enough? In many cases, yes it is.

dlech, 2016-05-29:
> think I'd like to bring the poll() or select() capability back

then please open a new issue :wink:

rhempel, 2016-05-29:
Get back to your Memorial Day Weekend :-)