ev3dev/ev3dev #448: update node to version 4.0

joshrtay, 2015-12-09:
Node 4 is the current stable release. It would be great if it was included in ev3dev.

WasabiFan, 2015-12-09:
This is not technically doable due to the removal of support for armv5 (the EV3's processor architecture) in V8, node's JavaScript engine. They have confirmed that V8 will not re-add armv5 support, so we are stuck with the older version for the foreseeable future.

That being said, Microsoft has a current effort in play to get their Chakra JavaScript engine running inside of Node using a V8  "shim". They haven't yet open-sourced their code, which means that we don't know what architectures or platforms they plan to support. It is currently Windows-only, so that would need to change to make this option viable for us.

joshrtay, 2015-12-09:
Can the BrickPi support node 4? Do you think 4 could be included in that image?

dlech, 2015-12-09:
There is a package for nodejs 4.2.2 in debian testing. There is even an armel arch version. I think you should try installing it an see what happens.

https://packages.debian.org/stretch/nodejs

dlech, 2015-12-16:
Has anyone tried this yet? If it works, I will include it in the next kernel release.

jabrena, 2015-12-16:
The problems was form EV3 because it has an ARM5. Raspi has a better hardware so, It should work fine and run no de 4.2.3 LTS.

kdawgwilk, 2016-02-04:
I'm curious why do they market the ev3 as having a ARM9 CPU?  Is there something I am missing? I get these results on the ev3 brick and it confused me:
```
robot@ev3dev:~$ lscpu
Architecture:          armv5tejl
Byte Order:            Little Endian
CPU(s):                1
On-line CPU(s) list:   0
Thread(s) per core:    1
Core(s) per socket:    1
Socket(s):             1
Model name:            ARM926EJ-S rev 5 (v5l)
```
Why does it report both ARM5 & ARM9?
(http://botbench.com/blog/2013/01/08/comparing-the-nxt-and-ev3-bricks/)

moriarty, 2016-02-04:
There is v that makes the difference.

ARMv5 is the instruction set, while ARM926EJ-S the actual processor model number.

dlech, 2017-08-12:
Closing since this is not technically possible on EV3. ev3dev-stretch has node v4.8.2 on non-EV3 hardware.

closed 2017-08-12 by dlech

fogbridgegit, 2017-12-11:
i download dotnet-linux-arm.latest.tar.gz on 12/10 and im getting a
robot@ev3dev:~/dotnetlinux$ ./dotnet
-bash: ./dotnet: No such file or directory

i installed it in a local directory as well as
robot@ev3dev:~$ /usr/bin/dotnet/dotnet
-bash: /usr/bin/dotnet/dotnet: No such file or directory

is there some preinstallation i need to install?
i did a sudo apt-get upgrade before trying.

here are some diagnostics from my linux system
cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 8 (jessie)"
NAME="Debian GNU/Linux"
VERSION_ID="8"
VERSION="8 (jessie)"
ID=ev3dev
ID_LIKE=debian
HOME_URL="http://www.ev3dev.org/"
SUPPORT_URL="http://www.ev3dev.org/support"
BUG_REPORT_URL="https://github.com/ev3dev/ev3dev/issues"
robot@ev3dev:$ cat /etc/debian_version
8.5
robot@ev3dev:$

robot@ev3dev:~$ lscpu
Architecture: armv5tejl
Byte Order: Little Endian
CPU(s): 1
On-line CPU(s) list: 0
Thread(s) per core: 1
Core(s) per socket: 1
Socket(s): 1
Model name: ARM926EJ-S rev 5 (v5l)

====update...

i tried hacking around by opening the dotnet executable and saw a path in there.  so i copied one of the sym links in the lib directory so it could match that path and something changed.

now i get a segmentation fault.

the path was in the first few lines of the dotnet exe its: /lib/ld-linux-armhf.so.3
when i put everything back the way it was i get the original

 -bash: ./dotnet: No such file or directory  silent crash.

any ideas?

dlech, 2017-12-11:
dotnet "arm" is armv7, so it won't run on EV3, which is armv5 (for the same reason nodejs doesn't work anymore). See https://github.com/dotnet/corert/issues/477

fogbridgegit, 2017-12-11:
thanks for your help...  ahh man im so stuck then.   originally just wanted the node.js signalr-client to work on node.js but number of problems led me to beleive that i needed to update nodejs which is not updateable.    plan b was to try to run dotnet console app connect back to signalr.. but that doesnt work either.    maybe the answer is either Window IoT core but im guessing thats not going to work on my ev3 robot.    not sure i can $afford to try and fail by buying a brickpi + raspbperrypi3 to replace the ev3  ( cost me 350$ already )  but nodejs signalr client would probably work there as would .net core 2.0   ..  what do you think i should do?

dlech, 2017-12-11:
If you have $150, BrickPi3 + RapsberryPi 3 is nice and can run just about anything you want way faster. On the EV3, probably best to stick with Python.

fogbridgegit, 2017-12-11:
wifes gonna kill me. lol   #thanks