ev3dev/ev3dev #449: Formatting of attribute docs on website

WasabiFan, 2015-12-09:
Currently, attribute docs for drivers are generated as lists in markdown and then converted, unstyled  (as far as I can tell), to HTML. This leaves us with pages looking like this: (<http://www.ev3dev.org/docs/drivers/tacho-motor-class/>)

[image]

What can we do about this? I was under the impression that this was formatted as a table, but it doesn't look like that at the moment...

dlech, 2015-12-09:
It is using dictionary html elements (`dl`, `dd`, `dt`).  Using a table is a nice idea, but these come from documentation comments in the kernel source code. I don't know of a way to keep it readable in the source code and make it a table on the website.

Perhaps you could work some css magic with the dictionary elements?

ddemidov, 2015-12-09:
Something as simple as adding html anchors to the attributes would be great already.

WasabiFan, 2015-12-09:
@dlech What's parsing the values that look like this?
```c
/**
 * @vendor_name: LEGO
 * @vendor_part_number: 45502
 * @vendor_part_name: EV3 Large Servo Motor
 * @vendor_website: http://shop.lego.com/en-US/EV3-Large-Servo-Motor-45502
 */
```

And what is generating the dictionary elements from the raw text?

>  I don't know of a way to keep it readable in the source code and make it a table on the website.

I think there are three main options to improve the readability and styling of these data:
- Add more structure to the comment data
- Roll our own styles for the dictionary elements
- Write some JavaScript to convert the dictionary in its current form to a table on the site

Currently, it looks like the content of each page is a large block of markdown (1 comment per class, 1 class per page) -- is this correct? If so, I think that the most elegant solution would be to add attributes such as the ones in the code block above to the comment text and then separate that text into a single comment per attribute. It seems that there is one method defined for each action that the class performs (or two if it's get-set), so in my mind splitting it into separate comments and relating them to the corresponding code would be optimal.

That being said, making changes like that would require at least an hours' worth of work, and I don't think I have that time to spend right now -- so I think we should just look at the second and third items on my list above. I will take a look at the structure that it is currently producing and see what options we have that just involve modifying the site.

WasabiFan, 2015-12-09:
OK, it looks like our issue is that Bootstrap is formatting the titles as bolded headings, but because the titles are mostly code it doesn't visually divide the text much. I have a very simple fix for this -- expect a PR shortly.

WasabiFan, 2015-12-09:
@dlech If I want to add a class to all those "sysfs attributes" sections, where do I do it? In each class file (the `*.c` files) or in the generator script you wrote?

I'm looking to add the below line after the last key-value pair in the dictionary.
```
{: class="dl-horizontal" }
```

dlech, 2015-12-09:
> What's parsing the values that look like this?

https://github.com/ev3dev/ev3dev-kpkg/blob/ev3dev-jessie/sensor-defs-to-markdown.py

> And what is generating the dictionary elements from the raw text?

http://kramdown.gettalong.org/syntax.html#definition-lists

> Currently, it looks like the content of each page is a large block of markdown (1 comment per class, 1 class per page) -- is this correct?

Yes.

> splitting it into separate comments and relating them to the corresponding code would be optimal.

That would be good but...

> making changes like that would require at least an hours' worth of work

indeed and would require modifying https://github.com/ev3dev/ev3dev-kpkg/blob/ev3dev-jessie/kernel-doc-text-to-markdown.py to paste the pieces back together as well.

dlech, 2015-12-09:
> If I want to add a class to all those "sysfs attributes" sections, where do I do it?

Hmmm... I would rather keep that out of the source code. You could add it to https://github.com/ev3dev/ev3dev-kpkg/blob/ev3dev-jessie/kernel-doc-text-to-markdown.py, but that would get tricky because you would have to write a regex or something that can identify http://kramdown.gettalong.org/syntax.html#definition-lists. 

I think your idea of using javascript to add it at pageload is probably the best option.

WasabiFan, 2015-12-09:
> I think your idea of using javascript to add it at page load is probably the best option.

OK, I'll do that.

closed 2015-12-09 by dlech