ev3dev/ev3dev #450: Add anchors for attributes in website documentation pages.

dlech, 2015-12-09:
Idea from @ddemidov in https://github.com/ev3dev/ev3dev/issues/449#issuecomment-163338196

I think this is a good idea. We can use the javascript function for adding the css class to do this too. However, first we should modify the source code files so that only the attribute name is in the `dt` element.

WasabiFan, 2015-12-09:
By anchors, you mean anchor elements with an `href` right? So, links? If so, what would be the link target?

dlech, 2015-12-09:
The link target would be the individual attribute. So, for example, if someone asked "how do motor commands work?", you could answer "http://www.ev3dev.org/docs/drivers/tacho-motor-class/#commands"

dlech, 2015-12-09:
Actually, anchor might be the wrong terminology. really, all you would need is to add an `id=attribute`

WasabiFan, 2015-12-09:
OK, that makes sense. In that case it would just be an ID on each -- no links required.

dlech, 2015-12-09:
right. I was thinking anchor as in `<a name="attribute_name" />`, but I think `id="attribute_name"` does the same thing essentially.

dlech, 2017-03-10:
Since the driver documentation is moved to readthedocs.org, the anchors are created for that site.

closed 2017-10-28 by dlech