ev3dev/ev3dev #452: ev3dev on non-EV3 platforms (Rasberry Pi/Brick Pi/PiStorms,  BeagleBone/EVB) propaganda ;-)

bmegli, 2015-12-12:
I was going to dissasemble one of my robots and took a "propaganda" movie for ev3dev on RaspberryPi/BrickPi platform (before dissasembling, but ugh, I couldn't make myself to dissasemble it)

The information that ev3dev supports other platforms doesn't seem to be widespread.

http://www.youtube.com/watch?v=dZfVPe_U0yA

Yeah, let's brainwash people a bit ;-)

@dlech - how are you going with BBB/EVB?

I am not sure if we want "Cam-Robot" in projects, I like it, I would say it's a classic. The good thing about it is we would have information about making proper power supply (for BrickPi/Raspberry Pi) + gstreamer pipelines + info on camera integration with Lego Technic. 

 Mapping Robot this or other way will be there when I find time.

dlech, 2015-12-12:
> how are you going with BBB/EVB?

I haven't really touched it yet. (Well I couldn't resist when first got it, so it is booting an image created from brickstrap, but zero drivers implemented yet).

Right now, my priority is getting an image release out that supports PiStorms. Then we can start spreading propaganda about it.

> We should also document all the quirks of other platforms in one place.

Yes, definitely. [Here](https://github.com/ev3dev/ev3dev/issues/412#issuecomment-154242776) is a table that I started on the differences between platforms. 

> I am not sure if we want "Cam-Robot" in projects

We want anything made with ev3dev in projects. :wink:

bmegli, 2015-12-27:
I probably had some brain malfunction when I was writing this post

I am sorry.

Both robots are ev3dev.

closed 2015-12-27 by bmegli

dlech, 2015-12-27:
I think it is a good post. Just a little premature. :wink: