ev3dev/ev3dev #453: Ramping in tacho-motor class

dlech, 2015-12-12:
We need to have a discussion on how to implement ramping that will work with all motor controllers. I'm pretty happy with how ramping is in the dc-motor class right now, but ramping in the tacho-motor class is most likely broken in the case of EV3 or not implemented in the case of BrickPi and PiStorms.

I have some thoughts that have been floating around in my head about this and I need to take some time to organize them and write them down, so more to come...

Also, see previous discussions in #282.

pootle, 2016-01-03:
I have just started messing about some lego motors for some automation, so this may be a silly comment....  I'm using python (2.7.9)

It would be really nice to have a way to change the target speed in speed regulation mode without the motor stopping and starting (unless there is a way and I am missing something), and to be able to ramp the change to the new speed. This is sort of like being able to change duty_cycle_sp with run_direct. (I did try setting speed_sp with run_direct, but nothing happens). I do really like the way speed regulation works with varying loads.

The ramping currently seems to work (I've only used ramp up). I am writing a simple class to synch to motors for a trike platform (2 driven, 1 idler wheel), the comments on pair in the previous thread are also interesting

dlech, 2016-01-03:
> It would be really nice to have a way to change the target speed in speed regulation mode without the motor stopping and starting

This isn't a silly comment, but it has nothing to do with ramping, so it doesn't really belong here on this issue. It's best to open a new issue if you can't find one that exactly matches.

If you want to change the speed in `speed_sp` while the motor is running, write the new value to `speed_sp` and then write the same command (presumably `run-forever`) to `command` to make the change take effect. You don't have to stop and then start again.

rhempel, 2016-01-03:
I am much happier with the way ramping is implemented in the DC and WeDo motor drivers than in the Tacho drivers.

dlech, 2016-01-03:
I like it to, but the problem that we still need to work out is how does ramping work for the extra commands that tacho motors have that dc motors don't. Maybe some commands just don't support ramping?

pootle, 2016-01-03:
OK thanks.

pootle, 2016-01-04:
Bit more testing and I have found out that if ramp_up_sp is >0 then a re-issue of run_xxx (with motor running) stops the motor and runs the ramp_up to the new speed. Setting ramp_up_sp to 0 before a re-issue results in an instant transition to the new speed. This happens with both duty_cycle_sp and with speed_sp

It would be really nice if a new value of duty_cycle_sp or speed_sp (with speed_regulation appropriately set) resulted in a smooth transition to the new setting using either ramp_up or ramp_down as appropriate.

In the meantime I can do some simple ramping 'by hand' so to speak.

Not sure how this compares with the current DC motor behaviour!

I'll raise an issue for this if you like.

rhempel, 2016-01-04:
When you issue a new command, it's as if the ramp is re-started. There's a fine line between putting too much in the driver, but on the other hand you want users to be able to do most motor control the simplest possible way. If you find yourself changing setpoints and issueing new commands while the motor is running, you are essentially doing the motor control by hand. Set the drive mode to `run-forever` and write to the setopints manually...

dlech, 2016-01-04:
> When you issue a new command, it's as if the ramp is re-started.

Yes, but the ramp should start at the current state, not at 0.

> Set the drive mode to `run-forever` and write to the setopints manually...

I think you mean `run-direct`. And only `duty_cycle_sp` works in this case. This means that the regulated speed mode won't work and you have to implement your own PID if you want any kind of regulation.

---

Before raising any other issues, we need to precisely define what the expected behavior is. For example, here is a diagram I came up with to partially explain how ramping works for the dc motor class.

[image]

The ramp up setpoint defines the time it will take to ramp from 0 to 100% If the duty cycle setpoint is only 50%, then the ramp time is only 1/2 of the ramp up setpoint. The ramp down setpoint works similarly. The ramp down setpoint also determines how long the motor will continue to run after the stop command is sent before the stop action takes place.

I need to expand this to show how ramping works if you change the duty cycle setpoint and send another run command before stopping the motor. But I did this in Libre Office Draw and it is a real pain to work with for this type of diagram. If anyone knows of any other software to make this type of diagram, let me know.

rhempel, 2016-01-04:
Nice job! And your explanation of ramping behaviour when the motor is already running also makes sense. So if the motor is set to a higher duty cycle, and then you change to a lower value and issue `run-direct` does the ramp-up time get applied even though the speed is decreasing? :-)

pootle, 2016-01-04:
Great explanation dlech, this was my understanding / assumption, glad to see it confirmed. Also that ramping should start from the current state - currently it restarts from 0.

Sorry, yes, should have been run-direct.... 

It would be fandabidozy with ramping working AND run_direct able to work with speed regulation on as well as off (i.e. be able to dynamically update speed_sp as well as duty_cycle_sp)

To reach somewhat further with ramping, here are a couple of ideas (of increasing complexity, but also would make ramping very much more powerful.

1) Define min and max values for duty_cycle. Most motors don't move until part way up the scale (sometimes quite a long way up!). Also they may reach maximum speed before 100% is set. Setting min and max values would mean the motor would always start immediately,and the ramp time could be spread over the actual range of change. Here is a starting from zero diagram (drawn in Libre Office Impress)

[image]

and here is what happens if the motor is already running...
[image]

2) when there are min and max values, duty_cycle_sp can be scaled so that input values are re-interpreted within the bounds defined by min and max. This would make it much easier to control multiple motors - even with the same type of motor behaviour varies. Probably scaled input values and internal calculations should use floats rather than ints to avoid banding. I've assumed a new control **duty_cycle_scaled_sp** here, but there are other ways this could be handled for the same result.
[image]

rhempel, 2016-01-04:
I'd need to think long and hard before adding more complexity or attributes to the tacho driver.

The duty cycle is a poor way to set motor speed  - especially under varying load conditions. A much better regulation of motor speed is the `speed_sp` which will encourage any motor to operate at the setpoint which is described in pulses per second. The LEGO tacho motors large and medium) are both 360 pulses per revolution.

Scaling as you describe it can be handled by your application script (I think) - the driver should provide only the bare minimum to accomplish accurate driving of the motor.

I have a "mechanism over policy" philosophy when it comes to drivers, but I am flexible if you can manage to get a good idea through my thick skull :-)

pootle, 2016-01-04:
One more option for ramping - user supplied graphs.

I may have got a bit carried away here ;)

Current ramping only allows for linear interpolation between 2 points. By allowing the user to supply a simple list of values, arbitrary motor behaviours can made really simple for a whole variety of purposes. For example to mimic a car starting up and changing gear, the movement of an old fashioned railway semaphore signal, specific movement of a robots joint, or a track or wheel driven vehicle to change speed or direction without jerks. Without getting into the complexity of using bezier curves, a simple list of up to (say) 256 values are supplied. One for ramp_up, another for ramp_down. Each graph is interpreted as an equally spaced set of time points, spread over the ramp_up / ramp_down time as appropriate. Target speeds are interpreted as positions within the list. Of course a graph is the easiest way to show this:
[image]
and the value list is made by simply reading off the values at equally spread intervals along the list.

Here's one for an old 3 speed manual gearbox car (original VW beetle for example):
[image]

Assuming the car has a max speed of around 70mph (old beetle again) then running at 30mph corresponds to about 25% along the graph. Accelerating to 50mph would mean just changing the graph_cycle_sp to about 65%

rhempel, 2016-01-04:
You are describing (in very nice graphs) what should be a userland application, not a driver function :-)

pootle, 2016-01-04:
Yes, speed regulated was in my mind as I wrote these. If run_direct (or an alternative command ) could allow dynamic update of speed_sp then all of this is practical in userland. I assume that a dynamic update of speed_sp without the need to call a **run_xxx** each time would be more efficient.

dlech, 2016-01-04:
> Yes, speed regulated was in my mind as I wrote these

With speed regulation on, we can't ramp the duty cycle output. We would have to ramp the speed pid target value.

> I assume that a dynamic update of speed_sp without the need to call a run_xxx each time would be more efficient

Ever so slightly. It saves you one file write, but that is all. Setting `speed_sp` is a memory operation, so takes virtually no time other than the file write. The `run-*` command can trigger communication with the motor controller. For the output ports of the EV3, this all happens in memory, but if you are using NxtMMX, PiStorms, WeDo or some other motor controller that uses some sort of communication bus, then the operation takes longer. This is why these are separate. These other types of motor controllers don't have `run-direct` at all, even for duty cycle. `run-direct` is intended for special high-performance applications where you write your own PID. The rest of the run commands are for typical applications where a few extra milliseconds doesn't matter.

pootle, 2016-01-04:
>With speed regulation on, we can't ramp the duty cycle output.

Sorry, yes of course, I wasn't expecting it to work using duty_cycle! 

>Ever so slightly. It saves you one file write, but that is all.

Ah! less overhead than I expected, I expect I can easily run a separate thread that can loop round 
10 - 20 times a second to keep things behaving.

rhempel, 2016-01-05:
For most applications, a separate thread that interacts directly with the motors is the way to go - you get to choose your language and the drivers stay simple (or close to it).

We will, of course, address this defect in the coming days...

dlech, 2016-04-11:
This fix has been released in the `3.16.7-ckt26-10-ev3dev-ev3` kernel (and other `10-ev3dev` kernels on other platforms). Be aware the tacho-motor class has breaking changes in this release, so don't expect motors to work with your existing programs/libraries if you upgrade to this version.

If there are still issues with ramping, please open a new issue.

closed 2016-04-11 by dlech