ev3dev/ev3dev #454: Synchronized starting of motors

dlech, 2015-12-12:
This is something that has been discussed a bit in #282.

In some applications, it is important for 2 motors to start at exactly the same time. The current thought is to create a `pairN` device node in addition to the `motorN` device nodes in `/sys/class/tacho-motor/` that has a `command` attribute that can be used to start and stop the motors.

For now, this will be limited to 2 motors because that is how most 3rd party motor controllers are implemented. In the future we may consider larger groups.

pootle, 2016-01-03:
Also once the motors start it needs to be possible to keep them in sync. It is feasible to do this by using speed regulation on the individual motors, so this class would be very small. Is this the current intention?

dlech, 2016-01-03:
It would be up to your program to set each motor to use speed regulation. To keep things simple and consistent the actual kernel driver will just synchronize the starting of the motors.

jabrena, 2016-01-09:
How to stop syncronize 2 motors?

rhempel, 2016-01-09:
This will be addressed in the next tacho motor update - proposed operation coming up shortly ...

jabrena, 2016-01-09:
Great! @rhempel & @dlech 

It is pretty interesting feature.

Cheers

jabrena, 2017-01-15:
Any upgrade with this issue?

WasabiFan, 2017-02-21:
Am I correct in assuming that this will be waiting until ev3dev-stretch?

If so, what's the recommended way of emulating this behavior with as much accuracy as possible?

dlech, 2017-02-21:
Yes, we won't be making a major change like this in ev3dev-jessie at this point. The best option is to have the `command` attributes open for both motors already and seek both to 0, then write the command to both to get them to start at close to the same time using the lowest level write function as possible (e.g. direct system call).

jabrena, 2017-03-09:
Nice news!

More or less, @dlech, when are you planning to release a early ev3dev-stretch version?

Cheers

dlech, 2017-03-13:
I'm not sure how much drivers are going to change in stretch. I suspect we will move to something that uses character devices for improved latency, so driver could be very different.

However, I do think it would be useful to hash out what it might look like in python without worrying about what the underlying drivers might look like.

WasabiFan, 2017-03-13:
OK, that's good to know; if the drivers are going to change that much, we'll be breaking things anyway :grin:

JakubVanek, 2019-12-23:
Hi,

I think I have recently discovered an issue with simultaneous ramping of two motors.

I have built a differential robot with EV3. In order to prevent slippage, I wanted to increase ramp setpoints to somewhere around 500 or 1000 milliseconds. However, the beginning of a straight movement wasn't straight - it got skewed to the side. A less pronounced example can be seen on the following video when the robot starts moving on the first step: https://photos.app.goo.gl/j22iSjbVJr9pUxf16

I think that this might be caused by a delay between writes to sysfs (possibly combined with a process switch). If a write to the `command` attribute of a second motor is delayed, the underlying ramping timer could be delayed too and this would produce the steering motion.

On the other hand, the problem might also do with the fact that I am trying to keep the robot going straight with a gyro sensor. The program is therefore issuing the run command each 100 milliseconds. This partially fixes the skew introduced, but it works only once the ramping has finished and it isn't perfect.

The program is written in C++ (available [here](https://gitlab.com/robovanek/robo2019-ss/tree/master)) and the FDs of the motor attributes are pre-opened. The critical path looks basically like this (if the compiler hasn't done some tricks):
* decode an enum to a std::string
* `lseek()` to the beginning of the left motor command FD
* `write()` the `c_str()` to the left motor command FD
* `fsync()` on the left motor command FD (probably unnecessary)
* decode an enum to a std::string
* `lseek()` to the beginning of the right motor command FD
* `write()` the `c_str()` to the right motor command FD
* `fsync()` on the right motor command FD (probably unnecessary)

From the point of view of my program, I see three solutions (given that sysfs is preserved):
1. Do not change the kernel at all. Rather, implement a motor regulator in userspace and use the `run-direct` attribute with interleaving between right and left motors. With regulator in userspace, I can configure the ramp timer with the same start value directly.
2. Add an atomic `run-direct` and optionally `position`&`speed` interface for multiple motors and implement a regulator in userspace on top of that.
3. Add an atomic (or synchronized) start to kernel and use that directly.

Option (1) is the most compatible, but it is not very reusable (i.e. it's language-specific). However I think that combining parts of (2) and (3) into a `pairN` or `groupN` sysfs device could be the way to go.
* Atomic start would require just a `command` attribute that would internally hook into the underlying motors. The attributes/parameters could be read from the motor devices.
  * This might be tricky on platforms with an external motor controller.
* Atomic query of `position` and `speed` attributes could be implemented by combining the individual values into a multiline string.
* Atomic run-direct could be implemented as multiline string writes to `duty_cycle_sp`.

On the other hand, I like the idea of having motors exposed (also) through a character device. At minimum, this way it would be possible to avoid conversions from integer to string and back. It would also be possible to implement some sort of a TLV protocol for setting multiple values in one syscall. However going that way seems to be more difficult.

Best regards,

Jakub

JakubVanek, 2019-12-24:
Hmmm, it might be enough to use Linux AIO (or io_uring in the future), with which it should be possible to group multiple FD writes into one syscall.