ev3dev/ev3dev #456: Log errors when running programs from brickman

dlech, 2015-12-13:
If a program crashes or otherwise terminates abnormally while brickman is running, there is no chance for the user to see the error message.

Idea: If a program started by brickman, pipe stderr to a file name `<program-name>.log`. If the program terminates abnormally (exit code != 0), then show a message dialog that suggests checking the log file for more information.

WasabiFan, 2015-12-13:
I suspect that for many cases, one only needs the last line or two to debug the issue; if a device isn't connected or there is a syntax error in the code, the helpful message is almost guaranteed to be in the last few lines.

Why not have a button (maybe "more details" or "expand") that shows the contents of the log file, truncated to a number of lines that easily fit on the screen? That would make it much easier to diagnose what I expect is the majority of problems. You could even open up a whole page with a scroll bar and automatically scroll to the end, with the up and down buttons corresponding to page up and page down.

WasabiFan, 2015-12-13:
Also, just piping stderr could get awkward depending on the program -- if it is a user-logged message in a script, the author might just do a normal print instead of specifying stderr. In that case, the message would get lost and the log would be pretty useless.

dlech, 2015-12-13:
> Why not have a button...

Pull requests welcome. :wink: (In other words, not a bad idea, but I'm not going to spend the time implementing it.)

> the author might just do a normal print instead of specifying stderr. In that case, the message would get lost and the log would be pretty useless.

Actually, stdout is printed on the screen of the EV3. If we pipe stdout to a file, then no one can make a text mode program.

WasabiFan, 2015-12-13:
> Pull requests welcome.

Once the core logging feature is added, I'd be happy to take a look and see if I can get that implemented myself -- from what I remember of the last time I made additions to brickman, it should be reasonably easy to figure out.

> Actually, stdout is printed on the screen of the EV3. ... no one can make a text mode program.

Hmmm... this does seem like an issue. I guess we'll just have to tell people to write to stderr if there are any messages they want the user to be able to see after the app exits.

dlech, 2017-08-31:
feature released in brickman 0.10.0

closed 2017-08-31 by dlech