ev3dev/ev3dev #458: Add pages on website to replace GitHub links

WasabiFan, 2015-12-13:
It would be great to create some pages to replace the need for "download" links which point directly to GitHub without context. I think this would mainly be:
- A "downloads" page which lists the most recent releases
- An "ev3dev on GitHub" page which lists the core repos that we have with descriptions of each and metadata on them

Some potential functionality on these pages (accessing GitHub metadata in Jekyll metadata) had been partially blocked up until recently due to a bug (jekyll/jekyll-redirect-from#89) in one of the Jekyll plugins that we use. This has been resolved, but until the updated versions trickle down to be depended upon by the `github-pages` gem, our site might be very hard to work with on local dev servers if we use this functionality.

I hope to get this done when I have time, but I figure keeping track of the work item here is best.

WasabiFan, 2015-12-16:
It would also be interesting to investigate a widget (combination of Jekyll and JavaScript, most likely) which shows an alert along the lines of "Hey, look! A new release with new features!" on the ev3dev.org homepage for a few weeks following a new release. This'll require some more thought on specifics though.

WasabiFan, 2015-12-16:
...Or, maybe it could be a "featured news article" that would also be highlighted on the news page...

dlech, 2016-01-03:
> Some potential functionality on these pages (accessing GitHub metadata in Jekyll templates)

This is also blocked by the fact that this is an organization website. The "releases" returned by the metadata will be for the `ev3dev/ev3dev.github.io` repository, not the `ev3dev/ev3dev` repository where actual releases live. We would have to move `ev3dev/ev3dev.github.io/master` to the `gh-pages` branch of the `ev3dev/ev3dev` repository to make this work.

WasabiFan, 2016-01-03:
> This is also blocked by the fact that this is an organization website.

If the organization site can't access the releases from other repos, it isn't really an issue; we can either use JavaScript or hard-code the data with JSON. But based on [their docs](https://help.github.com/articles/repository-metadata-on-github-pages/), you can get an array of objects for all the repos in the organization, and I can't tell whether or not those objects contain releases. We'll see when I get to doing this.

dlech, 2016-01-03:
[This is what you get for each repository.](https://developer.github.com/v3/repos/#list-organization-repositories) Just a `releases_url`, not individual releases.

dlech, 2016-01-20:
Looks like you patch has trickled down: https://pages.github.com/versions/

:trophy:

dlech, 2017-03-10:
The downloads page is done. Do we still need a "core repos" page? <https://github.com/ev3dev/> kind of does this already.

closed 2017-08-12 by dlech