ev3dev/ev3dev #459: ev3 to ev3 bluetooth messaging with ev3dev

rukiii, 2015-12-15:
Hi,

I am new to programming in general and lego mindstorms. I have a project in mind and would like some advice/guidance on how to proceed.

The project involves two ev3 bricks that I wish to connect with bluetooth. At the moment pressing the buttons on the first brick controls the motors/sensors on the second brick. The idea is a boardgame (similar to tic tac toe) whereby the player with the first ev3 brick can input their move with the brick buttons and play against a computer player, or in this case, the first ev3 brick. I want to keep the motion/programming simple in the second brick by using the lego mindstorm software. But want the first brick to be able to use python to calculate the best move and know when the game is over and send "text messages" to the second brick to tell it where to move for both "computer" and manual player. So far I have been able to do this using the basic lego mindstorm software but I am unable to make the "computer" player.

Does this sound possible to do with ev3dev? Are there any guidance on how to initiate bluetooth between two ev3 bricks and send messages between them?

Many thanks for any help or advice in advance.

dlech, 2015-12-15:
For connecting two EV3s with Bluetooth using ev3dev, right now, the simplest way to do this (without programming) is to enable tethering over Bluetooth on one EV3 and then connect to it from the other EV3 just as you would a host computer. Check out the [networking tutorials](http://www.ev3dev.org/docs/tutorials/).

Then in your  program, you just have to establish a network connection between the two EV3s. Before you even get into programming though, you could play around with the [netcat](http://manpages.debian.org/cgi-bin/man.cgi?query=nc&apropos=0&sektion=0&manpath=Debian+8+jessie&format=html&locale=en) command line tool to get a feel for how it works.

rukiii, 2015-12-17:
Many thanks for your advice.

When the connection is established, is there a way to send messages that can be received by the lego mindstorm software?

dlech, 2015-12-18:
If by "the LEGO MINDSTORMS software" you mean the desktop software for EV3, the answer is "not yet" (see #329).

WasabiFan, 2015-12-18:
> If by "the LEGO MINDSTORMS software" you mean the desktop software for EV3

I think the goal here is to send messages to a stock EV3 from an ev3dev EV3 so that one doesn't need to have ev3dev on both.

Although I don't have a PC with the LEGO software on it at the moment, I'm pretty sure that it has a generic Bluetooth text block; I don't know what that does exactly, but if it just accepts ASCII chars over the Bluetooth connection or something similar it should be pretty easy.

You can probably find the necessary information on EV3 bluetooth online with a simple search, although I suspect that the majority of the search results will be StackOverflow questions; @dlech is actually the one that answers a fair number of this type of question over there, so if he doesn't know how to do it I doubt it can be done.

closed 2017-07-25 by dlech

HTMLandJSVSCode, 2019-03-01:
bluetooth is the best