ev3dev/ev3dev #461: UART sensors are no longer autodetected

ddemidov, 2015-12-20:
It seems that after upgrading to the latest kernel (
```
$ uname -a
Linux ev3dev 3.16.7-ckt21-9-ev3dev-ev3 #1 PREEMPT Tue Dec 15 15:16:17 CST 2015 armv5tejl GNU/Linux
```
) UART sensors are no longer autodetected. For example, connecting EV3 infrared sensor emits the following message in `dmesg`:
```
[  512.127214] lego-port port3: Added new device 'in4:ev3-uart-host'
```
but the sensor is not listed in `/sys/class/lego-sensor/`.

Same thing happens with EV3 color sensor:
```
[  739.804522] lego-port port0: Added new device 'in1:ev3-uart-host'
```
Again, no node in `/sys/class/lego-sensor/`.

Analog sensors (tested EV3 touch sensor) and i2c sensors (tested ms-absolute-imu and pixycam adapter) work fine.

ddemidov, 2015-12-20:
Also tried rebooting with and without sensors connected -- same result.

dlech, 2015-12-20:
Should be fixed in `ev3-system 5.1.0`. I'm pushing it to the server, so please wait a minute or two, then `apt-get update; apt-get upgrade`.

ddemidov, 2015-12-20:
It works now, thanks for the quick fix!

closed 2015-12-20 by ddemidov