ev3dev/ev3dev #462: Brickman font size too large on EV3 display

WasabiFan, 2015-12-21:
Personally, I think that the new font size is too unwieldy on the EV3 display -- I admit that the previous one could get a bit hard to read, but this new one is big enough that buttons are pushed below the fold and you have to scroll to see all the options in most views. This might just be something I have to get used to, though.

Taking a brief look through the brickman code, I see that there's a method that returns the default -- could it be modified so that the font size is a setting? More simply, I think it would be great if the font could be a size in between the current and the previous; but I expect that it's this size because letters don't align well with other sizes.

dlech, 2015-12-21:
I am trying very hard to avoid settings of any kind in brickman because that means more bugs (and harder to reproduce bugs).

The current font is actually rather small on PiStorms because the display has a higher DPI than the EV3, so the choice was really a compromise between being too big on EV3 and too small on PiStorms. So, yes, I agree that it is a little bigger than I would like on the EV3, but it's what we are going to be living with for the time being. It took me several weeks of working full time to make the changes needed so that layouts worked correctly with the current font. I speculate that it would take even more time than that to come up with something that works for variable font sizes and still looks good.

On the other hand, as I already mentioned, if we want touchscreen support on PiStorms, then having an even larger font there is a must. (The touchscreen is really the only built-in input method there, so supporting it is rather important since otherwise you have to use an external keyboard/keypad of some sort.)

WasabiFan, 2015-12-21:
> I am trying very hard to avoid settings of any kind in brickman because that means more bugs (and harder to reproduce bugs).

That makes sense.

> The current font is actually rather small on PiStorms because the display has a higher DPI than the EV3, so the choice was really a compromise between being too big on EV3 and too small on PiStorms. 

Given that we currently need to support the EV3 display an the PiStorms display and we may need to support more later, at some point I suspect we'll need DPI scaling so that it adjusts the sizes for the display size. But that'll be a lot of work, so putting it off until we need it is probably the best option.

> It took me several weeks of working full time to make the changes needed so that layouts worked correctly with the current font.

Does that mean that some of those changes haven't been published yet, or are those changes not quite done? I'm seeing things like this:

[image]

Where you can't scroll down to see the buttons. Earlier I also saw a dialog box somewhere in networking with buttons that were mostly hanging off the dialog box.

dlech, 2015-12-21:
> Does that mean that some of those changes haven't been published yet, or are those changes not quite done?

I consider it "done" (that's why I closed the previous issue), so please open a new issue for each problem like this that you find.

ddemidov, 2015-12-21:
> The current font is actually rather small on PiStorms because the display
has a higher DPI than the EV3, so the choice was really a compromise
between being too big on EV3 and too small on PiStorms.

Does it make sense to select the font size in brickman based on e.g. screen
dimensions? So that larger font I'd selected for PiStorms, and smaller -
for ev3?

dlech, 2015-12-21:
It would be better to base it on dpi, but the framebuffer device doesn't provide that information. I think it would be better to try to guess the dpi from other information other than the display resolution. And actually, I was thinking that we detect if there is a touchscreen an modify more than just the font for PiStorms so the buttons are easier to press, scrollbars are way bigger, etc., but that is a separate issue.

dlech, 2018-04-20:
Closing this since font size is subjective and there is now a workaround via the `GRX_DRIVER` environment variable.

closed 2018-04-20 by dlech