ev3dev/ev3dev #464: unmet dependences

pgrudzien12, 2015-12-21:
### Problem

I have a problem with upgrading packages on my ev3. After `apt-get update` I get the following error:

```
root@ev3dev:/home# apt-get upgrade brickman
Reading package lists... Done
Building dependency tree
Reading state information... Done
Calculating upgrade... Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libev3devkit-0.1-0 : Depends: ev3devkit-data (= 0.2.1) but 0.3.0 is to be installed
E: Broken packages
```

Same goes when upgrading ev3devkit-data. I have upgraded `ev3-config` and `ev3-systemd` without problems. 

What am I missing?

### system info

```
root@ev3dev:/home# uname -rv
3.16.7-ckt16-7-ev3dev-ev3 #1 PREEMPT Sun Sep 6 22:59:18 CDT 2015
```

I used `ev3-ev3dev-jessie-2015-09-13.img` image.

dlech, 2015-12-21:
You need to run `apt-get dist-upgrade`.

`apt-get upgrade` won't remove packages and this upgrade requires the removal of one or more packages. Also, the newer brickman requires a newer kernel that will also be installed with the `dist-upgrade`.

pgrudzien12, 2015-12-21:
Ok that helps with brickman but I have network issues now. I'll try to resolve them or rise another issue. Thanks

closed 2015-12-21 by pgrudzien12