ev3dev/ev3dev #466: how to use python on ev3dev

smcri, 2015-12-27:
i want to use idle on windows(IDLE Python GUI3.4 32 bit)

dlech, 2015-12-28:
There are lot of options, but they are pretty much all broken right now. Pick one that you like and open an issue there to find out the current status.

* https://github.com/rhempel/ev3dev-lang-python
* https://github.com/EricPobot/ev3dev-lang-python-alt (probably best IDE support)
* https://github.com/in4lio/ev3dev-c/tree/master/python
* https://github.com/topikachu/python-ev3

smcri, 2015-12-28:
any idea how to upload and edit python files via ssh ?

dlech, 2015-12-28:
On Windows, I like WinSCP for copying files. Don't forget to set the executable bit before you upload though. See #438 and #439.

You can also edit files directly on the EV3 using a text editor. `nano` is pretty basic, `vi` requires that you read the manual first. Start a ssh session with PuTTY, then type `nano program.py`. Type in your program and then press CTRL+X to exit. Type `chmod +x program.py` to make your program executable.

WasabiFan, 2015-12-28:
Note: Before ctrl+x, you need ctrl+o to save in Nano.

theZiz, 2015-12-29:
But nano will also ask you, whether it should save at exit, so just do, what the editor tells you to do. :)

smcri, 2015-12-29:
is nano program.py pre installed?

rhempel, 2015-12-29:
`nano` is a text editor - `program.py` is the Python program you want to edit (or create) - you can also edit the program on your main computer and then send the file to the EV3, but it sounds like you need to get a bit of experience with the command line?

WasabiFan, 2015-12-29:
> is nano program.py pre installed?

Let's explain the terminal in basic terms to make sure we both understand what's happening.

When you run a command in the terminal, you are telling the device (in this case the EV3) to run a program and are giving it some information that tells it what to do (called "command-line arguments"). In the command
```
nano program.py
```
...there are actually two parts:
1. You are telling it to find and run a program called `nano`
2. You are telling it to give the program the information `program.py`

We know that nano is a text editor, and it comes preinstalled with ev3dev. We also know that nano accepts information that tells it what file to open. So, if we run nano and give it the file that we want (as shown above), it will open that file and allow us to edit it. If the file `program.py` exists, it will load the existing contents of the file. If it doesn't exist, it will create a new file and allow you to edit it. If you'd like some more info on using nano once you have it open, see [here](https://wiki.gentoo.org/wiki/Nano/Basics_Guide) for a reasonably good guide.

Knowing this, we are able to create and edit files on the EV3. But if we create a Python (`.py`) file, how do we run it?

```
python program.py
```

dlech, 2016-01-28:
Closing as duplicate of #492.

closed 2016-01-28 by dlech