ev3dev/ev3dev #471: OS X 10.11 does not have CDC driver?

dlech, 2015-12-27:
This was mentioned by @burf2000 in #25.

I ran into the same problems while trying to update documentation for the upcoming image release.

I can't find any relevant theads where other users have complained of this problem, but I did notice that AppleUSBCDCDriver is present in the [10.10 open source](http://www.opensource.apple.com/release/os-x-1010/) but missing in the [10.11 open source](http://www.opensource.apple.com/release/os-x-1011/).

I will file a bug with Apple and see if I can figure out what is going on.

dlech, 2015-12-28:
https://forums.developer.apple.com/thread/28541

dlech, 2015-12-28:
I just tried to build the most recent version of AppleUSBCDCDriver on 10.11.2 an it fails miserably. From the forum post, it sounds like 10.11 comes with a new USB stack, so unfortunately, it sounds like we have to wait for Apple to fix it.

I suggest that everyone effected [file a bug with Apple](https://bugreport.apple.com/) so that they get the message that this is important to us.

f-steinbauer, 2016-01-04:
Just a side note, using ev3-ev3dev-jessie-2015-12-30.img in combination with OS X 10.11.2 the CDC connection is working. At least on my system.

dlech, 2016-01-04:
Did you already have something setup before you upgraded? Can you post a screenshot of what you have or explain how you set it up?

I'm seeing the same thing shown [here](https://github.com/ev3dev/ev3dev/issues/25#issuecomment-167441753) that there is no "CDC Composite Gadget" when I try to add a device.

f-steinbauer, 2016-01-04:
once back from vacation, I will get another SD card to more easily switch systems and make some screen shots.
For now, I can say, I did no fancy setup. Just booted the mentioned image pluged in the USB cable and OS X provided me the "CDC Composite Gadget" in the drop down within the new network setup

251, 2016-01-08:
CDC works for me on 10.11.1. I can connect to ev3-ev3dev-jessie-2015-09-13 after loading the OS X defaults in the IPv4 menu, but not to ev3-ev3dev-jessie-2015-12-30 (not detected). 

[image]

[image]

dlech, 2016-01-08:
Thanks for the feedback. This makes it sound like we might be able to do something on the ev3dev side after all.

251, 2016-01-20:
Maybe it was introduced with 1af7d53447d9823a9bfac616231f8602fea49e30. When I disable `ev3-usb.service`, reboot and `modprobe g_cdc` the CDC Gadget pops up in the network settings.

dlech, 2016-01-20:
The new `ev3-usb.service` works on older versions of OS X and Apple confirmed that my bug report was a duplicate of another bug, but hasn't been otherwise forthcoming with details. The advantage of the new service is that it works on Windows 10 without any massive workarounds and works on Linux and (supposed to work on) OS X without having to choose between the two. If Apple doesn't come up with a fix soon though, we may have to do something. The workaround you have found is good in the meantime.

andrewwjensen, 2016-02-06:
The workaround helped me, but it took some digging to figure out the exact steps to get it to work. In case this helps someone else, here is what I did.

1. Create the cdc-gadget service:
 * sudo vi /lib/systemd/system/cdc-gadget.service 
 * Copy and paste the content of the file from below.
 * Note: if you don't know how to use 'vi', press 'i' to begin insert mode, then paste the content, then press ESC, type ":wq" (without the quotes) and hit RETURN.
1. sudo systemctl stop ev3-usb
1. sudo systemctl disable ev3-usb
1. sudo systemctl enable cdc-gadget
1. sudo systemctl start cdc-gadget

File contents of /lib/systemd/system/cdc-gadget.service:
```
# This file is part of the ev3-systemd package

[Unit]
Description=CDC Ethernet USB gadget service
Conflicts=rndis-gadget.service

[Service]
Type=oneshot
RemainAfterExit=yes
ExecStart=/sbin/modprobe g_cdc
ExecStop=/sbin/rmmod g_cdc

[Install]
WantedBy=network.target
```

dlech, 2016-02-07:
I've added a "better" workaround in `ev3-system 5.2.0`. (What makes it better is that A. it uses persistent MAC addresses and B. you won't have to connect via Bluetooth or something else first using the next image release).

For existing installs, here is what you need to do:

    sudo apt-get update
    sudo apt-get upgrade
    echo "cdc_only=true" | sudo tee /boot/flash/ev3dev.txt

Also undo any of the above workaround it you tried them.

thbar, 2016-07-30:
Another data point - I'm on 10.11.4 and followed the tutorial [here](http://www.ev3dev.org/docs/tutorials/connecting-to-the-internet-via-usb/), and CDC composite gadget choice does not appear after connecting via mini USB.

dlech, 2016-07-30:
The `ev3dev.txt` file I mentioned should be available in the latest nightly builds.

 [ [image] ](https://bintray.com/ev3dev/nightly/ev3dev-jessie/_latestVersion)

dlech, 2016-07-30:
Also file a bug with apple. They will probably close it as a duplicate, but it will let them know that this is a common issue that should be fixed.

huba71, 2016-08-28:
I had the same issue: 
No CDC Composite Gadget available in the network device list (macosx 10.11.6) and I can not add one.

I have done an upgrade and changes mentioned in a previous post (connecting via bluetooth):
sudo apt-get update
sudo apt-get dist-upgrade # upgrade is not enough because of the "kept-back packages"
echo "cdc_only=true" | sudo tee /boot/flash/ev3dev.txt

After rebooting the brick a new network device "EV3-ev3dev" appeared in the network device list of the mac.
Following the instructions on ev3dev.org site to connect via USB but using this device instead of the "CDC Composite Gadget" I can successfully connect to the device (and connect to the internet).

im-bart, 2016-09-27:
Same on (now macOS) 10.12, follow instructions as per the post from @huba71, and this will work just fine. Make sure you put some charge in the brick before doing so though :-)

dlech, 2017-02-08:
WooHoo, I finally found something that will work for Window and macOS at the same time without a config file! https://github.com/ev3dev/ev3-systemd/commit/9ab448d6501c625bb2b72a0b65f8c33aba1809d4

Fix released in `ev3-systemd_5.4.0`. After you upgrade to this version, you should be able to delete `ev3dev.txt` and USB will still work with macOS.

closed 2017-02-08 by dlech