ev3dev/ev3dev #472: Opportunity to simplify Windows 10 USB instructions

WasabiFan, 2015-12-28:
You can simplify the instructions for Windows 10 USB configuration (remove one step) by having the user paste the full URL of the INF file from the web into the file selector box instead of pre-downloading it. The Windows file browser dialog automatically downloads web links into the AppData folder, so this works without any extra work on the user's end.

I decided not to open a PR (at least not yet) because @dlech has said he is working on docs at the moment for the next image release and I don't want to cause merge conflicts for this fairly small change. I can make the changes myself and open a PR if desired.

Here are the updated versions of the images that would be replaced:
[image]
[image]

dlech, 2015-12-28:
Window 10 (and 7/8) should work "out of the box" without installing drivers using the upcoming image release, so this should not be needed.

WasabiFan, 2015-12-28:
That's why I checked beforehand :wink: Thanks for the info!

closed 2015-12-28 by WasabiFan