ev3dev/ev3dev #473: What is maximum value for the attribute: duty_cycle_sp in a DC Motor?

jabrena, 2015-12-28:
Hi,

I was testing DC Motors but I discovered that some values return the following error.

``` bash
root@ev3dev:/sys/class/dc-motor/motor8# echo 50 > /sys/class/dc-motor/motor8/duty_cycle_sp
root@ev3dev:/sys/class/dc-motor/motor8# echo 300 > /sys/class/dc-motor/motor8/duty_cycle_sp
-bash: echo: write error: Invalid argument
```

What is the limit for the property: "duty_cycle_sp"?

I was reading the info about Motors but I didn't find anything:
http://www.ev3dev.org/docs/motors/

BertLindeman, 2015-12-28:
Look at http://www.ev3dev.org/docs/drivers/dc-motor-class/

```
Valid values are -100 to 100 (-100% to 100%).
```

jabrena, 2015-12-28:
Hi Bert!!

Many thanks, I will update my notes. :D 

Cheers

closed 2015-12-28 by jabrena