ev3dev/ev3dev #475: How to detect a EV3 Brick / Brick Pi+ or PiStorm?

jabrena, 2015-12-28:
Hi,

I was reading previous issues (https://github.com/ev3dev/ev3dev/issues/412), but I didn't understand the right way to know if EV3Dev is running on a EV3 Brick or another board for RaspberryPi (BrickPi+ or PiStorm)

@rhempel suggested these ways:

``` bash
/sys/class/ev3dev/ev3-mono
/sys/class/ev3dev/ev3-pistorms
/sys/class/ev3dev/ev3-brickpi
``` 

But I didn't understand the final way.
Is this feature implemented?

Cheers

dlech, 2015-12-28:
The consensus is that your code should not care what hardware it is running on. Instead, you should detect the individual components. See <https://github.com/ev3dev/ev3dev/issues/412#issuecomment-154247298>.

If you want to display the "platform" for informational purposes, the <https://github.com/ev3dev/ev3dev/issues/412#issuecomment-153570798> would work. The PiStorms example probably isn't the best though.

jabrena, 2015-12-28:
Hi @dlech,

Merry Christmas!

and the ports, do maintain the same names?

**Motors:**

``` java
	public static final String A = "outA";
	public static final String B = "outB";
	public static final String C = "outC";
	public static final String D = "outD";
```

**Sensors:**

``` java
    public static final String S1 = "in1";
    public static final String S2 = "in2";
    public static final String S3 = "in3";
    public static final String S4 = "in4";
```

Many thanks in advance.

Cheers

dlech, 2015-12-28:
Everything you need to know about input and output ports is at http://www.ev3dev.org/docs/ports/.

In particular...
* http://www.ev3dev.org/docs/ports/pistorms-in-port/#fn:address-prefix
* http://www.ev3dev.org/docs/ports/pistorms-out-port/#fn:prefix

dlech, 2016-01-28:
See new discussion in #523 and #524.

closed 2016-01-28 by dlech