ev3dev/ev3dev #476: How to detect a RCX Touch Sensor?

jabrena, 2015-12-28:
Hi,

I was trying to connect a RCX Touch sensor using a RCX-NXT cable to a EV3 Brick in port4.
http://alpha.bricklink.com/pages/clone/catalogitem.page?P=879#T=C

The sensor has the following modes:

``` bash
root@ev3dev:/sys/class/lego-port/port3# cat address
in4
root@ev3dev:/sys/class/lego-port/port3# cat modes
auto nxt-analog nxt-color nxt-i2c ev3-analog ev3-uart other-uart raw
root@ev3dev:/sys/class/lego-port/port3# ls
address  device  driver_name  mode  modes  power  set_device  status  subsystem  uevent
root@ev3dev:/sys/class/lego-port/port3# cat mode
auto
```

But i set the raw mode:

``` bash
root@ev3dev:/sys/class/lego-port/port3# echo raw > /sys/class/lego-port/port3/mode
````

but later, I didn't find some place to get the "value0" from the sensor.

Do we have any support for this sensor?

Cheers

dlech, 2015-12-28:
RCX sensors are currently not supported. Presumably you are using the [LEGO 8528 cable](http://www.bricklink.com/catalogItem.asp?S=8528-1). This cable uses pins 1 and 2 on the input port. On the NXT, pin 2 was GND, but on the EV3 it is not, so this cable will not work with the EV3.

Theoretically, you could make your own cable that connects to pin 3 instead of pin 2. Then you could use RCX sensor. But there would be some work needed for drivers first.

jabrena, 2015-12-28:
Yes, I am using that cable.

It is a pity, but I understood. If the standard cable is not ok for EV3 then this issue is closed.

dlech, 2015-12-28:
Let's leave this open in case anyone else is looking to use RCX sensors. I changed the title to be more generic.

jabrena, 2015-12-28:
Oki :)