ev3dev/ev3dev #477: Simplify process of writing and transferring code for new users (and advanced ones too)

WasabiFan, 2015-12-29:
I have seen a few issues posted recently by new users attempting to figure out how to get code onto their ev3dev device and run it. It occurs to me that we don't actually have _any_ information as far as I can remember that guides or informs users on this process. Let's see if we can fix that.

I see three major options (and they aren't mutually exclusive):

- Add tutorial(s) to explain the options/steps of writing code and getting it to its destination. I'd say we would want to document SCPing files, editing them from SSH (with Nano), and maybe whatever the most conventional methods are for using *NIX/OS X if they're reasonably simple. It would also be useful to have some language-specific ones (e.g. "Getting started with basic Python on ev3dev")
- Create and/or preconfigure and document some software to simplify transfer of files. This could be a script to easily autodetect and SCP files, a GUI client app to do the same, or even a simple web app that can be enabled from Brickman.
- Support more "cloud IDEs". As I understand it, we already have an Open Roberta server (which is a simple GUI-based drag-and-drop application) controller implemented in brickman. If we could find a web IDE to support that works with traditional source code (I think Python is pretty important), that would make it much easier for people who just want to flash an SD card and start coding.
- Document and/or develop integration with existing IDEs. If we had documentation for deploying directly from existing IDEs onto ev3dev-based devices, users could write code without having to interrupt their normal workflow. This is the type of thing that I could see going with individual libraries instead of core ev3dev, but putting that work on library maintainers and possibly duplicating efforts seems like it would be an issue.

I think the first one is definitely important to do regardless of the other options, and the others are more complex and would require significant work. Are there any other options that we should be considering? What do we want to actually attempt in the short-term?

laurensvalk, 2015-12-29:
> Support more "cloud IDEs" 

What about [this] (https://www.youtube.com/watch?v=vJHr6vIZk0k)? I was able to get it to run, though not all smoothly. However, I'm just a beginner with these type of things. Recently the web IDE project was renamed [Jupyter] (http://jupyter.org/), and I haven't tried it since.

It actually also has a terminal accessible from the web interface. So if it is possible to run this smoothly, and possible to start from brickman, not a single tool would have to be installed on either the ev3 or the PC side.

We could also preload it with example projects for a simple robot, like the one in the video.

> Document and/or develop integration with existing IDEs.

An ev3dev plugin (Either Python, C/C++, etc) for Eclipse would be very nice.

For a project at university I am now using [some custom programming language] (http://ii.tudelft.nl/trac/goal/wiki/Releases) based on Prolog. The developers provide all the necessary tools through a plugin for Eclipse. All I had to do as an Eclipse user was go to "Additional software", list their domain, and click install. It would download all the necessary tools (such as swi-prolog in this case), preconfigure a programming and debugging workspace, preload examples, terminals, and so on.

This is about as easy as it gets for the user. Even platform dependency could be handled by the plugin. Writing such a plugin would probably be a lot of work, but it's probably still less work than developing a new IDE.

dlech, 2015-12-29:
Related: #44, #85, #432

WasabiFan, 2015-12-29:
> What about this?

That was one of the past works I had in mind -- I think that would be great to have working. It's unfortunate that it was slow on the EV3... I expect that will be the case with many server-based apps.

> not a single tool would have to be installed on either the ev3 or the PC side.

That would be pretty great for new users!

> An ev3dev plugin (Either Python, C/C++, etc) for Eclipse would be very nice.

Although I agree, I think we'll have to put some serious thought into what platforms (IDEs) we want to support. There are _many_ IDE and language combinations that we would need to consider, and I would like to make sure that we use our time effectively. For example, yes, Eclipse is a good IDE, but it is a 400Mb download and requires a JDK to be installed. It also is aimed at knowledgeable developers, so it can be pretty fiddly at times and might not be easy to use for new users.

> Writing such a plugin would probably be a lot of work, but it's probably still less work than developing a new IDE.

I agree -- and chances are that the work we would need to put into it would be mostly just tying together preexisting functionality. But we need to make sure that we aren't duplicating work if there are already packages for embedded development that do what we want.

rhempel, 2015-12-29:
Another option might be to point users to editing tools that can access files over an IP connection - I'm thinking of `vi` but there may be others.

Also consider updating our original tutorials for setting up shared folders on remote machines, allowing the user to edit the files from the host machine and run them via `ssh` on the command line.

If possible, I'd like to avoid having to set up multiple IDEs for the little languages that we will support - maybe we can come to a compromise using file sharing that lets the EV3 be the "cloud" - maybe a minicloud?

WasabiFan, 2015-12-29:
> I'm thinking of  vi  but there may be others.

We're looking to simplify the process for new users, not scare them off :wink: But yeah, I see the point and I agree... I just think that teaching them `vi(m)` could be a problematic way of doing that.

> Also consider updating our original tutorials for setting up shared folders on remote machines, allowing the user to edit the files from the host machine and run them via  ssh  on the command line.

Like the NFS option mentioned in #44? I think it would be great to get that ported forward for our current docs for OSX/Linux, but I still maintain that it isn't a good option for most Windows users. Although there may be better NFS apps available now.

> If possible, I'd like to avoid having to set up multiple IDEs for the little languages that we will support - maybe we can come to a compromise using file sharing that lets the EV3 be the "cloud" - maybe a minicloud?

Mainly, I want to make sure that we have first-class Python support, as that is (based on my empirical and unscientific data) the most common language that people want to use. Other languages would be great to have full support for, but really any solution that gets the files where they need to be is "good enough" as I see it.

To be clear, when I say "cloud IDE" I am imagining something served on the EV3 in most cases. It would be optimal as I see it if we could get ev3dev set up with the option for users to do all their development without any installation -- from what I have heard [this product](https://store.particle.io/) does something like that.

jabrena, 2015-12-29:
Hi,

Take a look this project:
https://github.com/baseapp/liveide

Juan Antonio

laurensvalk, 2015-12-29:
> Another option might be to point users to editing tools that can access files over an IP connection - I'm thinking of vi but there may be others.

Notepad++  with an sftp plugin could be the beginner version of that. Dexter Industries are [using this for the BrickPi here] (http://www.dexterindustries.com/BrickPi/program-it/hints-and-tricks/).

> (...) we'll have to put some serious thought into what platforms (IDEs) we want to support. (...) Eclipse is a good IDE, but it is a 400Mb download and requires a JDK to be installed. It also is aimed at knowledgeable developers, so it can be pretty fiddly at times and might not be easy to use for new users.

Agreed. I just happened to know Eclipse had this option, but there may be better lightweight alternatives. At 600Mb and pretty slow performance, it should be easy to beat the original graphical software on this aspect :)

@antonvh - Maybe you can explain what you are using for ev3dev development?

WasabiFan, 2015-12-29:
> there may be better lightweight alternatives

I expect there are many options in this space -- the first one that comes to mind for me is [Visual Studio Code](https://code.visualstudio.com/), for example, because I use it for most of my ev3dev development and configure my projects to deploy and run there. Editors like that are only ~50Mb.

I imagine that a fair portion of incoming users are people who use EV3-G (or whatever it's called now) and want to be able to write actual code instead, so I think that is a good yardstick to use -- if our solution is easier to get started with than EV3-G, I'd count it as a success.

I think a major problem that we are seeing is that everyone working on ev3dev is a developer who knows the platform intimately and doesn't use much pre-made software for it. We aren't (in most cases) reflective of the typical user that we should be targeting. So it would be great to get some insights from someone who has recently attempted to get started with ev3dev.

laurensvalk, 2015-12-29:
I think most users moving away from the graphical software are new to *both* Linux and text-based programming, which makes the transition harder.

Even supporting just one language in one light IDE (and hiding any Linux related affairs) would make the transition easier. Brixcc used to provide this for NXT, and RobotC (non-free) has it for NXT and EV3.

Actually, it looks like Brixcc is open source...

> We aren't (in most cases) reflective of the typical user that we should be targeting. 

I run a [blog] (http://robotsquare.com/) with mostly entry level EV3 readers. I occasionally speak readers of my book about this issue as well. Maybe once we have some options to consider, I can run a poll to see what people think, and ask what they might want to use ev3dev for.

dlech, 2015-12-29:
:+1: for visual studio code. It's cross platform (Linux/OS X/Windows), can do code completion in some languages and is extensible.

rhempel, 2015-12-30:
Hey, that Visual Studio Code looks interesting - kind of like a minimal Eclipse. Notepad++ is what I suggest people in my company use if I ever catch them editing code in Notepad (yes, it happens more than you think). Personally I use `gvim` but that's just "baby duck" syndrome at work - it was the first editor I used on the Unix system at the University of Waterloo so it imprinted into my fingers.
Vim has the ability to do remote file edits - but it can send otherwise sane humans screaming for the exits.
I suggest we do an evaluation of a few popular IDEs that are cross platform, have reasonable traction, and allow editing files on a remote system.
@laurensvalk - I appreciate the offer to run a poll on your blog. When I started the ev3dev project, the idea was to give folks that were somewhat familiar with the command line and text based programming an alternative to EV3-G - if there is another class of users that we can help get into programming - that would be cool!

WasabiFan, 2015-12-30:
> Hey, that Visual Studio Code looks interesting - kind of like a minimal Eclipse.

Sort of.. it's more like Notepad++, but prettier, and with all the legacy stuff replaced with IntelliSense, debugging, and other IDE features. And it's cross-platform.

> Notepad++ is what I suggest people in my company use if I ever catch them editing code in Notepad

What programming language is that? Personally, I use N++ for data files, random code I download from the internet, and running advanced find-and-replace -- but as an IDE it's pretty lackluster. 

> I suggest we do an evaluation of a few popular IDEs that are cross platform, have reasonable traction, and allow editing files on a remote system.

I think that's what we will probably do eventually, but first we need to understand what we're looking for: do we want to find something that just supports remote editing, and document usage with that on ev3dev? Or find an extensible platform and develop/find functionality that makes it work well for embedded development? Maybe write our own plugin which auto-detects EV3s and interfaces with brickman/other ev3ev systems? We could even make a device browser similar to what's in Brickman available in the IDE for quick access and testing/debugging.

So, if we want to investigate finding a desktop IDE to then build upon, document, and promote, we need specific criteria and points of comparison. Or, if instead we want to look for web IDEs that we wan host, what do we want there? And where can we look to find one?

> the idea was to give folks that were somewhat familiar with the command line and text based programming an alternative to EV3-G ... if there is another class of users that we can help get into programming ...

Although I suspect that the "pro dev that wants to use the EV3" category is definitely one that we currently support and should continue to (we all fit into that category as I see it), ev3dev has large potential as expansion space for people who want to learn programming. Writing a "hello world" app that prints text to the console is pretty boring; if people can instead make their little EV3 robot drive forward for a second, for example, that is much more interesting. But we need to lower the bar to entry before that can happen.

I see ev3dev being used in classroom-type settings to bridge the gap between drag-and-drop languages and text-based languages. Imagine you learn EV3-G, then translate a simple graphical program into Python (for example) and see how it correlates. This is an easy way to bring the digital and abstract world of programming into the real world, and it has a much larger capability for allowing expansion into complex programming concepts than the graphical language.

And once you understand the basics, you can start adding cameras and complex algorithms and PC interfaces. That's the area that our current userbase occupies.

rhempel, 2015-12-30:
>> Notepad++ is what I suggest people in my company use if I ever catch them editing code in Notepad

>What programming language is that? Personally, I use N++ for data files, random code I download from the internet, and running advanced find-and-replace -- but as an IDE it's pretty lackluster.

This is for basic C programming. It's not as good as a well polished IDE or customized `vim/emacs` setup in good hands, but it's far, far better than the default Windows editor. I kid you not, I see this more often than I'd like from some of our remote team members.

My original post on making hacking more accessible [http://hempeldesigngroup.com/embedded/2013/09/25/making-hacking-accessible-lego-mindstorms-ev3/] has many of the same points that you make - and to see how far the ev3dev project has come over the past two years has been a bit humbling. There is no way that I could have done any of this by myself - not even full time.

I am grateful for the work and dedication that has been demonstrated by the key contributors to this project, and the short bursts of intense energy applied by the infrequent but still important occasional contributors.

If we can get ev3dev to a point where users can burn an image to a microSD card, and install the same basic IDE on Windows/Mac/Linux that allows remote editing of files, I'll be a happy fellow!

WasabiFan, 2016-01-02:
I'd like to propose the following course of action: we add an alert at the end of our "Getting Started" guide that says something along the lines of the following:

> We'd love to hear what you have to say about your experience getting started with ev3dev. Please go [wherever we want them to go] and share your thoughts!

And direct them to either an open issue or some other location where we have some questions for them. Those would ask about finding the getting started guide, actually completing all the steps, finding subsequent tutorials (networking, for example) and then writing code. That way, we can get a good idea of the experiences of a new user to see where the "pain points" are, as well as get ideas on how we could improve it. I suspect that the major issues will be writing code and finding next steps/tutorials, but I would prefer to get this directly from users instead of guessing.

Based on the number of issues that I see posted on ev3dev-related repos from new users, I think we _do_ have enough volume to get some responses if we do it right. It might take a few weeks (or longer) to get enough input, but I believe that it is worth it: I see some fairly major holes in the getting started "flow" (mentioned earlier) that we really should fix.

@laurensvalk had offered to run a poll on his blog to get a better understanding about the ideas discussed earlier. I think we should get the most out of that offer by waiting until we get a good idea of the things that need to be solved before asking for feedback on specific ideas.

So, what do you guys think?

laurensvalk, 2016-01-03:
Considering a poll, there is also the [LEGO MINDSTORMS EV3 Facebook page] (https://www.facebook.com/groups/legomindstorms/), which likely has even more beginning users. Due to the nature of Facebook, it may also generate more feedback. 

> (...) waiting until we get a good idea of the things that need to be solved before asking for feedback on specific ideas

Agreed.

laurensvalk, 2016-01-03:
I just tried out the latest ev3dev release, and I was impressed by how easy it was to get started, use brickman to set up a connection, and make a program with Open Roberta.

If we manage to find something equivalent (a web IDE which also provides a terminal) for a language like Python, it should also be pretty easy to get started with the more advanced stuff.

rhempel, 2016-01-03:
Thanks Laurens - the credit goes (again) to David Lechner for his tireless work on the nuts and bolts and also to the many users that took the time to enter defects and work with us to make ev3dev better.

I've been working on getting the Python binding docs cleaned up .

laurensvalk, 2016-01-04:
Adafruit has a web IDE for Raspberry Pi and Beaglebone: [install] (https://learn.adafruit.com/webide/installation) --- [github] (https://github.com/adafruit/Adafruit-WebIDE) --- [demo] (https://www.youtube.com/watch?v=8NoiBBgaKCI)

A possible downside I see is that it requires an account to save code (although cloud storage could also be a convenience). I don't know if the EV3 will be fast enough to run this, but it looks promising. Also has a terminal option.

WasabiFan, 2016-01-04:
That project looks promising! I think the first step is simply to see if it runs on the EV3, and then try it out and see if it runs _well_ with no slow-downs.

As for the account issue, it is open-source; if we want to add a run option to save locally and anonymously, we should be able to.

dlech, 2016-01-04:
Since the server would actually be running on the EV3, I expect it to be a similar experience to iPython in terms of speed. It's worth a try though. You should be able to just install the .deb package and see what happens.

laurensvalk, 2016-01-04:
> You should be able to just install

I tried following the steps on their installation page, but I'm stuck at "npm install" (outputs "killed", or the brick hangs). Will give it another go later and look for the .deb package.

moriarty, 2016-01-05:
:+1: #490 is another example of how this question often comes up in the issues. 

I assume the information on this old wiki page still working? https://github.com/ev3dev/ev3dev/wiki/Set-Up-An-nfs-FileShare.

Would it be beneficial to at least link to or copy the old wiki page until a new solution is found (maybe with a note that easier methods are being discussed)? The options you've mentioned in this issue with IDEs seem very heavy and they could take a while to discuss, perfect and document, compared to ```scp``` or ```nfs``` and ```ssh```. Or is there a reason this page isn't being linked to from the site?

WasabiFan, 2016-01-05:
I think that we need to, as a priority work item, get some docs on transferring files written. Can we decide on the systems that we want to document? I'd vote for `scp`/WinSCP as a start.

dlech, 2016-01-05:
> I'd vote for scp/WinSCP as a start.

:+1: NFS is a lot of work to setup, which is why we have let it fall by the wayside. SSH is already working with no setup required. I just created #492 for this. Any volunteers?

rhempel, 2016-01-05:
Agreed - `nfs` is a pain - and I *like* `nfs`. Has anyone tried that Microsoft Code Studio yet?

laurensvalk, 2016-01-05:
When I got started with Linux, I found it rather difficult to navigate around the system, and especially to use SCP. So I used Filezilla. It's an sftp client available for Linux, Mac, Windows.

You can drag and drop files between your local system and the EV3. More importantly, you can see the file system of the EV3. At the time, this made things much easier for me, since I could actually see what the Linux file system tree looked like. Perhaps trivial now, it was not so trivial coming from a Windows background, and the *ls* command just doesn't give you that overview. 

That also came in handy for making backups of files. (Just dragging a folder to the left hand side.) You can optionally directly open a remote file in a text editor. When you try to save the file, it will ask you to whether you want to save the changes to the remote system. Handy when you want to make a small change but haven't mastered vi or nano yet.

I'm not saying we should be using this, but this is my perspective :)

[image]

WasabiFan, 2016-01-05:
> Has anyone tried that Microsoft Code Studio yet?

You mean Visual Studio Code? Yeah, I use it for the majority of my code editing -- ev3dev-lang, ev3dev-lang-js, ev3dev-lang-fake-sys, ev3dev.github.io, etc. I think it's great!

> I just created #492 for this. Any volunteers?

I might, but won't have time today and would prefer to get some of my other PRs finished before starting anything new.

rhempel, 2016-01-05:
Laurens, I use FileZilla all the time, but not for EV3 - nice because it's cross platform. I feel a bit dumb asking this, but do we have `ftp`/`sftp` enabled by default on the EV3?

dlech, 2016-01-05:
We don't have ftp enabled and sftp uses ssh. (ftps is to ftp as https is to http, ftps != sftp)

laurensvalk, 2016-01-05:
Yes. No change on the EV3 side required. 

In Filezilla:

[image]

rhempel, 2016-01-05:
Oh - that is nice indeed !

rhempel, 2016-01-05:
So forget the Microsoft Studio Code - go with whatever native editor you like and use FileZilla to copy the files over. If you can set the executable bit and have a nice shebang line, then you can run right from BrickMan!

laurensvalk, 2016-01-08:
As mentioned, Jupyter (Ipython Notebook) runs rather slowly because the server itself runs on the EV3.

There is also [Jupyter Hub] (https://github.com/jupyter/jupyterhub). If I understand correctly, this is still not exactly what we need, but maybe a step in the right direction. It looks like it runs both the server and the python code remotely (something you might use on a [supercomputer] (http://zonca.github.io/2015/04/jupyterhub-hpc.html)).

But we need the server and IDE stuff to run remotely (on the user's PC, or on ev3dev.org), and the python code locally on the EV3, right? I wonder if Jupyter Hub could do that also.

WasabiFan, 2016-01-08:
> But we need the server and IDE stuff to run remotely (on the user's PC, or on ev3dev.org), and the python code locally on the EV3, right?

ddemidov, 2016-01-08:
With [RPyC](https://rpyc.readthedocs.org/en/latest/) its possible to run a lightweight process on ev3, and then connect to it from a host computer and run python scripts remotely. There is an example in my old README: https://github.com/ddemidov/ev3dev-lang-python/tree/master-deprecated#remote-operation.

antonvh, 2016-01-08:
A bit late to the table but my 2 cts:
- I am working on a tutorial on how to set up a development environment using PyCharm Community Edition and Git (my favorite tools). I'll soon submit it as pull request to the docs. When I code I check in my code on github and then do a pull request and run on the robot. This has the nice side effect that code is versioned. I have little scripts/macro's that make the process real easy. 
- I will experiment tonight with automatic deployment. You can setup the ev3dev machine as a git remote and run a post-receive script that runs your code when you push to it. This is a little harder to setup than the above and you won't have ctrl-c as an abort button.
- For simple scripts or starters I like Juypiter/iPyhton notebook as @laurensvalk suggests. It has no code completion that I know of and seems to run rather slow for some people. Some sample notebooks would make this really accessible for newcomers.

dlech, 2016-01-08:
I had a look at PyCharm a couple weeks ago. It looks like remote execution and debugging is not available in the community edition. I hope you can find a way around this.

ddemidov, 2016-01-08:
Re Jupiter Hub: Jupiter Hub still needs to spawn a single-user ipython notebook server per compute node. In our case that means it will need to run ipython server on the EV3.

johnagr, 2016-01-10:
How about using atom. It has several plug-ins,  including sftp.

rhempel, 2016-01-10:
@johnagr - Can you please provide links so we're talking about the same thing? For me, atom is a pseudo RSS format :-)

johnagr, 2016-01-10:
Sorry. https://atom.io/ It's cross platform,  support all the ev3dev languages,(code syntax,  debug,  code completion,  etc) it is  highly extensible. Maybe it is possible to create a common plug in for remote file sync. Right now it has scp plug-in too.

rhempel, 2016-01-10:
Cool - I think that it's probably useful to add this to a list of options. @johnagr - have you actually used this to send a file to an ev3dev device?

I'm cautious of adding "all-in-one" type apps - some programmers (myself included) suffer from what I call "baby duck syndrome". The first editor you ever use it the one you use forever :-) So you'll pry `gvim` from my cold, dead, fingers before I give it up.

The preference is for something like FileZilla to transfer the files because it does not tie you to an editor - but for some users, atom.io will be a good alternative.

@johnagr - please give atom a try and report back on how well it works for you.

thski, 2016-01-10:
Only want to assure you didn't miss jabrena comment 12 days ago:

>Hi,
>
>Take a look this project:
>https://github.com/baseapp/liveide
>
>Juan Antonio
[image]

Perhaps this editor is more simple and thus faster and not so overwhelming?

WasabiFan, 2016-04-08:
On the side of "extremely new users," I think I have something that might be of interest.

Recently, I discovered a project called ["Blockly,"](https://github.com/google/blockly) an actively-maintained graphical Scratch-like editor that runs in the browser. This is similar to other projects that we've seen, except for two key differences: it is build to be embedded into other apps and it generates code in the language of your choosing instead of running it directly. So, I figured I'd try my hand at writing an ev3dev server and blocks for it -- they support Python generation, so it was a good opportunity to go deeper into Python than I had before.

I wrote two blocks, a "connect to motor" block and a "start motor" block. I then wrote a simple server that serves the Blockly files and responds to POST requests at a specific URL -- when you POST with a file name and code, it saves the file, runs it, and then returns the output. I also rigged a beautifully-designed button and div so that one could create a program in the web interface and then click a button to run the code. It's obviously a pretty hideous thing, so consider it a technical demo.

In the end, I did this mainly to get some experience with Python, but I was wondering if anyone thought it useful enough to get it fully configured for actual use. In theory, one could load up the web site, build an app, see the Python code being generated, and then run it remotely to see what it does on their EV3.The interesting part as I see it is that the generated code is written just as a human would natively write it, so you can take the generated Python and use it directly if you outgrow the graphical editor.

Anyway, is anyone interested in this? I dumped what I have into a fork of the Blockly repo at [WasabiFan/blockly-ev3dev](https://github.com/WasabiFan/blockly-ev3dev). If you'd like to test it, clone it onto your brick and then run the `server/server.py` file. It's a big repo because it has all the source files, so if that takes too long you can clone it on a desktop PC and just copy over [these files](https://gist.github.com/WasabiFan/0f4546123b25d66b05cd681a1e997022).

--------

As for the "Real IDE in a browser" discussion, should I go and investigate the options that have been brought up? Or is there something else that would be more helpful on the bootstrapping new users front? I think the discussion we have had here so far is definitely on the right track and would like to make sure that it doesn't get lost.

dlech, 2016-04-09:
FYI, OpenRoberta's NEPO is based on Blockly. http://lab.open-roberta.org

WasabiFan, 2016-04-09:
Interesting! I'm going to take a look at the source code for our Open Roberta server to see how that works.

psychemedia, 2016-05-20:
@laurensvalk I've just tried Filezilla on a mac and whilst I can connect and see files on the ev3, and pull files from the EV3 to the Mac, I can't go the other way, instead getting a  "Could not start transfer"? 

Poking around the brick, it's showing `/home/robot` as:

`drwxr-xr-x  9 robot robot 4096 May 19 22:20 robot`

but also as `read-only file system`. Any hints as to how it could have got into this state and how to resolve it?

Only thing I think I've done is add git to try out the pycharm recipe for getting files from pycharm to the ev3 via git commits?

WasabiFan, 2016-05-20:
@psychemedia In the future, please open a new issue for problems that you face instead of commenting on unrelated ones. It makes discussion easier to track.

psychemedia, 2016-05-20:
@WasabiFan sorry, I was picking up on a point earlier in this thread about Filezilla.

antonvh, 2016-05-23:
I have two tutorials on ev3dev.org on how to transfer and edit files. One uses nfs, the other uses git. NFS is nice for quick parameter tuning. Git is nice because you can make the ev3 run your code automatically after you push it. No need for open terminal windows, where you press arrow-up and enter all the time. But you need a hardware stop in this case, as you can't ctrl-c out of your code.

laurensvalk, 2016-05-24:
Here's something in between NFS and Filezilla, tested on windows and Linux, with files directly accessible from the file browser without any command line stuff on either PC or EV3:

As usual, in Windows a few things need to be installed first (SSHFS). There's a GUI to set up the connection, which will mount it as a drive in File Explorer. After that, just use any text editor. And you can launch your code from Brickman or Putty.

[image]

For Ubuntu, the functionality is built into the default file browser Nautilus:

[image]

psychemedia, 2016-05-25:
@laurensvalk is that [dimov-cz/win-sshfs](https://github.com/dimov-cz/win-sshfs) you're using?

laurensvalk, 2016-05-25:
Yes. I followed this [guide] (http://igikorn.com/sshfs-windows-8/), which lists the following dependencies with associated releases:
Dokan Library (dokany | dokan-dev fork, [0.7.X] (https://github.com/dokan-dev/dokany/releases/tag/v0.7.4))
Win-SSHFS (dimov-cz fork, [1.5.12.X] (https://github.com/dimov-cz/win-sshfs/releases/tag/1.5.12.8))

There might actually be something similar with an easier installation method. I use the Ubuntu method myself, but I just thought I'd try looking for a Windows equivalent method before posting. I haven't looked for Mac, but it might have the functionality built-in like Ubuntu.

dlech, 2016-05-25:
There is some sshfs stuff for OSX. Check out <http://osxfuse.github.io/>. Last time I used it, it wasn't really maintained and was buggy, but it seems there is some new activity on the project lately.

psychemedia, 2016-05-25:
FWIW, I've been exploring the pycharm route. One thing I;ve noticed id if you're in the file browser on the brick, and make pushes to it, then you can get the same filename showing multiple times in the listing in the File Browser display on the ev3; the listing is refreshed if you move to another folder then back again?

psychemedia, 2016-05-25:
I just had a quick look at Spyder - if you copy a remote connection file for ipython kernel running on ev3 over to host, you can then use it to launch an IPython console in Spyder that will run highlighted code in an editor window.

I can't see an immediate way of doing a fileshare though?

[image]

psychemedia, 2016-05-25:
Using Atom, with [Remote Sync](https://atom.io/packages/remote-sync) plugin:

- from Settings/Install menu, install *remote-sync*
- create new atom project;
- right click on project folder, `Remote Sync -> Configure`

Files are then synched on save:

[image]

psychemedia, 2016-05-26:
Does the [Adafruit Raspberry Pi IDE](https://github.com/adafruit/Adafruit-WebIDE) run on the ev3 (I had a quick go and it seemed to install, but not run anywhere I could see it?)

Are there any other lite RPi editors that might work with the EV3?

dlech, 2016-08-27:
Great info at https://sites.google.com/site/ev3python/

WasabiFan, 2016-08-27:
I think we should reach out to the author of that site to see if he'd be interested in contributing to the main docs with either new writing or the existing text from his website. Not only is that site thorough, that content would be a large part of #492 (which is currently stalled). At a minimum, linking to there would be nice, but I would like to avoid relying on external sites for important content.

Additionally, hearing from him on the issues he has faced and getting that feedback would be great.

[4 more replies]

closed 2017-10-03 by dlech