ev3dev/ev3dev #479: Need an easy way to change hostname

dlech, 2015-12-30:
With the new release, all devices (EV3, RPi, RPi2) will have the same hostname by default. This causes problems when you are lucky enough to have 2 or more of these devices.

You can change the name using `hostnamectl`, but there are a number of services that have to be restarted to pick up this change (bluetooth, avahi, nmbd, etc.)

Probably should be able to do this from brickman too. Also see #316.

dlech, 2016-02-25:
The new `ev3dev-config` allows you to change hostname (uses `hostnamectl` behind the scenes). It requires a reboot to be seen by all of the services that depend on it.

jabrena, 2016-02-25:
Does exist some doc about "ev3dev-config" ?

dlech, 2016-02-25:
There aren't any docs, but it should be self explanatory.

closed 2016-12-08 by dlech