ev3dev/ev3dev #480: Our website code is too valid -- titles are broken on mobile now

WasabiFan, 2015-12-31:
After removing that stray byte from our head template to make our HTML valid, titles are now cut off on mobile. They need more margin above them to not be hidden under the header. Damn HTML validation...

dlech, 2015-12-31:
Also, `img-responsive` for screenshots doesn't seem to be working on mobile anymore either.

dlech, 2015-12-31:
Never mind about the images. I think firefox is just being weird. Works in chromium.

ddemidov, 2015-12-31:
Another small thing about mobile: the main image gets out of borders (android chrome):
[image]

dlech, 2015-12-31:
@ddemidov, that image should be fixed now.

ddemidov, 2015-12-31:
@dlech, confirmed!

ddemidov, 2016-01-16:
Titles are cut off not only on mobile, but on desktop as well if browser window is thin enough. See #507.

ddemidov, 2016-01-16:
And another thing: the drop down menus do not do drop down when desktop browser is that thin. That happens when narrow the window, as soon as search box disappears.

ddemidov, 2016-01-16:
The change proposed by @moriarty in #507 helps with headers cutoff, but not with dropdown menus.

moriarty, 2016-01-16:
For the dropdown menus, there are currently two drop down links with, docs :

    <a class="dropdown-toggle" data-toggle="dropdown" href="#main-nav" ...

It should be

    <a class="dropdown-toggle" data-toggle="dropdown" href="#" ...

This changes the behaviour, now if you click on a link, dropdown will stay open until you click it closed again. When you hover on the dropdown it will display, and go away when you're not over it. But clicking on it will keep it open until you click on something else. 

The new behaviour is similar to the behaviour in the [default bootswatch darky](https://bootswatch.com/darkly/) preview site which I was comparing it to, except that they do not expand the menus on mouse over.

WasabiFan, 2016-02-27:
Migrated to ev3dev/ev3dev.github.io#112

closed 2016-02-27 by WasabiFan