ev3dev/ev3dev #481: "Broken" link on tutorial: connecting to the internet via usb

BertLindeman, 2015-12-31:
The generated url for a screenshot (I assUme) is prefixed with `http://www.ev3dev.org`
See in the generated html below: `http://www.ev3dev.orghttps://raw.githubusercontent.com`
Part of the original tutorial:
```
{% include screenshot.html source="https://raw.githubusercontent.com/ev3dev/PBrickIcons/master/demo1.png" %}
```

Part of the generated html (after `PROTIP`):
```
<span class="screenshot"><a href="http://www.ev3dev.orghttps://raw.githubusercontent.com/ev3dev/PBrickIcons/master/demo1.png"><img src="http://www.ev3dev.orghttps://raw.githubusercontent.com/ev3dev/PBrickIcons/master/demo1.png" class="img-responsive" /></a></span></p>
```

An "edit on github" would not help here (I think)

dlech, 2015-12-31:
Thanks for letting us know. You are right, it was not as easy as editing that page.

closed 2015-12-31 by dlech

BertLindeman, 2015-12-31:
Thanks David,
Confirm: Fixed.