ev3dev/ev3dev #482: Brickman WiFi password prompt needs resized

G33kDude, 2015-12-31:
The WiFi password screen you get when connecting to a new network (not the one with the keyboard, but the one just before it) has some clipping issues and needs to be resized.

dlech, 2015-12-31:
can you get a screenshot with fbgrab?

WasabiFan, 2015-12-31:
If I remember correctly, this is the one where the buttons are hanging off the bottom of the dialog.

dlech, 2015-12-31:
@WasabiFan, I fixed #463

G33kDude, 2015-12-31:
[image]

WasabiFan, 2015-12-31:
Oh, I meant that they were hanging off like in the screenshot that @G33kDude posted.

dlech, 2016-05-31:
I sort of fixed this with <https://github.com/ev3dev/ev3devKit/commit/7be2b3e5b7fd8653cd293a6a5b0871e39a9ca664>. It will still look like the picture above if the SSID is very long, but calling it fixed for now.

dlech, 2016-06-17:
Fix has been released in `brickman 0.8.0`

closed 2016-06-17 by dlech