ev3dev/ev3dev #483: Need a system info tool

dlech, 2015-12-31:
We could use a command line tool that prints out the ev3dev "system info" for people reporting bugs.

Needs to print out:

* Kernel version (`uname -r` is sufficent)
* `Hardware` and `Revision` lines from `/proc/cpuinfo`
* `/proc/device-tree/model` if it exists
* Brickman package version
* Ev3devKit package version

We should also probably start adding a file to image releases so that we can print out which image release was used.

WasabiFan, 2015-12-31:
Maybe it would make sense to roll this into the ev3dev-config tool described in #316?

dlech, 2015-12-31:
that would be a good place for it if it existed :wink:

WasabiFan, 2015-12-31:
The `*-config` tool is written as a shell script, so if we write this as a shell script as well we can combine them down the road.

WasabiFan, 2016-01-07:
If I wanted to start on this before #316, where should the code go? Do you have a repo in mind?

I figure I want to get working on something that isn't just HTML and JS, and given my recent small success with the CI shell script I'd be interested to see how this goes. And I do still plan to work on the other open work items I have :wink:

dlech, 2016-01-07:
https://github.com/ev3dev/ev3dev-tools

WasabiFan, 2016-01-07:
Will-do.

dlech, 2016-02-25:
This has been released in `ev3dev-tools 1.0.0.

closed 2016-02-25 by dlech