ev3dev/ev3dev #484: Switch website to using Sass instead of plain CSS

WasabiFan, 2016-01-02:
[Jekyll natively supports Sass](http://jekyllrb.com/docs/assets/). Instead of using custom-compiled Bootstrap and CSS additions, we should see if we can get Jekyll to build Bootstrap for us. Then our custom theme modifications won't need to be manually compiled, and we can stop using CSS hacks to modify Bootstrap code.

jabrena, 2016-01-09:
Hi,

Are you going to use libsass?

WasabiFan, 2016-01-09:
Our website uses a static site generator called Jekyll. As mentioned on the page I linked to in my post, if you put `.scss` files with the right identification ("frontmatter") in your site directory, they will be compiled automatically. So we won't actually be doing it directly.

rhempel, 2016-01-09:
@WasabiFan and @dlech - thanks for all the work you're putting intomaking this part of the site even better! I've learned a lot from watching this issue...

WasabiFan, 2016-02-25:
I have a successful local test of building Bootstrap + Bootswatch in a Jekyll site. At some point soon I'll try to get our custom adaptations extracted from our site, re-written as Sass, and then put in the right place so that Bootstrap builds with our custom stuff.

From my tests, it looks like this adds around a second of build time; this isn't optimal, but hopefully with Jekyll 3.0's new profiling tool we can speed up our site build anyway.

dlech, 2016-02-25:
jekyll has incremental builds now too, which should be helpful for development.

WasabiFan, 2016-02-27:
Migrated to ev3dev/ev3dev.github.io#111

closed 2016-02-27 by WasabiFan