ev3dev/ev3dev #485: WiFi Latency

Azaldur, 2016-01-02:
Hi,
I'm working with ev3dev since a few weeks. I encounter no problems when I connect to the robot via ssh and usb. However, when I connect to the ev3 over WiFi the ssh connection is very slow from time to time. So sometimes it works great and in the next moment I can barely type text, because it appears seconds later. My current project is controlling the robot remotely over a webinterface from my computer with a live webcam view. This also works great except from the huge WiFI latency issues which occure sometimes... independent of how far I'm away from the router .

I tested with two WiFi usb Dongles:
- Netgear WNA1100
- Hama WLAN-USB-Stick 300

Does anyone else encountered similiar issues or does have an idea how to solve it?

jabrena, 2016-01-02:
Hi, I detected something similar with a tethering connection on my smartphone. I avoid the issue, opening another ssh connection and execute the command top. I suppose that exist better solutions bit it works for me and this WiFi connection.

WasabiFan, 2016-01-02:
I have noticed this too. From my brief attempts to understand what was going on, I don't remember seeing any processes using anomalous amounts of CPU, so I really have no idea what might be causing it.

rhempel, 2016-01-02:
The kernel modules for some WiFi dongles are in less than awesome shape which causes intermittent operation. Another thing to watch out for is low batteries - WiFi dongles can take a surprising amount of power.

I prefer to use the USB connection most times when developing code, and I also have used the BlueTooth for an ssh connection.

I don't know an easy way to figure out what's going on in your situation, but one thing to try might be to look in `dmesg` output to see if the kernel logs might give us a clue as to what's going on - or you can try `journalctl -f` in a separate `ssh` session to see what's going on in real time.

dlech, 2016-01-03:
Output of `lsusb` too please (this goes for everyone who is saying "me too" as well).

WasabiFan, 2016-01-03:
> this goes for everyone who is saying "me too" as well

Just reflashed my SD card with the new image 10 minutes ago, so no existing logs from me :wink: I'll post the log info if I see the issue on the new image.

jabrena, 2016-01-03:
In my smartphone I use this option un order to not sleep the connection.

https://scontent-mad1-1.xx.fbcdn.net/hphotos-xfp1/t31.0-8/fr/cp0/e15/q65/12469385_10153337464473527_7749521901193293152_o.jpg?efg=eyJpIjoidCJ9

I suppose that the router could have a similar option.

Azaldur, 2016-01-05:
Hi,

thanks for your help.

Output of lsusb (Usb Hub, logitech camera and Hama WiFi stick):
Bus 001 Device 004: ID 046d:082b Logitech, Inc. 
Bus 001 Device 003: ID 148f:2870 Ralink Technology, Corp. RT2870 Wireless Adapter
Bus 001 Device 002: ID 1a40:0101 Terminus Technology Inc. 4-Port HUB
Bus 001 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub

Output of lsusb (WNA1100 WiFi stick)
Bus 001 Device 002: ID 0846:9030 NetGear, Inc. WNA1100 Wireless-N 150 [Atheros AR9271]
Bus 001 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub

Output of "journalctl -f" while running the remote control application (Hama Stick):
Jan 05 20:16:37 ev3dev wpa_supplicant[381]: wlan1: CTRL-EVENT-SIGNAL-CHANGE above=1 signal=-43 noise=9999 txrate=104000
Jan 05 20:16:43 ev3dev wpa_supplicant[381]: wlan1: CTRL-EVENT-SIGNAL-CHANGE above=0 signal=-53 noise=9999 txrate=117000
Jan 05 20:16:48 ev3dev wpa_supplicant[381]: wlan1: CTRL-EVENT-SIGNAL-CHANGE above=0 signal=-58 noise=9999 txrate=65000
Jan 05 20:16:57 ev3dev wpa_supplicant[381]: wlan1: CTRL-EVENT-SIGNAL-CHANGE above=0 signal=-64 noise=9999 txrate=104000
Jan 05 20:17:43 ev3dev wpa_supplicant[381]: wlan1: CTRL-EVENT-SIGNAL-CHANGE above=0 signal=-66 noise=9999 txrate=117000
Jan 05 20:18:57 ev3dev wpa_supplicant[381]: wlan1: CTRL-EVENT-SIGNAL-CHANGE above=0 signal=-72 noise=9999 txrate=78000

The connection to the robot was sometimes very bad while running journalctl. For the remote control of the robot i use an unpowered usb hub (camera+WiFi stick). However, the ssh session is also temporarily slow without an usb hub. This is independent of the battery status. It's working great sometimes and then suddenly the connection is slow (even without moving the robot).

Let me know if you need more information.

rhempel, 2016-01-05:
Please also give us the result of `uname -a`

jabrena, 2016-01-06:
The bluetooth connection could be another alternative. 

Yesterday, I started using this way and the result are interesting. My OSX provide the internet connection and the brick connects with it. This alternative has another alternative, you have not provide Wifi passwords.

Today, I were deploying with mvn several artifacts without any problem.
Consider this way as another alternative.

rhempel, 2016-01-06:
I think you are saying BlueTooth PAN networking is an alternative wireless connection. Yes it is, and it works quite well!

joshrtay, 2016-03-18:
@dlech seeing some really weird wifi latency behavior. When I start pinging the ev3 a large percent (30%) of the time the wifi just seems to be down: `Host not found request timeout`. If I give the ping about a minute it will eventually recover. Once the pings start they never stop (i.e. I don't get `Host not found` errors no matter how long I leave it pinging). However, if I stop pinging and start back up again it's very likely to be down. 

Furthermore, if I ping every 1s, my latency is roughly 100ms, but if I ping every 500ms my latency is roughly 10ms. 

All this behavior seems pretty bizarre, but I've reproduced it over and over again. Any thoughts?

rhempel, 2016-03-18:
Power saving  mode on the dongle is a possibility - please provide the kernel version using"
```
uname -a
```
and also provide the wifi dongle model, the USB PID/VID might also help

danleavitt0, 2016-03-18:
Edimax EW-7811Un dongle. I did just notice that they advertise 'Power Saving designed to support smart transmit power control and auto-idle state adjustment'. Do you think that is the problem / do you have any ideas how to turn it off?
```
Linux ev3dev 3.16.7-ckt21-9-ev3dev-ev3 #1 PREEMPT Tue Dec 15 15:16:17 CST 2015 armv5tejl GNU/Linux
```

dlech, 2016-03-18:
https://github.com/ev3dev/ev3dev/issues/204#issuecomment-65330031

Fuzzyma, 2016-10-05:
Same problem here. High pigs without any reason. Dongle (Linksys AE3000 N900 Dual-Band Wireless-N USB Adapter) works fine on my laptop:

```sh
uname -a
Linux ev3dev 4.4.19-15-ev3dev-ev3 #1 PREEMPT Sun Sep 4 16:14:11 CDT 2016 armv5tejl GNU/Linux

lsusb
Bus 001 Device 002: ID 13b1:003b Linksys
Bus 001 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub

dmesg
[   13.217465] lego-port port7: Added new device 'outD:lego-ev3-l-motor'
[   13.569594] ti_omapl_pru_suart ti_omapl_pru_suart.1: fw size 3772. downloading...
[   13.588723] ti_omapl_pru_suart.1: ttySU0 at MMIO 0x1d00000 (irq = 3, base_baud = 8250000) is a suart_tty
[   13.633141] ti_omapl_pru_suart.1: ttySU1 at MMIO 0x1d00000 (irq = 4, base_baud = 8250000) is a suart_tty
[   13.646765] ti_omapl_pru_suart ti_omapl_pru_suart.1: ti_omapl_pru_suart device registered(pru_clk=150000000, asp_clk=132000000)
[   13.871724] input: EV3 speaker beep as /devices/platform/snd-legoev3/input/input1
[   14.886535] usb 1-1: reset full-speed USB device number 2 using ohci
[   14.919631] random: nonblocking pool is initialized
[   15.119384] ieee80211 phy0: rt2x00_set_rt: Info - RT chipset 3593, rev 0405 detected
[   15.707388] ieee80211 phy0: rt2x00_set_rf: Info - RF chipset 000d detected
[   15.729672] ieee80211 phy0: Selected rate control algorithm 'minstrel_ht'
[   15.814107] usbcore: registered new interface driver rt2800usb
[   19.009754] tacho-motor motor0: Registered 'lego-ev3-l-motor' on 'outB'.
[   19.026515] tacho-motor motor1: Registered 'lego-ev3-l-motor' on 'outD'.
[   20.732314] systemd-journald[84]: Received request to flush runtime journal from PID 1
[   29.263231] serial8250: too much work for irq53
[   29.468095] serial8250: too much work for irq53
[   29.832354] serial8250: too much work for irq53
[   30.117931] serial8250: too much work for irq53
[   30.327186] serial8250: too much work for irq53
[   30.539378] serial8250: too much work for irq53
[   30.747169] serial8250: too much work for irq53
[   30.957822] serial8250: too much work for irq53
[   31.169210] serial8250: too much work for irq53
[   31.329785] ttyS1: 3 input overrun(s)
[   31.380893] serial8250: too much work for irq53
[   32.503971] ttyS1: 7 input overrun(s)
[   34.088878] ttyS1: 7 input overrun(s)
[   34.297189] serial8250_interrupt: 12 callbacks suppressed
[   34.297242] serial8250: too much work for irq53
[   34.507150] serial8250: too much work for irq53
[   34.717179] serial8250: too much work for irq53
[   34.927132] serial8250: too much work for irq53
[   35.139437] serial8250: too much work for irq53
[   35.139907] ttyS1: 5 input overrun(s)
[   35.501493] serial8250: too much work for irq53
[   35.827055] serial8250: too much work for irq53
[   35.917150] serial8250: too much work for irq53
[   36.127160] serial8250: too much work for irq53
[   36.337095] serial8250: too much work for irq53
[   36.806693] using random self ethernet address
[   36.806898] using random host ethernet address
[   38.283247] ttyS1: 2 input overrun(s)
[   38.320653] using random self ethernet address
[   38.320753] using random host ethernet address
[   38.830147] usb0: HOST MAC 32:16:53:48:d5:e7
[   38.831396] usb0: MAC 22:16:53:48:d5:e7
[   38.888405] usb1: HOST MAC 12:16:53:48:d5:e7
[   38.889361] usb1: MAC 02:16:53:48:d5:e7
[   39.427832] ttyS1: 7 input overrun(s)
[   39.719502] serial8250_interrupt: 15 callbacks suppressed
[   39.719557] serial8250: too much work for irq53
[   39.932679] serial8250: too much work for irq53
[   40.142662] serial8250: too much work for irq53
[   40.350876] serial8250: too much work for irq53
[   40.484835] ttyS1: 8 input overrun(s)
[   40.557136] serial8250: too much work for irq53
[   40.772952] serial8250: too much work for irq53
[   40.983601] serial8250: too much work for irq53
[... 198 lines cut ...]
[  109.827139] serial8250: too much work for irq53
[  110.038298] serial8250: too much work for irq53
[  110.247139] serial8250: too much work for irq53
[  110.247401] ttyS1: 3 input overrun(s)
[  110.457126] serial8250: too much work for irq53
[  110.667134] serial8250: too much work for irq53
[  110.877153] serial8250: too much work for irq53
[  111.241884] serial8250: too much work for irq53
[  111.521322] serial8250: too much work for irq53
[  111.728160] serial8250: too much work for irq53
[  112.249514] ttyS1: 5 input overrun(s)
[  114.451124] ttyS1: 7 input overrun(s)
[  115.041841] serial8250_interrupt: 13 callbacks suppressed
[  115.041898] serial8250: too much work for irq53
[  115.290158] serial8250: too much work for irq53
[  115.500710] serial8250: too much work for irq53
[  115.707170] serial8250: too much work for irq53
[  115.917161] serial8250: too much work for irq53
[  116.131226] serial8250: too much work for irq53
[  116.338312] serial8250: too much work for irq53
[  116.547319] serial8250: too much work for irq53
[  116.562598] ttyS1: 7 input overrun(s)
```

Ideas?

dlech, 2016-10-05:
> High pigs

:pig:s?

Try the latest nightly image. It should fix the `too much work for irq53` anyway, which is unrelated to Wi-Fi. https://oss.jfrog.org/list/oss-snapshot-local/org/ev3dev/brickstrap/

Fuzzyma, 2016-10-05:
Yep - my pigs are all high :P
Will try the newest build. Any solutions for the wifi part? :)

dlech, 2016-10-05:
If a new image does not help, then try a different Wi-Fi dongle, I suppose. Google search knows way more than I do about Linux Wi-Fi drivers. So far, it seems everyone who has commented has a different Wi-Fi dongle, so this could 5 different issues. Or it could have to do with the fact that the EV3 only has a USB 1.1 port and these dongles are normally only tested with USB 2.0. Other than the Edimax EW-7811Un issue with power saving mode (linked above) I don't have any suggestions.

bmegli, 2016-10-06:
Hi Fuzzyma,

You may also try `lsmod`, localize kernel module for your wifi, then `modinfo yourkernelmodule` and check what parameters it accepts, then you may play with them.

Probably:

``` bash
modifno rt2800usb
```

But before doing that make research in the internet as @dlech suggests. There's little reason to reinvent the wheel and somebody may have solved this problem already.

bmegli, 2016-10-06:
@Fuzzyma,

Also I had similiar problems with Linux upstream driver for realtek with some dongles. Not only high pigs but also occasional disconnect and totally sleepy experience.

You may also want to check which driver RaspberryPi community uses. They are much bigger community than us.

ddemidov, 2016-10-06:
[image]

Fuzzyma, 2016-10-06:
Thanks for all these tipps.
Updating to the latest nightly build didnt chance anything.
However: I got 2 other wifi dongles from friends and tested them.

First one:
Bus 001 Device 007: ID 14b2:3c22 Ralink Technology, Corp. Conceptronic C54RU v3 802.11bg Wireless Adapter [Ralink RT2571W]

Pinging router with ~50ms

Which is a huge jump forward (mine were from 5-1500ms).

Second one is:
Bus 001 Device 008: ID 0bda:8179 Realtek Semiconductor Corp.

Which has mostly pings around 5ms (with a few peaks to 500-1500ms)

I googled a bit about wifi drivers (in special the drivers used with this devices) but could not find anything related to high pings. Closest link was [this](https://sites.google.com/site/easylinuxtipsproject/reserve-7#TOC-Realtek-RTL8188EU-chipset-0bda:8179-) where multiple realtek drivers are listed and how to fix them.

I also looked into `modinfo` but I dont know where I can change the params listed there (and I am scared of doing so^^)

For now I can give the advise to try different dongles and pick the best one.

bmegli, 2016-10-06:
@Fuzzyma 

Hmm, I am almost sure I have added also tip about using iw but it's not here, strange.

You can also use iw (as opposed to say iwconfig), this is the actual (modern, current) command line tool for wireless devices in linux.

Get it:
``` bash
sudo apt-get update
sudo apt-get install iw
```

Check power saving:
``` bash
sudo iw dev wlan0 get power_save
``` 

Disable power saving
``` bash
sudo iw dev wlan0 set power_save off
```
And check if it helps.

dlech, 2016-10-06:
Do you think we should install `iw` by default? if yes, open a new issue (or pull request to ev3dev/docker-library).

bmegli, 2016-10-06:
@dlech

 I don't think typical users will need it. It's somewhat low-level.  Connman should be enough for almost all of us.

Fuzzyma, 2016-10-07:
In my case `iw` finds no device (`command failed: No such device (-19)`).
When I run `sudo iw list` it prints nothing, too.
So... no idea whats wrong

bmegli, 2016-10-07:
Hmm, you might check the interface name with:

```
ifconfig
```
e.g. (wlan0, wlan1)

But there are currenlty at at least 2 ways that user space applications can configure wireless devices from user space:
- wireless extensions, `iwconfig` command uses that (old one)
- nl80211 (netlink), `iw command` uses that (new one)

But as usual not everything works with everything so it may be that your wireless device just works with wireless extensions (`iwconfig` from the command line, this one is in `wireless-tools` package) instead.

And naturally `ifconfig` and `iwconfig` are two distinct tools.

The two tools have even distinct syntax so I am not sure you want to waste time on that ;-)

bmegli, 2016-10-07:
@Fuzzyma 

No, they would not persist. If you want them to persist:

Create file:

``` bash
sudo nano /etc/modprobe.d/8188eu.conf
```

With those lines:
``` bash
# here add some comment so that you know 7 months from now what it is ;-)
options 8188eu rtw_power_mgnt=0 rtw_ips_mode=0
```