ev3dev/ev3dev #486: Launching Apps with Brickman

laurensvalk, 2016-01-03:
With the recent addition of Open Roberta Lab, and with other IDE's possibly added in the future, how about placing these under an 'Apps' menu?

In this list of apps, maybe we can add "App Settings", allowing the user to select which apps should be launched at startup. That would eliminate the need to activate apps from the command line.

WasabiFan, 2016-01-03:
:+1: I have been thinking that the Open Roberta option was awkwardly placed -- an "apps" menu would make it easier to manage these mostly separate modules.

It would be great if the apps system were more like a plugin area, so that people could write Brickman apps without needing to modify the core code.

rhempel, 2016-01-03:
:+1: and then we can have a section for each "user" (most users will stcik with the default ``robot`` and never add their own userid) under "Apps" ?

laurensvalk, 2016-01-03:
> a section for each "user" (...) under "Apps" ? 

I'm not sure. Maybe user management could go under a "settings" page on on the main menu? Actually, I'll make a new issue for that: #487

dlech, 2016-01-03:
I would defiantly like to do something along these lines. Some of the other "apps" that we currently have enabled by default that I think some people would like to turn off are the avahi daemon (network device discovery for Linux/OS X) and nmbd (network name service for window).

Some of the other future apps we have discussed before, like an lms2012 compatibility layer would be appropriate here.

> It would be great if the apps system were more like a plugin area, so that people could write Brickman apps without needing to modify the core code.

When I first started working on brickman, I intended to have plugin support as a core feature. I quickly decided that it was way too much work. So unless there are 5 people lined up ready to make plugins for brickman, it's all going to be integrated for the time being.

> and then we can have a section for each "user"

I'm thinking this would be "system" apps as opposed to "user" apps. For "user" apps , see #129.

rhempel, 2016-01-03:
When @dlech does things defiantly, watch out!

dlech, 2016-01-03:
One of these days, I will learn to spell definitely correctly. I can't tell you how many times I've done that. :stuck_out_tongue_closed_eyes: